lvgl/lvgl #4087: Does LVGL only support updating the display by 1 pixel?

cexianl, 2023-03-24:
### Introduce the problem
When I adapt a display that can only update data according to 4 pixels, it seems that LVGL only supports 1 pixel to update the display, so the data cannot be displayed normally.

### Examples and cases

### Suggested solution
I temporarily modified the code here to solve my problem. Is there a better way for LVGL?
[image]

kisvegabor, 2023-03-27:
You can use the [`rounder_cb`](https://docs.lvgl.io/8.3/porting/display.html#optional-fields) (see line 231).

We use GitHub Issues for development-related discussions.

--

Please ask similar questions in the [Forum](https://forum.lvgl.io) to make them visible for a larger community. You can simply log in with your GitHub account.

closed 2023-03-27 by kisvegabor

cexianl, 2023-03-28:
Thanks for your help.