lvgl/lvgl #4088: lv_disp_draw_buf_t' does not name a type

copyCATandpasteDOG, 2023-03-24:
Hi everybody
This error is reported many times since 2021  over the net and seems to be related to the LVGL versions over 8.0,
but i didn't find a clear solution.
i've tried with 7.8.1 but have the same error.
I'm new to LVGL and simply want to try the lvgl widgets arduino demo .
the setup uses thoses snippets:

```
/* Change to your screen resolution */
static uint32_t screenWidth;
static uint32_t screenHeight;
static lv_disp_draw_buf_t draw_buf;
static lv_color_t *disp_draw_buf;
static lv_disp_drv_t disp_drv;

```

```
/* Initialize the display */
    lv_disp_drv_init(&disp_drv);
    /* Change the following line to your display resolution */
    disp_drv.hor_res = screenWidth;
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);

```

```
/* Initialize the (dummy) input device driver */
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register(&indev_drv);
```
Where can i find a LVGL version that fit with those snippets?

kisvegabor, 2023-03-27:
We have [this example ino](https://github.com/lvgl/lvgl/blob/release/v8.3/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino) file in the docs which should work.

copyCATandpasteDOG, 2023-03-27:
Sadly always the same error :

```
LVGL_Arduino:16:8: error: 'lv_disp_draw_buf_t' does not name a type; did you mean 'lv_draw_ctx_t'?

```

kisvegabor, 2023-03-28:
Please check in `lvgl.h` the version of LVGL.

copyCATandpasteDOG, 2023-03-28:
The error has gone :) , but replaced by a multitude of other ones,
i'm sorry but i prefer to quit.
Goodbye lgvl ,so long lgvl...

kisvegabor, 2023-03-29:
I'm sorry for that :slightly_frowning_face:

closed 2023-03-29 by kisvegabor

eagl1, 2023-05-12:
How to know the version ?

The version is set to this arrangement, but I didn't know what it mean ?

`
/***************************
 * CURRENT VERSION OF LVGL
 ***************************/
#define LVGL_VERSION_MAJOR 9
#define LVGL_VERSION_MINOR 0
#define LVGL_VERSION_PATCH 0
#define LVGL_VERSION_INFO "dev"
`

I have the same error.

kisvegabor, 2023-05-12:
You can either install LVGL v8.3.x from Arduino's library manager
[image]

or download it from the [`release/v8.3`](https://github.com/lvgl/lvgl/tree/release/v8.3) branch from GitHub.

eagl1, 2023-05-12:
I get this error:

`
Arduino: 1.8.19 (Windows 10), Board: "Generic STM32F4 series, BlackPill F411CE, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), None, Newlib Nano (default)"

D:\Program_Files\Arduino\libraries\lvgl\src\examples\arduino\LVGL_Arduino\LVGL_Arduino.ino: In function 'void my_disp_flush(lv_disp_t*, const lv_area_t*, lv_color_t*)':

LVGL_Arduino:41:26: error: cannot convert 'lv_disp_t*' {aka '_lv_disp_t*'} to 'lv_disp_drv_t*' {aka '_lv_disp_drv_t*'}

   41 |     lv_disp_flush_ready( disp );

      |                          ^~~~

      |                          |

      |                          lv_disp_t* {aka _lv_disp_t*}

In file included from d:\program_files\arduino\libraries\lvgl\src/hal/lv_hal.h:16,

                 from d:\program_files\arduino\libraries\lvgl\lvgl.h:33,

                 from D:\Program_Files\Arduino\libraries\lvgl\src/lvgl.h:17,

                 from D:\Program_Files\Arduino\libraries\lvgl\src\examples\arduino\LVGL_Arduino\LVGL_Arduino.ino:4:

d:\program_files\arduino\libraries\lvgl\src/hal/lv_hal_disp.h:335:67: note:   initializing argument 1 of 'void lv_disp_flush_ready(lv_disp_drv_t*)'

  335 | LV_ATTRIBUTE_FLUSH_READY void lv_disp_flush_ready(lv_disp_drv_t * disp_drv);

      |                                                   ~~~~~~~~~~~~~~~~^~~~~~~~

D:\Program_Files\Arduino\libraries\lvgl\src\examples\arduino\LVGL_Arduino\LVGL_Arduino.ino: In function 'void setup()':

LVGL_Arduino:100:23: error: cannot convert 'lv_disp_t*' {aka '_lv_disp_t*'} to 'lv_disp_drv_t*' {aka '_lv_disp_drv_t*'}

  100 |     lv_disp_drv_init( &disp_drv );

      |                       ^~~~~~~~~

      |                       |

      |                       lv_disp_t* {aka _lv_disp_t*}

In file included from d:\program_files\arduino\libraries\lvgl\src/hal/lv_hal.h:16,

                 from d:\program_files\arduino\libraries\lvgl\lvgl.h:33,

                 from D:\Program_Files\Arduino\libraries\lvgl\src/lvgl.h:17,

                 from D:\Program_Files\Arduino\libraries\lvgl\src\examples\arduino\LVGL_Arduino\LVGL_Arduino.ino:4:

d:\program_files\arduino\libraries\lvgl\src/hal/lv_hal_disp.h:206:39: note:   initializing argument 1 of 'void lv_disp_drv_init(lv_disp_drv_t*)'

  206 | void lv_disp_drv_init(lv_disp_drv_t * driver);

      |                       ~~~~~~~~~~~~~~~~^~~~~~

LVGL_Arduino:102:14: error: 'lv_disp_t' {aka 'struct _lv_disp_t'} has no member named 'hor_res'

  102 |     disp_drv.hor_res = screenWidth;

      |              ^~~~~~~

LVGL_Arduino:103:14: error: 'lv_disp_t' {aka 'struct _lv_disp_t'} has no member named 'ver_res'

  103 |     disp_drv.ver_res = screenHeight;

      |              ^~~~~~~

LVGL_Arduino:104:14: error: 'lv_disp_t' {aka 'struct _lv_disp_t'} has no member named 'flush_cb'
[... 85 lines cut ...]
                 from D:\Program_Files\Arduino\libraries\lvgl\src/lvgl.h:17,

                 from D:\Program_Files\Arduino\libraries\lvgl\src\examples\arduino\LVGL_Arduino\LVGL_Arduino.ino:4:

d:\program_files\arduino\libraries\lvgl\src/hal/lv_hal_indev.h:202:61: note:   initializing argument 1 of 'lv_indev_t* lv_indev_drv_register(_lv_indev_drv_t*)'

  202 | lv_indev_t * lv_indev_drv_register(struct _lv_indev_drv_t * driver);

      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~

LVGL_Arduino:127:5: error: 'lv_demo_widgets' was not declared in this scope

  127 |     lv_demo_widgets();            // OK

      |     ^~~~~~~~~~~~~~~

exit status 1

cannot convert 'lv_disp_t*' {aka '_lv_disp_t*'} to 'lv_disp_drv_t*' {aka '_lv_disp_drv_t*'}

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

`

kisvegabor, 2023-05-18:
How did you added the LVGL Arduino library?

eagl1, 2023-05-20:
I just downloaded it from the library manager and did the lvgl_conf.h, copy the file and pasted it in the libraries folder and removed the 0 and put 1. And got the same error.

Princore, 2023-05-24:
Hello there!
I have the same problem. Is there already a solution? Version 8.3.7 installed.

kisvegabor, 2023-05-24:
Hi,

Ahhhh, I terrible sorry but I've just realized that a I've messed up something in the example INO file. I've fixed it [here](https://github.com/lvgl/lvgl/blob/release/v8.3/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino) and now it should work.

brightproject, 2023-06-02:
> Ahhhh, I terrible sorry but I've just realized that a I've messed up something in the example INO file. I've fixed it [here](https://github.com/lvgl/lvgl/blob/release/v8.3/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino) and now it should work.

Don't work
```
/***************************
 * CURRENT VERSION OF LVGL
 ***************************/
#define LVGL_VERSION_MAJOR 9
#define LVGL_VERSION_MINOR 0
#define LVGL_VERSION_PATCH 0
#define LVGL_VERSION_INFO "dev"
```
```
In file included from C:\Users\Admin\Documents\Arduino\libraries\lvgl-master\src/lvgl.h:17:0,
                 from C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino:4:
c:\users\admin\documents\arduino\libraries\lvgl-master\lvgl.h:122:2: warning: #warning "You are using the development version of LVGL which is not stable at this moment. For production use the release/v8.3 branch. To silence this warning add #define LV_USE_DEV_VERSION to lv_conf.h" [-Wcpp]
 #warning "You are using the development version of LVGL which is not stable at this moment. For production use the release/v8.3 branch. To silence this warning add #define LV_USE_DEV_VERSION to lv_conf.h"
  ^~~~~~~
In file included from C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino:5:0:
C:\Users\Admin\Documents\Arduino\libraries\TFT_eSPI-master/TFT_eSPI.h:973:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
LVGL_Arduino:19:8: error: 'lv_disp_draw_buf_t' does not name a type; did you mean 'lv_draw_ctx_t'?
 static lv_disp_draw_buf_t draw_buf;
        ^~~~~~~~~~~~~~~~~~
        lv_draw_ctx_t
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino: In function 'void my_disp_flush(lv_disp_t*, const lv_area_t*, lv_color_t*)':
LVGL_Arduino:41:45: error: 'lv_color_t {aka struct lv_color16_t}' has no member named 'full'
     tft.pushColors( ( uint16_t * )&color_p->full, w * h, true );
                                             ^~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino: In function 'void my_touchpad_read(lv_indev_t*, lv_indev_data_t*)':
LVGL_Arduino:52:24: error: 'class TFT_eSPI' has no member named 'getTouch'; did you mean 'getColor'?
     bool touched = tft.getTouch( &touchX, &touchY, 600 );
                        ^~~~~~~~
                        getColor
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino: In function 'void setup()':
LVGL_Arduino:97:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch( calData );
         ^~~~~~~~
LVGL_Arduino:99:29: error: 'draw_buf' was not declared in this scope
     lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * screenHeight / 10 );
                             ^~~~~~~~
LVGL_Arduino:99:5: error: 'lv_disp_draw_buf_init' was not declared in this scope
     lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * screenHeight / 10 );
     ^~~~~~~~~~~~~~~~~~~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino:99:5: note: suggested alternative: 'lv_obj_draw_dsc_init'
     lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * screenHeight / 10 );
     ^~~~~~~~~~~~~~~~~~~~~
     lv_obj_draw_dsc_init
LVGL_Arduino:103:5: error: 'lv_disp_drv_init' was not declared in this scope
     lv_disp_drv_init( &disp_drv );
     ^~~~~~~~~~~~~~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino:103:5: note: suggested alternative: 'lv_fs_drv_init'
     lv_disp_drv_init( &disp_drv );
     ^~~~~~~~~~~~~~~~
     lv_fs_drv_init
LVGL_Arduino:108:14: error: 'lv_disp_t {aka struct _lv_disp_t}' has no member named 'draw_buf'; did you mean 'draw_buf_1'?
     disp_drv.draw_buf = &draw_buf;
              ^~~~~~~~
              draw_buf_1
LVGL_Arduino:109:5: error: 'lv_disp_drv_register' was not declared in this scope
     lv_disp_drv_register( &disp_drv );
     ^~~~~~~~~~~~~~~~~~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino:109:5: note: suggested alternative: 'lv_fs_drv_register'
     lv_disp_drv_register( &disp_drv );
     ^~~~~~~~~~~~~~~~~~~~
     lv_fs_drv_register
LVGL_Arduino:113:5: error: 'lv_indev_drv_init' was not declared in this scope
     lv_indev_drv_init( &indev_drv );
     ^~~~~~~~~~~~~~~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino:113:5: note: suggested alternative: 'lv_fs_drv_init'
     lv_indev_drv_init( &indev_drv );
     ^~~~~~~~~~~~~~~~~
     lv_fs_drv_init
LVGL_Arduino:116:5: error: 'lv_indev_drv_register' was not declared in this scope
     lv_indev_drv_register( &indev_drv );
     ^~~~~~~~~~~~~~~~~~~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino:116:5: note: suggested alternative: 'lv_fs_drv_register'
     lv_indev_drv_register( &indev_drv );
     ^~~~~~~~~~~~~~~~~~~~~
     lv_fs_drv_register
LVGL_Arduino:129:5: error: 'lv_demo_widgets' was not declared in this scope
     lv_demo_widgets();
     ^~~~~~~~~~~~~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_736808\LVGL_Arduino.ino:129:5: note: suggested alternative: 'lv_clamp_width'
     lv_demo_widgets();
     ^~~~~~~~~~~~~~~
     lv_clamp_width
exit status 1
'lv_disp_draw_buf_t' does not name a type; did you mean 'lv_draw_ctx_t'?

```

kisvegabor, 2023-06-02:
You need to use LVGL v8.

Please confirm that this part in your ino file is the same:
https://github.com/lvgl/lvgl/blob/75e6ef4e1d66a85d68160b97ef931b0717e77212/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino#L109-L113

Pay attention to `lv_indev_drv_t`.

brightproject, 2023-06-02:
> Pay attention to `lv_indev_drv_t`.

Was:
```
108  /*Initialize the (dummy) input device driver*/
109  static lv_indev_t indev_drv;
110  lv_indev_drv_init( &indev_drv );
111  indev_drv.type = LV_INDEV_TYPE_POINTER;
112  indev_drv.read_cb = my_touchpad_read;
113  lv_indev_drv_register( &indev_drv );
```
became:
```
108  /*Initialize the (dummy) input device driver*/
109  static lv_indev_drv_t indev_drv;
110  lv_indev_drv_init( &indev_drv );
111  indev_drv.type = LV_INDEV_TYPE_POINTER;
112  indev_drv.read_cb = my_touchpad_read;
113  lv_indev_drv_register( &indev_drv );
```
version 8
```
/***************************
 * CURRENT VERSION OF LVGL
 ***************************/
#define LVGL_VERSION_MAJOR 8
#define LVGL_VERSION_MINOR 3
#define LVGL_VERSION_PATCH 7
#define LVGL_VERSION_INFO ""
```
Error:
```
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_487082\LVGL_Arduino.ino: In function 'void my_disp_flush(lv_disp_t*, const lv_area_t*, lv_color_t*)':
LVGL_Arduino:41:26: error: cannot convert 'lv_disp_t*' {aka '_lv_disp_t*'} to 'lv_disp_drv_t*' {aka '_lv_disp_drv_t*'}
   41 |     lv_disp_flush_ready( disp );
      |                          ^~~~
      |                          |
      |                          lv_disp_t* {aka _lv_disp_t*}
In file included from c:\users\admin\documents\arduino\libraries\lvgl\src/hal/lv_hal.h:16,
                 from c:\users\admin\documents\arduino\libraries\lvgl\lvgl.h:33,
                 from C:\Users\Admin\Documents\Arduino\libraries\lvgl\src/lvgl.h:17,
                 from C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_487082\LVGL_Arduino.ino:4:
c:\users\admin\documents\arduino\libraries\lvgl\src/hal/lv_hal_disp.h:335:67: note:   initializing argument 1 of 'void lv_disp_flush_ready(lv_disp_drv_t*)'
  335 | LV_ATTRIBUTE_FLUSH_READY void lv_disp_flush_ready(lv_disp_drv_t * disp_drv);
      |                                                   ~~~~~~~~~~~~~~~~^~~~~~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_487082\LVGL_Arduino.ino: In function 'void my_touchpad_read(lv_indev_t*, lv_indev_data_t*)':
LVGL_Arduino:49:24: error: 'class TFT_eSPI' has no member named 'getTouch'
   49 |     bool touched = tft.getTouch( &touchX, &touchY, 600 );
      |                        ^~~~~~~~
C:\Users\Admin\AppData\Local\Temp\arduino_modified_sketch_487082\LVGL_Arduino.ino: In function 'void setup()':
LVGL_Arduino:94:9: error: 'class TFT_eSPI' has no member named 'setTouch'
   94 |     tft.setTouch( calData );
      |         ^~~~~~~~
LVGL_Arduino:100:23: error: cannot convert 'lv_disp_t*' {aka '_lv_disp_t*'} to 'lv_disp_drv_t*' {aka '_lv_disp_drv_t*'}
  100 |     lv_disp_drv_init( &disp_drv );
      |                       ^~~~~~~~~
      |                       |
      |                       lv_disp_t* {aka _lv_disp_t*}
c:\users\admin\documents\arduino\libraries\lvgl\src/hal/lv_hal_disp.h:206:39: note:   initializing argument 1 of 'void lv_disp_drv_init(lv_disp_drv_t*)'
  206 | void lv_disp_drv_init(lv_disp_drv_t * driver);
      |                       ~~~~~~~~~~~~~~~~^~~~~~
LVGL_Arduino:102:14: error: 'lv_disp_t' {aka 'struct _lv_disp_t'} has no member named 'hor_res'
  102 |     disp_drv.hor_res = screenWidth;
      |              ^~~~~~~
LVGL_Arduino:103:14: error: 'lv_disp_t' {aka 'struct _lv_disp_t'} has no member named 'ver_res'
  103 |     disp_drv.ver_res = screenHeight;
      |              ^~~~~~~
LVGL_Arduino:104:14: error: 'lv_disp_t' {aka 'struct _lv_disp_t'} has no member named 'flush_cb'
  104 |     disp_drv.flush_cb = my_disp_flush;
      |              ^~~~~~~~
LVGL_Arduino:105:14: error: 'lv_disp_t' {aka 'struct _lv_disp_t'} has no member named 'draw_buf'
  105 |     disp_drv.draw_buf = &draw_buf;
      |              ^~~~~~~~
LVGL_Arduino:106:27: error: cannot convert 'lv_disp_t*' {aka '_lv_disp_t*'} to 'lv_disp_drv_t*' {aka '_lv_disp_drv_t*'}
  106 |     lv_disp_drv_register( &disp_drv );
      |                           ^~~~~~~~~
      |                           |
      |                           lv_disp_t* {aka _lv_disp_t*}
c:\users\admin\documents\arduino\libraries\lvgl\src/hal/lv_hal_disp.h:231:50: note:   initializing argument 1 of 'lv_disp_t* lv_disp_drv_register(lv_disp_drv_t*)'
  231 | lv_disp_t * lv_disp_drv_register(lv_disp_drv_t * driver);
      |                                  ~~~~~~~~~~~~~~~~^~~~~~
LVGL_Arduino:112:25: error: invalid conversion from 'void (*)(lv_indev_t*, lv_indev_data_t*)' {aka 'void (*)(_lv_indev_t*, lv_indev_data_t*)'} to 'void (*)(_lv_indev_drv_t*, lv_indev_data_t*)' [-fpermissive]
  112 |     indev_drv.read_cb = my_touchpad_read;
      |                         ^~~~~~~~~~~~~~~~
      |                         |
      |                         void (*)(lv_indev_t*, lv_indev_data_t*) {aka void (*)(_lv_indev_t*, lv_indev_data_t*)}
LVGL_Arduino:127:5: error: 'lv_demo_widgets' was not declared in this scope
  127 |     lv_demo_widgets();            // OK
      |     ^~~~~~~~~~~~~~~
exit status 1
cannot convert 'lv_disp_t*' {aka '_lv_disp_t*'} to 'lv_disp_drv_t*' {aka '_lv_disp_drv_t*'}
```

pubFbx, 2024-02-15:
I think I'm going to give up too, I've been looking for an example that works with the current version of LGVL (^9.0.0) for 15 days. None of them work; you have to change a lot of types and functions and in the end it doesn't work. This is really worrying for the sustainability of developments. Providing an example compatible with each new version seems to me to be a minimum if you want to convince new users. Furthermore, the planned end of the cooperation with SquareLine seems to me to be one more argument for going elsewhere.

kisvegabor, 2024-02-15:
Well, the [built-in examples](https://github.com/lvgl/lvgl/tree/master/examples) are compatible with LVGL itself.  That's why they are built in. 

However if you are using Arduino be sure to:
- use the [updated ino](https://github.com/lvgl/lvgl/blob/master/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino) file
- follow the [Arduino guide](https://docs.lvgl.io/master/integration/framework/arduino.html)

Several people already confirmed that it works so I think it's really working. If you still have some fortitude, please give it an other try. :crossed_fingers: 

> Furthermore, the planned end of the cooperation with SquareLine seems to me to be one more argument for going elsewhere.

I understand that it' raises some questions so let me summarize the implications briefly. Only 2 things will change:
1. LVGL doesn't advertise SquareLine  in the future
2. I personally don't give advice to the SquareLine team. 

So if you are happy with SquareLine, you have no reason to stop using it.

ronaldtro, 2024-03-04:
Hi everyone, after a long time of reading I was able to solve all the
I was able to fix all the bugs and run an example. Here are the problems I had.

Microcontroller: Esp32 c3 super mini
screen: GC9A01A

Error #1: Verify that the project is in the correct path, I don't know if this has to do with any error but better to avoid:
C:\Usersers_TuDocuments_Arduino***My_Project***.

Error #2: File "lv_conf.h" not created:
solution: copy, paste and rename the file:
 
C:\UsersersTuDocuments\Arduino\libraries\lv_conf_template***.

In the path: 
C:\Userstu\Documents\Arduino\libraries\*lv_conf.h***.

Modify some lines of code:

/* clang-format off */
#if 1 /*Set it to "1" to enable content */

/*Color depth: 8 (A8), 16 (RGB565), 24 (RGB888), 32 (XRGB8888)*/
#define LV_COLOR_DEPTH 16 //My display is 16 bit

Error #3: lv_disp_draw_buf_t.
solution: change LVGL version to v8.3.

Error #4: ../../lv_conf.h: No such file or directory.
solution: Read well the error path, locate the file and modify:

Example error in path:

1. c:\ c:\users\tu\documents\arduino\libraries\lvglossrc\ *** lv_conf_internal.h *** ***.

2. Invoke the file "lv_conf.h" from the current file, keeping in mind how many folders upwards are found in each case:

Example: #include "../../lv_conf.h".

It will not only be in a single file, do it the necessary times until the error stops
that are necessary until the error stops coming out.

I leave an example of the example project that I used in case it is useful. Greetings.

`/*******************************************************************************
 * LVGL Hello World
 * This is a simple example for LVGL - Light and Versatile Graphics Library
 * import from: https://github.com/lvgl/lv_demos.git
 *
 * Dependent libraries:
 * LVGL: https://github.com/lvgl/lvgl.git
 *
 * LVGL Configuration file:
 * Copy your_arduino_path/libraries/lvgl/lv_conf_template.h
 * to your_arduino_path/libraries/lv_conf.h
 *
 * In lv_conf.h around line 15, enable config file:
 * #if 1 // Set it to "1" to enable content
 *
 * Then find and set:
 * #define LV_COLOR_DEPTH     16
 * #define LV_TICK_CUSTOM     1
 *
 * For SPI/parallel 8 display set color swap can be faster, parallel 16/RGB screen don't swap!
 * #define LV_COLOR_16_SWAP   1 // for SPI and parallel 8
 * #define LV_COLOR_16_SWAP   0 // for parallel 16 and RGB
 ******************************************************************************/
 
#include <lvgl.h>
// #define DIRECT_MODE // Uncomment to enable full frame buffer
[... 131 lines cut ...]
    lv_label_set_text(label, "Hello Arduino! (V" GFX_STR(LVGL_VERSION_MAJOR) "." GFX_STR(LVGL_VERSION_MINOR) "." GFX_STR(LVGL_VERSION_PATCH) ")");
    lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);

    Serial.println("Setup done");
  }
}

void loop()
{
  lv_timer_handler(); /* let the GUI do its work */

#ifdef DIRECT_MODE
#if (LV_COLOR_16_SWAP != 0)
  gfx->draw16bitBeRGBBitmap(0, 0, (uint16_t *)disp_draw_buf, screenWidth, screenHeight);
#else
  gfx->draw16bitRGBBitmap(0, 0, (uint16_t *)disp_draw_buf, screenWidth, screenHeight);
#endif
#endif // #ifdef DIRECT_MODE

#ifdef CANVAS
  gfx->flush();
#endif

  delay(5);
}`

kisvegabor, 2024-03-05:
Thank you for sharing it! I think most of these points are also mentioned [here](https://docs.lvgl.io/master/integration/framework/arduino.html). Or is there something missing?

Have you tried with LVGL v9 as well?

witaradya, 2024-08-05:
I use LVGL v8.3.10 and the problem solved

brightproject, 2024-10-12:
Good day @kisvegabor.
After some time and version `lvgl 8.4` I managed to run graphics on my device.
First attemp
[image]
Second attemp
[image]
And benchmark

https://github.com/user-attachments/assets/007b8a9d-f719-49f7-a317-cb057a66e172

I hope my [first](https://github.com/lvgl/lvgl/issues/7005) question will be answered.
And how can I set the rotation of the `rollers` not from the touchscreen, but by the value of a variable?

>     ui_Roller = lv_roller_create(ui_Screen);
>     lv_roller_set_options(ui_Roller, "0\n1\n2\n3\n4\n5\n6\n7\n8\n9", LV_ROLLER_MODE_INFINITE);

Thanks and I'm back with `lvgl`🙂

kisvegabor, 2024-10-14:
Hi,

It seems you questions are not related to the original topic of this issue, so please open new issues for these.