lvgl/lvgl #4091: virtual/multi display

kdschlosser, 2023-03-26:
### Introduce the problem
using multiple physical displays so they appear to be one display to LVGL

### Examples and cases
The i8080 and also SPI interfaces allow for more than one display to be connected to an io bus. The cost is one additional pin used on the MCU for each additional display. The hardware cost being low means it is something that a user may want to do.

### Suggested solution
I know that LVGL supports multiple displays but not in the context of being able to add the displays together to form a single display.

say you have a display that is 480 x 320 and you want to arrange them side by side horizontally. That would give you a total display resolution of 960 x 320. This is not currently how LVGL works. It keeps the display independent of each other and data can be written to one or the other but not both in the sense that they act as a single display. . Currently mirroring can be done and also if I want to have something drawn on one display or the other but if say I wanted to render an image that is 480 x 320 and have half of it displayed on one display and the other half displayed on the other display using a single lv_img object that is not possible.

LVGL would need to be set up so it is able to determine if data needs to be written to one display or the other. This is how I imagine that working.

displays get set up and callbacks get set and the resolutions for each display get set as usual. those displays would get added to a virtual display container  The first display added would have it's upper left corner positioned at 0, 0. displays added after that would have their upper left corner positioned where the user instructs it to. buffers get set to this container and not to the driver. The container has all the same functions as a typical display except those functions would handle the sorting of the buffer data and calling the callbacks to update the displays.

There would be a memory hit for doing this if the area of the virtual display spans more than a single physical display because the data contained in the buffer would have to be iterated over and separated to go to the proper display. This would only occur for horizontally positioned displays as vertical orientation can easily be handled by pointing to the start address in the buffer where a display would be and altering the x, y coords appropriately.

kdschlosser, 2023-03-28:
Here is the issue with handling the splitting of the pixel data in the flush callback is the loss of DMA transfer for one of the screens. For example on the ESP32 using i8080 you are able to add additional displays simply by paralleling everything but the CS to the second display. The second display would need a CS line of it's own. The flush CB would send to the first display and then have to wait until it was finished before it could write to the second display. If LVGL handled this internally while the data is being written to the first display LVGL could be filling a buffer for the second display. More efficient use of processor time then just sitting there waiting. This would require that double buffering is used in order to work properly.

It would also be easier from a user standpoint because they supply a flush CB for each display and LVGL hammers out what CB needs to be called.

If there is a container that is a virtual display it would also handle offsets meaning if a user didn't want to have the displays aligned with each other. Maybe they want one of them positioned 1/2 way down the first display. This is able to be set when the display gets registered with the virtual display container. they would pass the lv_disp_t along with the coords of how to position the display in reference to the first display that has been added. 

The 3 images below show what I am talking about.

[image]

[image]

[image]

kisvegabor, 2023-03-29:
>  If LVGL handled this internally while the data is being written to the first display LVGL could be filling a buffer for the second display. More efficient use of processor time then just sitting there waiting. This would require that double buffering is used in order to work properly.

Note that the invalidated ares can be arranged like this:
- disp1-area1
- disp1-area2
- disp1-area3
- disp1-area4
- disp2-area1
- disp2-area2
- disp3-area1
- disp3-area2
- disp3-area3

So LVGL will complete a display first and go to the next display after that. So the parallel port and DMA could be used effectively.

Is the 3rd picture really useful in the practice? I mean I haven't seen any devices so far where displays were arranged like this, and were considered as one large display.

kdschlosser, 2023-03-29:
> Is the 3rd picture really useful in the practice? I mean I haven't seen any devices so far where displays were arranged like this, and were considered as one large display.

I don't believe that it is. But if the ability to allow the user to set the position of a display exists then it could be done. Remember the displays don't need to be physically next to each other. This is a virtual representation of them and not a physical representation. In an automotive application I could see something like this being done for an HMI interface with rotary encoders that have displays integrated into them. or possibly a button setup using a plethora of small screens (one for each button). as an example. say you have 2 columns of buttons 5 buttons on each column each button is a 0.96" display that has a resolution of 160 x 80. That would be a total resolution of 320x400. the buttons don't align exactly with each other. the second column is slightly shifted down.

Real world application? SURE!! with the total resolution being 320, 400 this can be run on a lot of different MCUs. Using SPI or even 8 bit i8080 there would be more than enough pins available to do something like this on a number of different MCUs.

[image]

[image]

because multiple instances of LVGL are not able to be run at the same time on an MCU if you wanted to do something like the buttons and have them controlled by a single MCU you would have to do it as a single virtual display. Being able to use SquareLine Studio and arrange the graphics in an orientation that best represents the physical locations of the displays would make it easier to develop.

kdschlosser, 2023-03-29:
gotta think outside the box. This is the kind of thing that if built to show what LVGL is capable of it could push LVGL's use into the automotive world which is where LVGL would have the most use considering the environment and the number of small displays that could be used in a vehicle. think of all of the buttons and switches that are inside a vehicle where this kind of a system would work.

How hard would it be to make a physical switch where the button cap on the top was an actual display? It would actually be pretty easy to do. all 10 buttons would only require a single GPIO to read the state. Multiplexed signals using resistance. each button would have it's own resistance value applied to it and the voltage read on that pin is what would determine what switch has been pressed. button technology that has been used in vehicles for decades. almost all vehicles that have steering wheel controls only have a single wire that handles all of the buttons. The user gets the tactile feedback of a typical momentary contact button and a visual added to that for state. 

3D printer is all that would be needed to build this.

kisvegabor, 2023-04-13:
I see that this feature can be used in some cases. Please comment here if it'd be required for you. 

If it gets some traction we can can figure out how to implement it.

closed 2023-04-13 by kisvegabor

kdschlosser, 2023-04-13:
cool deal.