lvgl/lvgl #4092: Suggestion: add a unique identifier for lv_obj

bjsylvia, 2023-03-27:
### Introduce the problem

During the LVGL development process, we cannot access an lv_obj instance without saving it at creation time.

### Examples and cases

### Suggested solution

If lv_obj has a unique identifier (string name or integer), we can get the object by the root object and its ID. 

For example:
    
    /*
     1. add 'uint16_t id' in lv_obj_t struct
     2. new apis:
         void lv_obj_set_id(lv_obj_t* obj, int16_t id);
         lv_obj_t* lv_obj_get_obj_by_id(lv_obj_t* root, int16_t id);
    */

    // test

    // uid
    int16_t imgid = 201;

    // creation
    {
      lv_obj_t * img = lv_img_create(lv_scr_act());
      lv_obj_set_id((lv_obj_t*)img, imgid); 
      lv_img_set_src(img, LV_SYMBOL_OK "Ok");
      ...
    }
    // ...

    //operation 
    {
      lv_obj_t* img = (lv_img_t*)lv_obj_get_obj_by_id(lv_scr_act(), imgid);
      if (img) {
        lv_img_set_src(img, LV_SYMBOL_CLOSE "Close");
      }
    }

kisvegabor, 2023-03-28:
I was thinking about it too but for a different reason. In your example now you need to save `imgid` into a global variable. With the same effort you could save the image object too into an `lv_obj_t * img1` variable. 

I was thinking about this feature to make debuging easier. If you see only a lot of `lv_obj_t*`s it's hard to say which one is which. Adding a name or ID would help a lot with it.

kisvegabor, 2023-03-30:
Okay, we can easily by adding a new field to `lv_obj_t`. It should be optional though. 

What do you think, shall we let the user `typedef` the type of the ID or is it ok if we hard code it to `int` or `void*`?

bjsylvia, 2023-03-31:
It seems that the user to typedef the type of the ID can add flexibility and customization to the system. But void* maybe more convenient. Which idea do you like more?

kisvegabor, 2023-04-03:
I'd prefer `void *` because it's more unified. If we typedef-ed it the UIs would be less portable as the widgets would rely on the parent projects config.

fvanroie, 2023-04-17:
> During the LVGL development process, we cannot access an lv_obj instance without saving it at creation time.

I'm sure you can do this already with LVGL. I've been using `lv_obj_user_data_t` to store user specific IDs since v6.

This proposed solution looks very similar to `lv_obj_user_data_t`. The only difference is that it is specific to an ID where the user_data struct is generic and you can also store other data and IDs if needed...

I've implemented some [functions in my application](https://github.com/HASwitchPlate/openHASP/blob/f76dee8c88bb5375feba93f2090e006f78ad7d72/src/hasp/hasp_object.cpp#L21-L60) to walk the children of a screen to find the object pointers based on the ID.

> Okay, we can easily by adding a new field to lv_obj_t. It should be optional though.

My question is if this field is really needed at all? And if it is, indeed it needs to be optional because _imho_ adding a pointer _just_ to serve as an ID is not very memory efficient on a microcontroller. _(e.g. I use `uint8_t` for IDs and a pointer would be 4 bytes on ESP32)_

kisvegabor, 2023-04-18:
The only reason I think a `ID` field can be useful besides `user_data` is that `user_data` might be already used for something and therefor the `ID` can't be stored.

Posejdos, 2023-05-06:
I think that if we go this way, we should implement a mechanism that assigns IDs to every object when it's created. I think (although I'm not sure) that other GUI toolkits do that. I can see that coming in handy.

So when any object gets created, we could call something like `lv_set_id(lv_next_id())` that would set the next free ID. 
Then, when a user wants to explicitly set a known ID, it's a matter of calling `lv_set_id(known_id)`.

If we were to go with simple integer IDs, then the default IDs could always be in range _<0, 0xFFFF)_ and user IDs could always be in range _<0xFFFF, 0xFFFFFFFF)_. 

There's obviously a matter of memory usage and performance (how to get the next free ID when creating a new object? how to retrieve an object purely by its ID?). As @fvanroie asked - does LVGL even need this?

kisvegabor, 2023-05-08:
I think it'd be quite useful in a wide variety of cases, so I agree to implement it in some form. 

What about names like: `"btn1", btn2, "label1", "label2", "textarea1", "textarea1"`?

If this feature is not enabled of course nothing will happen. We could even disable to automatically assign names.

What do you think?

kisvegabor, 2023-05-10:
> How do you propose we implement automatic name assignment? Traverse through object tree and find the last one of the same type? For example, when adding a button, traverse through all the objects, find the button with the biggest ID and just set the next one?

Each widget type should have a counter which can be used for the indices. I'm not sure where is the best place to assign the the IDs though. E.g. in case of a slider (which is inherited from `lv_bar` which is inherited from `lv_obj`) the naive solutoin would be that when the `lv_obj` part is created it assigns a name (e.g. `"obj_47"`), when the the `lv_bar` part is created, it overwrites this name (e.g. `"bar_8"`), and the `lv_slider` overwrites it again (e.g. `"slider_4"`). So it could be handled in the constructors with a single static counter for each widget. The only issue is that it involves several `realloc`s.  Probably it's not a big issue though especially if we use the automatic names only for debugging. 

> Also, do we just store the ID as a char * or use void * as you suggested earlier? I think in that case char * would work better as we would know what to expect when handling the ID field (a null-terminated char array).

I also vote for `char *` here. It'd be similar to the [IDs in HTML](https://www.w3schools.com/html/html_id.asp).

XuNeo, 2023-08-24:
Hi Gabor,

Let's continue the discussion. This is a really useful debug feature beause currently when a crash happends, we have no idea of what's the faulty obj. An example crash trace is attached below.

My rough idea is that we use custom callback functions to generate `ID`, and stringify `ID` which will be used when assert happends.

We can ignore the details of how to generate `ID` for now since it could be platform dependent. For Nuttx, I'm thinking of adding a string of backtrace to where `lv_obj` is create.

So when crash happends, with the backtrace of how faulty obj is created, I can now check the source code around faulty object.

[image]

XuNeo, 2023-08-28:
Another callback should be provided to free any resource allocated in `generate` ID.

closed 2023-09-22 by kisvegabor