lvgl/lvgl #4093: Deleting animation with `lv_anim_del` does not work as expected when using `custom_exec_cb`

DCSBL, 2023-03-28:
### Introduce the problem
A animation with a `custom_exec_cb` is not deleted when the animatable object is deleted or `lv_anim_del` is called

### Examples and cases
With the code below, I create an animation where I set the object var. When I call `lv_anim_del` or `lv_obj_del` the animation is _not_ deleted/stopped, causing unexpected behaviour.

```c
lv_anim_t a;
lv_anim_init(&a);
lv_anim_set_values(&a, 0, 1000);
lv_anim_set_time(&a, 1000);
lv_anim_set_path_cb(&a, lv_anim_path_ease_out);
lv_anim_set_var(&a, (void*) animatable_object);
lv_anim_set_custom_exec_cb(&a, update_cb); // ← Uses 'context' to add some extra logic for a basic animation
lv_anim_set_user_data(&a, (void*) context);

lv_anim_start(&a);

// ... (some moment later in the program)

lv_anim_del(animatable_object, NULL);
// Animation still runs

lv_obj_del(animatable_object);
// Animation still runs, causing crashes
```

I think it happens because the animation itself is set as `var` instead of what was set before. Why is that?
https://github.com/lvgl/lvgl/blob/e5279bdbece4f5640ec48b444795a5b0de19ee6f/src/misc/lv_anim.h#L184

### Suggested solution
Why is `var` set? Anyway, I think we can at least add some documentation or a warning? Or is there a better way of doing this?

DCSBL, 2023-03-28:
After looking a bit in the code I see there is no difference internally between 'custom' and _normal_ (both use `exec_cb`), so I understand why var is set.

But is is still a bit confusing for the implementation

kisvegabor, 2023-03-29:
It's really confusion now. We will update it in v9 so that the animation will be deleted when the widget is deleted.

closed 2023-04-20 by lvgl-bot