lvgl/lvgl #4094: release/v8.3 Manually setting the value of the slider is invalid

huenrong, 2023-03-29:
### Introduce the problem

call the `lv_slider_set_value()` function without animation in the slider's LV_EVENT_VALUE_CHANGED event callback function, and the setting is invalid

### Examples and cases

```c
#define SLIDER_KNOB_MIN_VALUE 25
#define SLIDER_KNOB_MAX_VALUE 75

static void start_slider_event_handler(lv_event_t *event)
{
    if (!event)
    {
        return;
    }

    lv_event_code_t code = lv_event_get_code(event);
    lv_ui_t *ui = lv_event_get_user_data(event);

    switch (code)
    {
    case LV_EVENT_VALUE_CHANGED:
    {
        int32_t value = lv_slider_get_value(start_slider);

        value = value < SLIDER_KNOB_MIN_VALUE ? SLIDER_KNOB_MIN_VALUE : value;
        value = value > SLIDER_KNOB_MAX_VALUE ? SLIDER_KNOB_MAX_VALUE : value;

        // setting is invaid
        lv_slider_set_value(start_slider, value, LV_ANIM_OFF);

        // setting is vaid
        // lv_slider_set_value(start_slider, value, LV_ANIM_ON);

        break;
    }

    default:
        break;
    }
}
```

### Suggested solution
1. commit id `f29514aa5` don't have this problem
2. a test available modifications
- In the `lv_slider_event()` function of `lvgl/src/widgets/lv_slider.c`, make the following modifications:
```c
static void lv_slider_event(const lv_obj_class_t * class_p, lv_event_t * e)
{
    ...

    /*Advanced hit testing: react only on dragging the knob(s)*/
    if(code == LV_EVENT_HIT_TEST) {
        ...
    }
    else if(code == LV_EVENT_PRESSED) {
        ...
    }
    else if(code == LV_EVENT_PRESSING && slider->value_to_set != NULL) {
        ...

        new_value = LV_CLAMP(real_min_value, new_value, real_max_value);
        if(*slider->value_to_set != new_value) {
            
            // add this
            *slider->value_to_set = new_value;

            if(slider->value_to_set == &slider->bar.start_value) {
                lv_bar_set_start_value(obj, new_value, LV_ANIM_ON);
            }
            else {
                lv_bar_set_value(obj, new_value, LV_ANIM_ON);
            }
            res = lv_event_send(obj, LV_EVENT_VALUE_CHANGED, NULL);
            if(res != LV_RES_OK) return;
        }

    }
    else if(code == LV_EVENT_RELEASED || code == LV_EVENT_PRESS_LOST) {
        ...
    }
    else if(code == LV_EVENT_FOCUSED) {
        ...
    }
    else if(code == LV_EVENT_SIZE_CHANGED) {
        lv_obj_refresh_ext_draw_size(obj);
    }
    else if(code == LV_EVENT_REFR_EXT_DRAW_SIZE) {
        ...
    }
    else if(code == LV_EVENT_KEY) {
        ...
    }
    else if(code == LV_EVENT_DRAW_MAIN) {
        draw_knob(e);
    }
}
```
- user code set value and redrawn area
```c
lv_slider_set_value(start_slider, value, LV_ANIM_OFF);
lv_obj_invalidate(start_slider);
```

kisvegabor, 2023-03-29:
Hi,

Which LVGL version do you use? Interestingly on `master` for me it works well with `LV_ANIM_OFF` and sometimes ignores the limits with `LV_ANIM_ON`.

huenrong, 2023-03-29:
> Hi,
> 
> Which LVGL version do you use? Interestingly on `master` for me it works well with `LV_ANIM_OFF` and sometimes ignores the limits with `LV_ANIM_ON`.

i am using `release/v8.3` branch
```bash
hu@hu-virtual-machine:~/lvgl$ git branch 
  master
* release/v8.3
hu@hu-virtual-machine:~/lvgl$ 
```

kisvegabor, 2023-04-03:
Thanks, I've fixed it!

commit aa313806 (2023-04-03): fix(bar): delete running animations when a new value is set without animation

fixes #4094
commit fd08c817 (2023-04-03): fix(bar): delete running animations when a new value is set without animation

fixes #4094
closed 2023-04-03 by kisvegabor