lvgl/lvgl #4098: LV_STATE_ additions

kdschlosser, 2023-03-30:
### Introduce the problem
I guess technically speaking it is kind of a problem but also maybe an addition as well.

LV_STATE_PRESSED doesn't exactly work like it should. I have an example of the issue in the simulator (link below). If you click the knob and release immediately everything works as it should and the animation carries out like it should. But if you click the knob and hold the animation stalls and when the release finally takes place the animation does not continue it stops where it stalled.

II believe that the addition of 2 more enum items "LV_STATE_HELD" and "LV_STATE_RELEASED" would allow for an expansion of different things that can be done. To solve the issue above having an animation for when the button gets pressed and then have a style when it is held that would match the end point of the pressed animation and then set up an animation for released that would complete the animation. With something like the slider it is not known how long the knob is going to be pressed for and knowing if the knob is being held and when the knob get released would allow for handling this scenario. 

I do not feel these additional enum items should be specific to the slider widget as they would also apply to the arc, button and possibly others.

### Examples and cases
[](https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/widgets/slider/lv_example_slider_2.py&script_direct=0b5d7b19f8c9125da349d4a9aa0182514dd0fde6)

### Suggested solution

add LV_STATE_HELD and LV_STATE_RELEASED and applicable code in the widgets to handle the added states.

kdschlosser, 2023-03-31:
I was running into the "save" script problem with the simulator. I managed to work around the issue using events. The code isn't pretty using animations and events but it does work.

Now.. from what you are saying this script should work properly. The animation should render when the button is pressed and then again when the button is released.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/02b56115173e8824cd0ac97947c32cf190a01e92/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/02b56115173e8824cd0ac97947c32cf190a01e92/examples/widgets/slider/lv_example_slider_1.py&script_direct=8dd2fde308cc71054ed4b1470324606263ba7910

I want to have the same style animation run when the button gets released. it doesn't work. There would need to be a specific "released" state in order to have the button run the animation when being released.

I was not able to do what I wanted to using style transition animations and I ended up using events and running custom animations. I am still mucking about with this and it's only for example. The code is ugly and it is not exactly what I wanted.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/widgets/slider/lv_example_slider_2.py&script_direct=89a6ad91dd677c6422edbbaa5c531b287273e6ba

kisvegabor, 2023-04-03:
Could you describe exactly what you would like to achieve, so that I can try to put together an example?

kdschlosser, 2023-04-03:
Lets use an effect for a button press where it makes the ring that disappears like the ripples that happen when you throw a rock in the water.. then do the exact same thing on release (not in reverse)

kisvegabor, 2023-04-04:
I've just tried it out and realize that the main problem is that you can have transitions only on the main part...

If I replace the slider with a button it works, but to not play the ripple effect backward on release, you really need real animations.

But, it's a feature, not a bug. :slightly_smiling_face: Style transitions just add animations between the styles in each state. They are not good for any arbitrary animations.

kdschlosser, 2023-04-04:
You get what I am saying now at least..

There is a gray area as to what would be considered an animation and what is considered a style transition. and they overlap. gonna take a while to hammer out where the line is exactly.

kisvegabor, 2023-04-05:
To put it simple, style transitions are used to make state changes smooth and animated. For more magic you should use normal animations.

kdschlosser, 2023-04-05:
not a big deal learning the different things in LVGL is all. There is a lot that can be done with it just have to learn the ins and outs of how it works and what some of the different components are called.

closed 2023-04-27 by lvgl-bot