lvgl/lvgl #4100: Webserver issues

kdschlosser, 2023-03-30:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Unable to save anything in the simulator.  I click the button and it acts like it saves it but no link is given. I opened a previously saved sim and it loaded fine then made a change and clicked the save button. The link was there so I copied it and pasted it into a new browser tab  and it loaded the unsaved code.

Something is broken.

### To Reproduce

### Expected behavior
to be able to save

### Screenshots or video

kisvegabor, 2023-03-31:
cc @amirgon and @embeddedt 
Is it the certificate issue?

kdschlosser, 2023-03-31:
Let me get a screen recording of what it is doing. maybe browser specific?

There is an error in the java console when I click the button.

```
XHRPOSThttps://forum.lvgl.io:8443/save_file
CORS Missing Allow Origin

Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://forum.lvgl.io:8443/save_file. (Reason: CORS header ‘Access-Control-Allow-Origin’ missing). Status code: 413.
ERROR: Network Error app.c1bea19f5aaafe2a9941.js:2:3083815

```

and here is the expanded information for the error

**headers**
```
XHRPOSThttps://forum.lvgl.io:8443/save_file
CORS Missing Allow Origin

	
POST
	https://forum.lvgl.io:8443/save_file
Status
413
Payload Too Large
VersionHTTP/1.1
Transferred424 B (144 B size)
Referrer Policystrict-origin-when-cross-origin
Request PriorityHighest

    	
    Connection
    	keep-alive
    Content-Length
    	144
    Content-Security-Policy
    	default-src 'none'
    Content-Type
    	text/html; charset=utf-8
    Date
    	Fri, 31 Mar 2023 17:37:24 GMT
    Keep-Alive
    	timeout=5
    X-Content-Type-Options
    	nosniff
    X-Powered-By
    	Express
    	
    Accept
    	application/json, text/plain, */*
    Accept-Encoding
    	gzip, deflate, br
    Accept-Language
    	en-US,en;q=0.5
    Connection
    	keep-alive
    Content-Length
    	131454
    Content-Type
    	application/json;charset=utf-8
    Host
    	forum.lvgl.io:8443
    Origin
    	https://sim.lvgl.io
    Referer
    	https://sim.lvgl.io/
    Sec-Fetch-Dest
    	empty
    Sec-Fetch-Mode
[... 2 lines cut ...]
    	same-site
    User-Agent
    	Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/110.0
```

**request**
```
XHRPOSThttps://forum.lvgl.io:8443/save_file
CORS Missing Allow Origin

	
original_revision	"c431c7b4dfd2cc0dd9c392b74365d5af6ea986f0"
text	"import math\nimport lvgl as lv\n\nLV_PART_KNOB_CENTER = 0x090000\nLV_PART_KNOB_RING = 0x0A0000\nLV_PART_KNOB_GRIP = 0x0B0000\n\nui_img_ticks = b'\\x89PNG\\r\\n\\x1a\\n\\x00\\x00\\x00\\rIHDR\\x00\\x00\\x00\\xfa\\x00\\x00\\x00\\xfa\\x08\\x06\\x00\\x00\\x00\\x88\\xecZ=\\x00\\x00\\x00\\x01sRGB\\x00\\xae\\xce\\x1c\\xe9\\x00\\x00\\x00\\x04gAMA\\x00\\x00\\xb1\\x8f\\x0b\\xfca\\x05\\x00\\x00\\x00\\tpHYs\\x00\\x00\\x0e\\xc3\\x00\\x00\\x0e\\xc3\\x01\\xc7o\\xa8d\\x00\\x00 ;IDATx^\\xed\\xdd\\x07\\x98tiU\\xedqZ\\xe1\"F$…lue):\n super().set_width(value)\n self.__set_sizes()\n\n def set_height(self, value):\n super().set_height(value)\n self.__set_sizes()\n\n def set_size(self, width, height):\n super().set_size(width, height)\n self.__set_sizes()\n\n\nif __name__ == '__main__':\n\n import display_driver # NOQA\n \n screen = lv.scr_act()\n screen.set_style_bg_color(lv.color_hex(0x2D2D2D), 0)\n screen.set_style_bg_opa(255, 0)\n\n volume = knob_ctrl(screen)\n"
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://forum.lvgl.io:8443/save_file. (Reason: CORS header ‘Access-Control-Allow-Origin’ missing). Status code: 413.
ERROR: Network Error app.c1bea19f5aaafe2a9941.js:2:3083815

```

So what I am seeing is a limitation on how large the script can be. because of the inability to access any web resources from the simulator (images) the only way to use any kind of an external resource is it has to be embedded into the python source file. this would make it exceedingly easy to hit some kind of a transfer cap using POST. The cap is a setting that is server side.

Here is the entire JSON that is being sent

```json
{"text":"import math\nimport lvgl as lv\n\nLV_PART_KNOB_CENTER = 0x090000\nLV_PART_KNOB_RING = 0x0A0000\nLV_PART_KNOB_GRIP = 0x0B0000\n\nui_img_ticks = b'\\x89PNG\\r\\n\\x1a\\n\\x00\\x00\\x00\\rIHDR\\x00\\x00\\x00\\xfa\\x00\\x00\\x00\\xfa\\x08\\x06\\x00\\x00\\x00\\x88\\xecZ=\\x00\\x00\\x00\\x01sRGB\\x00\\xae\\xce\\x1c\\xe9\\x00\\x00\\x00\\x04gAMA\\x00\\x00\\xb1\\x8f\\x0b\\xfca\\x05\\x00\\x00\\x00\\tpHYs\\x00\\x00\\x0e\\xc3\\x00\\x00\\x0e\\xc3\\x01\\xc7o\\xa8d\\x00\\x00 ;IDATx^\\xed\\xdd\\x07\\x98tiU\\xedqZ\\xe1\"F$\\t\\xa8W\\xc2\\x15\\x19\\xa2^\\x01\\xc9\\xa2\\x12$#\\x8a\\x01\\xc9AD\\xafJ\\x14D\\xc0@P\\x11\\x01\\x05\\x04\\xae\\x80H\\xd0KPDTD\\x06\\x04\\x11AI\"\\xa3\\xa0\\x08C\\x94\\xa4dQ\\x0c\\x17\\xd7\\xaf\\xaa\\xcf|g\\x9a/TwWUW\\xf7Y\\xff\\xe7YO\\xf5\\xa9\\xe9\\xa9\\xee\\xae\\xaf\\xf6\\xd9\\xfb\\xdd\\xef\\xde\\xfb\\xdd\\xfal8G)\\xe5H\\xf3y\\xdb\\x8fe:<1\\xfa\\x95\\xf9\\x97e*\\xd4\\xa3O\\x8fkl?\\xbej\\xfb\\xb1L\\x80z\\xf4R&@\\r\\xbd\\x94\\tPC/e\\x02\\xd4\\xd0K\\x99\\x005\\xf4R&@\\r\\xfdh\\xf0\\xf1\\xe8\\xef\\xe6_\\x9e\\xc5\\x19\\xd1\\x95\\xa2w\\xcf\\xaeN\\xce\\x07\\xa3o\\x8a^=\\xbb:\\xc6\\xdb\\xa3\\x7f\\x9a\\x7fY\\x0e35\\xf4\\xc3\\xc5\\xa7\\xa2\\xfbE;\\x8d\\xf7\\xa1\\xd17G\\xff9\\xbb\\x9a\\xf3\\xde\\xad\\xad\\xad\\xd7\\xe7\\x91\\xb1\\x9e\\x8a\\xf7F\\xaf\\xdd~\\x1cs\\xfd\\xe8\\'\\xe7_\\x9e\\x85\\x9b\\xc2\\xbd\\xa2\\x8f\\xce\\xae\\xca\\xa1\\xa0\\x86\\xbe\\xb9\\xbc1\\xbaK\\xf4\\xe9\\xd9\\xd5\\x9c\\x0fD\\x8a]\\x9e=\\xbb:\\xc6\\xb9\"\\xdf\\xf7/\\xb3\\xab9_\\xb6\\xfd\\xf8\\x89\\xed\\xc7\\x93\\xf1\\x89\\xdc\\x14\\xd4S|\\xc9\\xfcr\\x86\\x9b\\xc6G\"\\xaf=\\xe6\\xf9\\xd1c\\xa3\\x7f\\x9c]\\xcd\\xf9\\xb7\\xe8\\xee\\xd1kfWe\\xe3\\xa8\\xa1\\x1f<\\xff\\x15\\xdd9z\\xd1\\xec\\xea\\x18\\x7f\\x13\\xe3{\\x8a\\xc7\\xf9\\xe5\\x8c\\x8bD\\xe7\\x8d\\xc6\\xcf\\xe1K#\\xaf\\xc3\\xe3\\x0f\\x0cF\\xbb\\x90\\xa1o?\\x0e7\\x07x-\\xaf\\xe9\\xb5\\xc7\\xfcm\\xe4\\xb9\\x0b\\xcf\\xae\\xe6\\xbc3zR\\xf4\\x86\\xd9\\xd51^\\x11\\xdd:\\x1aG\\x1a\\xe5\\x00\\xa8\\xa1\\xaf\\x17^\\xf3M\\xd1\\xff\\x9f]\\xcd\ [... cut ...] set_height(self, value):\n        super().set_height(value)\n        self.__set_sizes()\n\n    def set_size(self, width, height):\n        super().set_size(width, height)\n        self.__set_sizes()\n\n\nif __name__ == '__main__':\n\n    import display_driver  # NOQA\n    \n    screen = lv.scr_act()\n    screen.set_style_bg_color(lv.color_hex(0x2D2D2D), 0)\n    screen.set_style_bg_opa(255, 0)\n\n    volume = knob_ctrl(screen)\n","original_revision":"c431c7b4dfd2cc0dd9c392b74365d5af6ea986f0"}
```

kdschlosser, 2023-03-31:
I was able to save it once I removed some code that was actually no longer being used. IDK what the character limit is for being able to save but apparently I found there is one in place. LOL

in case you were wondering what I was working on that managed to go over this limit.

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0d9ab4ee0e591aad1970e3c9164fd7c544ecce70/examples/widgets/slider/lv_example_slider_2.py&script_direct=a6ce1641c1ebed3eab425b674291af8ea1e2214b

closed 2023-04-10 by kdschlosser