lvgl/lvgl #4102: Discussion on `img` scaling behavior

W-Mai, 2023-03-31:
### Introduce the problem

From https://github.com/lvgl/lvgl/pull/4099#issuecomment-1491934912

In web development, the `object-fit` style is related to scaling in `img`, which includes:

```cpp
object-fit: contain;
object-fit: cover;
object-fit: fill;
object-fit: none;
object-fit: scale-down;
```

On the other hand, `repeat`, `repeat-x`, `repeat-y`, and others are included in the `background-repeat` style of CSS, which includes:

```cpp
background-repeat: repeat-x;
background-repeat: repeat-y;
background-repeat: repeat;
background-repeat: space;
background-repeat: round;
background-repeat: no-repeat;
```

However, in `lvgl`, both `lv_img` and `lv_obj` have the `repeat` behavior in their `bg_img`, and they both use the `draw_img` function. I am thinking of improving the `repeat` member in `draw_ctx` and implementing it in `draw_img`.

Currently, I have only implemented `repeat` in `lv_img`, and I opened this issue to discuss where it would be appropriate to implement it.

I have already implemented `obj-fit` locally, and it works effectively. I have created `lv_img_obj_fit_t` now.

### Examples and cases

repeat: https://github.com/lvgl/lvgl/pull/4099

### Suggested solution

discussion

W-Mai, 2023-03-31:
@kisvegabor

kisvegabor, 2023-04-03:
Ah, it's really not mutually exclusive as it can happen that `fit` doesn't cover the whole area and repeat has some space to repeat the image.

I suggest finding a smallest subset from the web spec which is still required and can extended later if needed. At first look it seems logical to me:

```
fit: do we really need all these?
  object_fit_none;
  object_fit_contain; 
  object_fit_cover;
  object_fit_fill;
  object_fit_scale-down;

repeat:
  none
  repeat_x
  repeat_y

align:
  align: LV_ALIGN_TOP/TOP_LEFT/...
  offset_x
  offset_x
```

These can be applied for `lv_img` for sure. For ` bg_img` 
- If we add 9 patch support I'm not sure we need it. 
- If we add only a center image of the 9 patch we can apply all these.

I think we can remove `LV_IMG_SIZE_MODE_REAL` and `LV_IMG_SIZE_MODE_VIRTUAL` because they are not standard. On web layouts doesn't consider the transformed coordinates (i.e. if an image is zoomed it doesn't push its siblings). We can add an option in LVGL for whether to use the normal or the transformed coordinates, but it should be added to `lv_obj` and it can be added in any minor version so it's not critical now. 

What do you think?

W-Mai, 2023-04-06:
> If we add 9 patch support I'm not sure we need it.
> If we add only a center image of the 9 patch we can apply all these.

I'm confused with this two cases. 
`9 patch` images should be a single image, why would there be a case where only the center image exists?

kisvegabor, 2023-04-06:
We could have 2 APIs:
```c
lv_style_bg_img(&style, &img1);  //a single image, all fit modes apply

static const void* imgs[9] = {...}
lv_style_bg_img_9patch(&style, imgs);  //set 9 images fort the 9 patc
```

W-Mai, 2023-04-06:
Generally speaking, in the Android standard, a 9-patch image is an image with 4 marks on each side of the image, as described in the article you sent me earlier.

Why do we need to use 9 images? Is it to process the 9 images on a PC first and then use them directly in lvgl to reduce the load on embedded devices?

W-Mai, 2023-04-10:
If we follow the standard and only use one image and its four-side identification as the format of 9-patch image, then we only need to set a flag to indicate whether it is a 9-patch image for rendering. If this flag does not exist, the various repeat and scaling situations above will apply.

In this case, we reserve all the styles for `obj-fit`, including `object_fit_none`, `object_fit_contain`, `object_fit_cover`, `object_fit_fill`, and `object_fit_scale-down`. 

For `repeat`, we have `none`, `repeat_x`, and `repeat_y`.

For `align`, we have `LV_ALIGN_TOP/TOP_LEFT/...`, `offset_x`, and `offset_x`.

For the case of `background-img`, we check if the flag exists, and if it does, none of the above styles will be effective.

It can be seen that the behaviors of `img` and `background-img` are basically similar, so we can put all these operations in the same place and reuse these features.

What do you think?

kisvegabor, 2023-04-11:
It'd be very clean, but we need to be sure that our APIs support selectively scaling and repeating parts of the image. 

Maybe 9-patch image drawing should have it's own draw_task? cc @GorgonMeducer

GorgonMeducer, 2023-04-11:
@W-Mai @kisvegabor Only four side isn't sufficient for the 9-patch image. You need four-corners and 4 sides + 1 filling-image (or filling-colour).

It is safe to let the 9-path image drawing has its own draw_task. If we find it unnecessary, we can remove it later and keep an API wrapper for backward compatibility if required.

W-Mai, 2023-04-12:
@kisvegabor @GorgonMeducer 
9-patch isn't an image? Are we supposed to use 9 images to make a 9-patch?

W-Mai, 2023-04-12:
@GorgonMeducer Of course, I'm not trying to argue about what's right or wrong. I just want to understand which approach we should use when integrating 9-patch as a background, whether it should be 9 individual images or a single image with marks. This may impact the underlying rendering implementation and the design of some APIs.

kisvegabor, 2023-04-12:
Let use this image to be sure we are talking about the same thing:
[image]

So I believe what we need is 4 points from which we can slice and get all 9 images. So for each 9 patch image we need to have a descriptor too which stores this 4 points. 

[Here](https://github.com/lvgl/lvgl/pull/3294#issuecomment-1374512677) we were talking about using an `event_cb` for the images too to manage all the decoding, metadata requesting, caching, etc things in an abstract way. So an image would be described mainly by an `event_cb` which can return any data (buffer, metadata, 9patch info, etc). This way we can extend the images in a very flexible way.

W-Mai, 2023-06-30:
No one is doing the 9-patch work yet, and the img decoder rework has not moved for a long time. Can we reopen these two pr first?

1. #4099
2. #4106

kisvegabor, 2023-07-03:
The image decoder update really wasn't updated so I removed it from the roadmap of v9.0.

The parallel rendering merge is blocking a lot of development now, so I'm working on it with top priority. Now only the MicroPython examples are missing so I hope all examples will be ready today and I can merge it.

Please, let's continue it after that.

kisvegabor, 2023-07-19:
Not stale

W-Mai, 2023-08-04:
Not stale

kisvegabor, 2023-08-20:
Not stale

kisvegabor, 2023-09-04:
Not stale, should be don in v9.

kisvegabor, 2023-09-22:
Not stale

GorgonMeducer, 2023-10-08:
not stale

W-Mai, 2023-10-23:
> Not stale, now we have `scale_x` and `scale_y` so it can be warmed up.

That's great!

kisvegabor, 2023-10-31:
Closing in favor of #4733.

closed 2023-10-31 by kisvegabor