lvgl/lvgl #4105: LV_EVENT_COVER_CHECK never gets sent

kdschlosser, 2023-04-01:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
LV_EVENT_COVER_CHECK never gets sent see simulator link below. It will explain everything.

### To Reproduce
https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/02b56115173e8824cd0ac97947c32cf190a01e92/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/02b56115173e8824cd0ac97947c32cf190a01e92/examples/widgets/label/lv_example_label_4.py&script_direct=92417f8c3abcf41e6381e4fdabca6cfd9e8b4ab4

### Expected behavior
for the event to fire

### Screenshots or video

kdschlosser, 2023-04-01:
I fixed the example so now it works. The event still never fires. 

https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/02b56115173e8824cd0ac97947c32cf190a01e92/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/02b56115173e8824cd0ac97947c32cf190a01e92/examples/widgets/label/lv_example_label_4.py&script_direct=dc8f2d925df0dc9f0232bcdea6f98011c0640e7d

kisvegabor, 2023-04-03:
This is what happens: LVGL uses `LV_EVENT_COVER_CHECK` to find the first object which fully covers the area to redraw. It checks the screen first, usually it covers any area, so it's a good candidate. After that it checks the other widgets if their area fully covers the area to redraw. If so, LVGL calls the `LV_EVENT_COVER_CHECK` to see if the object really covers that area. If so, the screen doesn't need to be drawn, so that object is better candidate. 

However, in the first run, when the whole screen needs to be redrawn `LV_EVENT_COVER_CHECK` is not called for the gradient obj because it's too small anyway. Later, if an area inside the gradient obj needs to be redrawn, the event will be fired.

kdschlosser, 2023-04-03:
ahhh OK I get the mechanics of it now. I thought it would have made those calls regardless to get the object that is on top.

kisvegabor, 2023-04-04:
> ahhh OK I get the mechanics of it now. 

Great! :slightly_smiling_face: 

I close this issue then.

closed 2023-04-04 by kisvegabor