lvgl/lvgl #4112: Removing lv_keyboard_def_event_cb used to be easier before v9

Johennes, 2023-04-04:
### Introduce the problem

In v8 the default event callback of the keyboard widget could be removed with

```
lv_obj_remove_event_cb(keyboard, lv_keyboard_def_event_cb);
```

On `master` this now seems to require looping over all events to find the correct index for `lv_obj_remove_event`.

```
uint32_t num_keyboard_events = lv_obj_get_event_count(keyboard);
for (uint32_t i = 0; i < num_keyboard_events; ++i) {
    if (lv_event_dsc_get_cb(lv_obj_get_event_dsc(keyboard, i)) == lv_keyboard_def_event_cb) {
        lv_obj_remove_event(keyboard, i);
        break;
    }
}
```

### Examples and cases

This occurs in any application that wants to replace the default callback with a custom one, e.g. [unl0kr](https://gitlab.com/cherrypicker/unl0kr) or [buffyboard](https://gitlab.com/cherrypicker/buffyboard).

### Suggested solution

Maybe a convenience method could be added to remove an event callback using its function pointer?

kisvegabor, 2023-04-05:
Hi,

> Introduce the problem
In v8 the default event callback of the keyboard widget could be removed with

lv_obj_remove_event_cb(keyboard, lv_keyboard_def_event_cb);
On master this now seems to require looping over all events to find the correct index for lv_obj_remove_event.

uint32_t num_keyboard_events = lv_obj_get_event_count(keyboard);
for (uint32_t i = 0; i < num_keyboard_events; ++i) {
    if (lv_event_dsc_get_cb(lv_obj_get_event_dsc(keyboard, i)) == lv_keyboard_def_event_cb) {
        lv_obj_remove_event(keyboard, i);
        break;
    }
}
Examples and cases
This occurs in any application that wants to replace the default callback with a custom one, e.g. [unl0kr](https://gitlab.com/cherrypicker/unl0kr) or [buffyboard](https://gitlab.com/cherrypicker/buffyboard).

Suggested solution
Maybe a convenience method could be added to remove an event callback using its function pointer?

The problem is that a function pointer doesn't identify an event perfectly. E.g. you can add the same function with a different `user_data` or `LV_EVENT_...` code.

I'd rather add a special `lv_keyboard_remove_default_event()` function. What do you think about it?

Johennes, 2023-04-05:
> The problem is that a function pointer doesn't identify an event perfectly. E.g. you can add the same function with a different user_data or LV_EVENT_... code.

Ah, good point. I hadn't thought of that.

> I'd rather add a special lv_keyboard_remove_default_event() function. What do you think about it?

I gave it a shot in #4116.

closed 2023-04-27 by lvgl-bot