lvgl/lvgl #4117: Unexpected screen change behaviour (ESP32-S3)

Syrinx98, 2023-04-06:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Hello everyone, recently I have been working on a small LVGL project consisting of 2 screens using ESP32-S3-WROOM1 and LVLGL 8.2.0 (latest available for esp32s3). In the main code, I switch between the screens every second using the "lv_scr_load" function. However, it seems that the screen is not updating all at once.
I will put links for the video

### To Reproduce
I have an ESP32-S3 

with this TFT
[TFT chip](https://drive.google.com/file/d/1uIfzoxnGeGm-Ye7aW-wmaQjFH97sIC3j/view?usp=share_link)
[TFT specs](https://drive.google.com/file/d/182QBTBAu4bqcO7B21Li2i8TN1qZLKCEu/view?usp=share_link)

```c

// this is the loop in main
bool switchh = false;
	
	while (1)
	{
		if (switchh)
		{
			lv_scr_load(ui_Screen2);
			switchh = false;
		}
		else
		{
			lv_scr_load(ui_Screen1);
			switchh = true;
		}
		
		vTaskDelay(1000 / portTICK_PERIOD_MS);
	}

// the LVLGL project is simply composed by the two screens
///////////////////// ANIMATIONS ////////////////////

///////////////////// FUNCTIONS ////////////////////

///////////////////// SCREENS ////////////////////
void ui_Screen1_screen_init(void)
{
    ui_Screen1 = lv_obj_create(NULL);
    lv_obj_clear_flag(ui_Screen1, LV_OBJ_FLAG_SCROLLABLE);      /// Flags

    ui_Panel1 = lv_obj_create(ui_Screen1);
    lv_obj_set_width(ui_Panel1, lv_pct(30));
    lv_obj_set_height(ui_Panel1, lv_pct(100));
    lv_obj_set_align(ui_Panel1, LV_ALIGN_CENTER);
    lv_obj_clear_flag(ui_Panel1, LV_OBJ_FLAG_SCROLLABLE);      /// Flags
    lv_obj_set_style_bg_color(ui_Panel1, lv_color_hex(0x0017FF), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_Panel1, 255, LV_PART_MAIN | LV_STATE_DEFAULT);

}
void ui_Screen2_screen_init(void)
{
    ui_Screen2 = lv_obj_create(NULL);
    lv_obj_clear_flag(ui_Screen2, LV_OBJ_FLAG_SCROLLABLE);      /// Flags

    ui_Panel2 = lv_obj_create(ui_Screen2);
    lv_obj_set_width(ui_Panel2, lv_pct(30));
    lv_obj_set_height(ui_Panel2, lv_pct(100));
    lv_obj_set_align(ui_Panel2, LV_ALIGN_CENTER);
    lv_obj_clear_flag(ui_Panel2, LV_OBJ_FLAG_SCROLLABLE);      /// Flags
    lv_obj_set_style_bg_color(ui_Panel2, lv_color_hex(0xFF000D), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_Panel2, 255, LV_PART_MAIN | LV_STATE_DEFAULT);

}

void ui_init(void)
{
    lv_disp_t * dispp = lv_disp_get_default();
    lv_theme_t * theme = lv_theme_basic_init(dispp);
    lv_disp_set_theme(dispp, theme);
    ui_Screen1_screen_init();
    ui_Screen2_screen_init();
    ui____initial_actions0 = lv_obj_create(NULL);
    lv_disp_load_scr(ui_Screen1);
}
```

### Expected behavior
The expected behaviour is to change the screen all in the same moment, but it seems to update in 150ms  in 3 steps or so.
I also added a video where i show the usage of other widgets to show that the screen is working properly and the esp does manage it decently (You can see it with the oscilloscope screen photo, that frequency value is the VSYNC).

### Screenshots or video
VSYNC with the oscilloscope
[image]

Test for some widgets, works, cool
https://user-images.githubusercontent.com/71718486/230288609-47ebd134-da24-4014-a77e-8edf97467768.mp4

A simple change screen where the two screens do have a different coloured bar
https://user-images.githubusercontent.com/71718486/230288619-4ef44765-02ce-4e24-84ff-e7ee4358385d.mp4

Syrinx98, 2023-04-06:
i am using the RGB 565, the clock speed is 16MHzbecause i am using the PSRAM with SPI at 80Mhz and if i try to go with more than 16MHz, the image is glitched.

kisvegabor, 2023-04-06:
So you need to send 1024x600x16 = ~10Mbits. With a 16MHz SPI it takes 10/16=0.6sec. And we saw roughly this on the video. 

What I can say is that SPI is usually slow for larger screens like this. Even with a 80 MHz SPI (5 times faster) you could have max 8 FPS.

FASTSHIFT, 2023-04-08:
Is there an API for calling lvgl across threads?
Precautions: https://docs.lvgl.io/master/porting/os.html#operating-system-and-interrupts

Syrinx98, 2023-04-09:
@kisvegabor When i'm not refreshing the entire screen it works perfectly.
I managed to solve this with the double frame buffer, in fact now it refreshes the screen all at once every time with no problems but when i try to use a widget, it goes slow.
Is there any way to allocate memory for the double frame buffer in specific regions of the PSRam? This would be useful since (if i am not wrong) only some regions of the PSRam are accessibile through the GDMA.

Also, i tried the bounce buffer option so i can go at 30MHz (as reported in the esp LCD library) , but the problem with the screen refresh it is even worse, it refreshes the whole screen in smaller steps but it is slower (so i don't think the problem is the frequency)

Thank you for your time.

kisvegabor, 2023-04-11:
Great news!

> Is there any way to allocate memory for the double frame buffer in specific regions of the PSRam? This would be useful since (if i am not wrong) only some regions of the PSRam are accessibile through the GDMA.

I'm not an ESP expert, but for LVGL you only need to pass a pointer to the allocated memory. So it should be an ESp specific questions.

Syrinx98, 2023-04-11:
Thank you @kisvegabor, i am asking also on the ESP forum, however, i think the problem is that i must force lvgl to access the memory through GDMA, i don't know if it is possible, however, when i try to use the double buffer, the problem disappears, but i got less fps; Also, without the double buffer, when i have to change large quantities of pixels, the fps do go down.
However, maybe, i am using something wrong (also with LVGL functions), i will paste the .h and .c here, a hand would be really appreciated!

Thank you in advance for your time and your patience.

header
`
#pragma once

#include "lvgl.h"
#include "utils.h"
#include "esp_timer.h"
#include "esp_lcd_panel_ops.h"
#include "esp_lcd_panel_rgb.h"
#include "esp_lcd_touch.h"
#include "esp_lcd_touch_gt911.h"
#include "esp_lvgl_port.h"

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////// Please update the following configuration according to your LCD spec //////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define SETTE_POLLICI 0
#define DIECI_POLLICI 1

#define SELECTED_TFT DIECI_POLLICI

#if SELECTED_TFT == SETTE_POLLICI

#define EXAMPLE_LCD_PIXEL_CLOCK_HZ     (10 * 1000 * 1000)
#define EXAMPLE_LCD_H_RES              800
#define EXAMPLE_LCD_V_RES              480
#define HSYNC_BACK_PORCH 43
#define HSYNC_FRONT_PORCH 8
#define HSYNC_PULSE_WIDTH 4
#define VSYNC_BACK_PORCH 12
#define VSYNC_FRONT_PORCH 37
#define VSYNC_PULSE_WIDTH 37
#define MS_BETWEEN_FRAMES 32

#elif SELECTED_TFT == DIECI_POLLICI

#define EXAMPLE_LCD_PIXEL_CLOCK_HZ     (16 * 1000 * 1000)
#define EXAMPLE_LCD_H_RES              1024
#define EXAMPLE_LCD_V_RES              600
#define HSYNC_BACK_PORCH 160
#define HSYNC_FRONT_PORCH 16
#define HSYNC_PULSE_WIDTH 1
#define VSYNC_BACK_PORCH 23
#define VSYNC_FRONT_PORCH 12
#define VSYNC_PULSE_WIDTH 1
#define MS_BETWEEN_FRAMES 38 // lo schermo può andare a 60 fps a livello teorico

#endif

#define EXAMPLE_PIN_NUM_HSYNC          6
#define EXAMPLE_PIN_NUM_VSYNC          7
#define EXAMPLE_PIN_NUM_DE             15
#define EXAMPLE_PIN_NUM_PCLK           21

#define EXAMPLE_PIN_NUM_DATA0          42 // B3
#define EXAMPLE_PIN_NUM_DATA1          44 // B4
#define EXAMPLE_PIN_NUM_DATA2          43 // B5
#define EXAMPLE_PIN_NUM_DATA3          2 // B6
#define EXAMPLE_PIN_NUM_DATA4          1 // B7

#define CONFIG_EXAMPLE_DOUBLE_FB          0
#define CONFIG_EXAMPLE_AVOID_TEAR_EFFECT_WITH_SEM          0

#define EXAMPLE_PIN_NUM_DATA5          14 // G2
#define EXAMPLE_PIN_NUM_DATA6          46 // G3
[... 227 lines cut ...]
	lvgl_port_add_touch(&touch_cfg);
	

	ESP_LOGI(TAG, "Install LVGL tick timer");
	// Tick interface for LVGL (using esp_timer to generate 2ms periodic event)
	const esp_timer_create_args_t lvgl_tick_timer_args = {
		.callback = &increase_lvgl_tick,
		.name = "lvgl_tick"
	};
	esp_timer_handle_t lvgl_tick_timer = NULL;
	ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
	ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, EXAMPLE_LVGL_TICK_PERIOD_MS * 1000));

	//ESP_LOGI(TAG, "Display LVGL Scatter Chart");
}

void lvgl_task(void *pvParameters) {
	for (;;) {
		// raise the task priority of LVGL and/or reduce the handler period can improve the performance
		vTaskDelay(pdMS_TO_TICKS(MS_BETWEEN_FRAMES));
		// The task running lv_timer_handler should have lower priority than that running `lv_tick_inc`
		lv_timer_handler();
	}
}
`

Syrinx98, 2023-04-12:
I cannot allocate the buffer like that because it will not fit the data RAM, since 1 fb is approx 1 MB for 1024x600.
What i meant with my message is this: allocate both frame buffers (using double frame buffer) in external RAM that is accessible through DMA. I don't know if It is possible because as you can see in the code i pasted

`#if CONFIG_EXAMPLE_DOUBLE_FB
ESP_LOGI(TAG, "Use frame buffers as LVGL draw buffers");
ESP_ERROR_CHECK(esp_lcd_rgb_panel_get_frame_buffer(panel_handle, 2, &buf1, &buf2));
// initialize LVGL draw buffers
lv_disp_draw_buf_init(&disp_buf, buf1, buf2, EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES);`

Here i do not allocate memory manually the library do that for me (this is a piece of code from an esp example) so i do not know if it is possible to specify in which area of PSRam i allocate that frame buffers

kisvegabor, 2023-04-14:
These really seem like ESP specific questions unfortunately I don't know these either. :slightly_frowning_face:

Syrinx98, 2023-04-14:
no problem, thank you @kisvegabor, however, i opened a topic on lvgl forum to have some answers about some lvgl optimizations in config, i leave here the [link](https://forum.lvgl.io/t/lvgl-config-optimization-to-increase-fps/11476) if you want to help me.

closed 2023-04-24 by Syrinx98