lvgl/lvgl #4118: javascript: import module throw memory access out of bounds

leidto, 2023-04-06:
I'd like to import another .py file in the javascript simulator with no luck.

I have an 5.31MB file ui_assets.py. It contains images:
`Image_data = b'\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\xFF\xFF\xFF\x00\...`

Import:
`import ui_assets`

Initialized with:
`mp_js_init(1000 * 1024 * 1024)`

And passed to:
`mp_js_do_str('import ui_assets')`

I get the following error:
```
firmware.wasm:0x579a Uncaught (in promise) RuntimeError: memory access out of bounds
    at nlr_push_tail (firmware.wasm:0x579a)
    at dynCall_ii (firmware.wasm:0x55293e)
    at ret.<computed> (micropython.js:8583:31)
    at micropython.js:1260:22
    at invoke_ii (micropython.js:9361:12)
    at imports.<computed> (micropython.js:8560:31)
    at do_str (firmware.wasm:0xff46a)
    at mp_js_do_str (firmware.wasm:0x100395)
    at ret.<computed> (micropython.js:8583:31)
    at micropython.js:1260:22
```

Do you have any idea @embeddedt?

embeddedt, 2023-04-22:
Hi, sorry for the delay here.

I have recently made some patches to the simulator to enable use of larger heaps. Can you try importing this script in the v9 simulator with `memory=104857600` (which means 100MB heap) added as a query parameter and see if it gives the same error?

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?memory=104857600

leidto, 2023-04-26:
Thank you @embeddedt. Unfortunately it doesn't work:
```
Aborted(Stack overflow! Stack cookie has been overwritten at 0x00000004, expected hex dwords 0x89BACDFE and 0x2135467, but received 0x31305f6e 0x695f7265)
RuntimeError: Aborted(Stack overflow! Stack cookie has been overwritten at 0x00000004, expected hex dwords 0x89BACDFE and 0x2135467, but received 0x31305f6e 0x695f7265)
```

It works with two screens. With the third screen it's crashed. The screens contain images, texts and fonts.

Do you have any idea?

embeddedt, 2023-04-26:
Hmm, I think this is the Emscripten internal stack overflowing so perhaps try giving it a lot (`-sSTACK_SIZE=10MB`) as a test?

leidto, 2023-04-26:
Wow, it's working!

closed 2023-04-26 by leidto