lvgl/lvgl #4124: lv_canvas doesn't respect alpha channel when filling background

kdschlosser, 2023-04-07:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
when filling the background with a color you are able to supply the value of the alpha channel. No matter what this value is set to it still renders as if 255 was passed.

### To Reproduce
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/53dcb8cd998a6f98701a5702f1c9cc62246b313a/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/53dcb8cd998a6f98701a5702f1c9cc62246b313a/examples/widgets/canvas/lv_example_canvas_3.py&script_direct=7e80fa03c1b7a1d1c91902396856d409354d1eef

### Expected behavior
The background should be transparent

### Screenshots or video

kdschlosser, 2023-04-07:
just to mention I have tried changing the color format and NATIVE is the only one that will render anything other than the background, so there is something going on there as well. I have also used `lv_set_style_bg_opa(canvas, 0, 0)` as well and that doesn't change anything either.

kisvegabor, 2023-04-11:
It's a limitation in the current version of v9. It will be fixed by the release.

The problem is that now only `LV_COLOR_FORMAT_NATIVE` (e.g. RGB565, RGBX8888) is supported on the canvas, and these color formats doesn't have alpha channel.

kdschlosser, 2023-04-11:
After poking about the code in LVGL I did discover that it is not supported. so I wrote the functions needed in python to get the job done.

kdschlosser, 2023-04-11:
This is another reason that adds to my suggestion of using lv_color32_t for everything except for writing to the frame buffer.

I am a little confused as to what RGBX8888 is then. That's not 8 bit reg, 8 bit green, 8 bit blue and 8 bit alpha?? RGBX is RGBA but RGBA is not RGBX. RGBX is RGB with an additional 8 bits that are either unused or used for alpha.

kisvegabor, 2023-04-12:
> This is another reason that adds to my suggestion of using lv_color32_t for everything except for writing to the frame buffer.

Yep, we will do it in v9. See in [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md#architecture).

> I am a little confused as to what RGBX8888 is then. That's not 8 bit reg, 8 bit green, 8 bit blue and 8 bit alpha?? RGBX is RGBA but RGBA is not RGBX. RGBX is RGB with an additional 8 bits that are either unused or used for alpha.

RGBX8888 is basically with RGB888 with 8 extra bits just to make the pixels 4 byte aligned, so the alpha channel is not used.

closed 2023-05-05 by lvgl-bot