lvgl/lvgl #4130: Use lv_style_set_arc_angle_xxx to set the angle of arc

fuxiujun, 2023-04-13:
### Introduce the problem

Lvgl does not have a suitable style to set the angle of the widget

### Examples and cases

take lv_arc as an example

````
typedef struct {
    lv_obj_t obj;
    uint16_t rotation;
    uint16_t indic_angle_start;
    uint16_t indic_angle_end;
    uint16_t bg_angle_start;
    uint16_t bg_angle_end;
   ....
} lv_arc_t
````

### Suggested solution

Is it more reasonable to add style api lv_style_set_arc_angle_start and lv_style_set_arc_angle_end to set the angle of lv_arc
````
lv_style_set_arc_angle_start
lv_style_set_arc_angle_end
````

Would this be more appropriate?

kisvegabor, 2023-04-14:
When I draw the line between style properties and and widget properties, my idea was that the style properties should be what a designer can set, and the widget properties should be the runtime dynamic data adjusted by the app or developer.

Following this the arc angles should be widget properties. 

But I'm really curious how you see it.

fuxiujun, 2023-04-17:
> When I draw the line between style properties and and widget properties, my idea was that the style properties should be what a designer can set, and the widget properties should be the runtime [...]

thank you for your reply

I hope to be able to set `lv_obj_set_style_arc_angle_xxx` in `lv_them`, so that I can only set `lv_them` to adapt to different screens.

In other words, I will use `lv_obj_set_style_arc_angle_xxx` to set some default values to adapt to the size and shape of different screens.

A common example is the scrolling progress bar of a circular watch. When it is modified, I hope to use `lv_obj_set_style_arc_angle_xxx` to set it.

So I hope to add `lv_obj_set_style_arc_angle_xxx`, but it is strange if `lv_obj_set_style_arc_angle_xxx` cannot set `lv_arc`

kisvegabor, 2023-04-17:
I still feel like the angles of the arc are rather values than appearance. :slightly_frowning_face:

fuxiujun, 2023-04-18:
I understand what you mean, is it necessary to simply add `lv_obj_set_style_arc_angle_xxx` for user-defined widgets? Since we have this use case right now, I figured others might need it too.

kisvegabor, 2023-04-18:
> I understand what you mean, is it necessary to simply add lv_obj_set_style_arc_angle_xxx for user-defined widgets? 

So I think we should add `style_arc_angle_xxx` properties to keep the separation between appearance and data.

closed 2023-05-11 by lvgl-bot