lvgl/lvgl #4133: lv_obj_add_style cannot distinguish between repeated additions of style

CaoZewei, 2023-04-14:
**Introduce the problem**
When I repeatedly add styles, LVgl will repeatedly add without distinguishing between duplicates. When the number of additions exceeds 64, obj ->style_ Cnt becomes 0, causing a segmentation,

**Examples and cases**

```c++
void PageTest::timeRun()
{
	lv_obj_add_style(label, &PageStyleManage::getInstance()->label_34b_ffffffff_center, 0);
	uint16_t styleNum = label->style_cnt;
	timeRunTimes++;
	Debug("add style times %d style num = %d", timeRunTimes, styleNum);
}
``` 
```c++
01-01 09:37:29.310|I HalMsgHandleTask.cpp::run [77]  send:{"Blast": 0}
01-01 09:37:29.467|D PageTest.cpp::timeRun [94]  add style times 3 style num = 5
01-01 09:37:29.633|D PageTest.cpp::timeRun [94]  add style times 4 style num = 6
01-01 09:37:29.798|D PageTest.cpp::timeRun [94]  add style times 5 style num = 7
01-01 09:37:29.965|D PageTest.cpp::timeRun [94]  add style times 6 style num = 8
01-01 09:37:30.132|D PageTest.cpp::timeRun [94]  add style times 7 style num = 9
01-01 09:37:30.299|D PageTest.cpp::timeRun [94]  add style times 8 style num = 10
01-01 09:37:30.467|D PageTest.cpp::timeRun [94]  add style times 9 style num = 11
01-01 09:37:30.637|D PageTest.cpp::timeRun [94]  add style times 10 style num = 12
01-01 09:37:30.807|D PageTest.cpp::timeRun [94]  add style times 11 style num = 13
01-01 09:37:30.978|D PageTest.cpp::timeRun [94]  add style times 12 style num = 14
01-01 09:37:31.148|D PageTest.cpp::timeRun [94]  add style times 13 style num = 15
01-01 09:37:31.319|D PageTest.cpp::timeRun [94]  add style times 14 style num = 16
01-01 09:37:31.488|D PageTest.cpp::timeRun [94]  add style times 15 style num = 17
01-01 09:37:31.657|D PageTest.cpp::timeRun [94]  add style times 16 style num = 18
01-01 09:37:31.825|D PageTest.cpp::timeRun [94]  add style times 17 style num = 19
01-01 09:37:31.991|D PageTest.cpp::timeRun [94]  add style times 18 style num = 20
01-01 09:37:32.156|D PageTest.cpp::timeRun [94]  add style times 19 style num = 21
01-01 09:37:32.320|D PageTest.cpp::timeRun [94]  add style times 20 style num = 22
01-01 09:37:32.484|D PageTest.cpp::timeRun [94]  add style times 21 style num = 23
01-01 09:37:32.646|D PageTest.cpp::timeRun [94]  add style times 22 style num = 24
01-01 09:37:32.810|D PageTest.cpp::timeRun [94]  add style times 23 style num = 25
01-01 09:37:32.975|D PageTest.cpp::timeRun [94]  add style times 24 style num = 26
01-01 09:37:33.139|D PageTest.cpp::timeRun [94]  add style times 25 style num = 27
01-01 09:37:33.305|D PageTest.cpp::timeRun [94]  add style times 26 style num = 28
01-01 09:37:33.470|D PageTest.cpp::timeRun [94]  add style times 27 style num = 29
01-01 09:37:33.636|D PageTest.cpp::timeRun [94]  add style times 28 style num = 30
01-01 09:37:33.803|D PageTest.cpp::timeRun [94]  add style times 29 style num = 31
01-01 09:37:33.971|D PageTest.cpp::timeRun [94]  add style times 30 style num = 32
01-01 09:37:34.140|D PageTest.cpp::timeRun [94]  add style times 31 style num = 33
01-01 09:37:34.308|D PageTest.cpp::timeRun [94]  add style times 32 style num = 34
01-01 09:37:34.474|D PageTest.cpp::timeRun [94]  add style times 33 style num = 35
01-01 09:37:34.639|D PageTest.cpp::timeRun [94]  add style times 34 style num = 36
01-01 09:37:34.803|D PageTest.cpp::timeRun [94]  add style times 35 style num = 37
01-01 09:37:34.967|D PageTest.cpp::timeRun [94]  add style times 36 style num = 38
01-01 09:37:35.130|D PageTest.cpp::timeRun [94]  add style times 37 style num = 39
01-01 09:37:35.293|D PageTest.cpp::timeRun [94]  add style times 38 style num = 40
01-01 09:37:35.455|D PageTest.cpp::timeRun [94]  add style times 39 style num = 41
01-01 09:37:35.616|D PageTest.cpp::timeRun [94]  add style times 40 style num = 42
01-01 09:37:35.777|D PageTest.cpp::timeRun [94]  add style times 41 style num = 43
01-01 09:37:35.938|D PageTest.cpp::timeRun [94]  add style times 42 style num = 44
01-01 09:37:36.098|D PageTest.cpp::timeRun [94]  add style times 43 style num = 45
01-01 09:37:36.258|D PageTest.cpp::timeRun [94]  add style times 44 style num = 46
01-01 09:37:36.420|D PageTest.cpp::timeRun [94]  add style times 45 style num = 47
01-01 09:37:36.581|D PageTest.cpp::timeRun [94]  add style times 46 style num = 48
01-01 09:37:36.743|D PageTest.cpp::timeRun [94]  add style times 47 style num = 49
01-01 09:37:36.906|D PageTest.cpp::timeRun [94]  add style times 48 style num = 50
01-01 09:37:37.069|D PageTest.cpp::timeRun [94]  add style times 49 style num = 51
01-01 09:37:37.233|D PageTest.cpp::timeRun [94]  add style times 50 style num = 52
01-01 09:37:37.398|D PageTest.cpp::timeRun [94]  add style times 51 style num = 53
01-01 09:37:37.563|D PageTest.cpp::timeRun [94]  add style times 52 style num = 54
01-01 09:37:37.728|D PageTest.cpp::timeRun [94]  add style times 53 style num = 55
01-01 09:37:37.894|D PageTest.cpp::timeRun [94]  add style times 54 style num = 56
01-01 09:37:38.059|D PageTest.cpp::timeRun [94]  add style times 55 style num = 57
01-01 09:37:38.224|D PageTest.cpp::timeRun [94]  add style times 56 style num = 58
01-01 09:37:38.388|D PageTest.cpp::timeRun [94]  add style times 57 style num = 59
01-01 09:37:38.555|D PageTest.cpp::timeRun [94]  add style times 58 style num = 60
01-01 09:37:38.721|D PageTest.cpp::timeRun [94]  add style times 59 style num = 61
01-01 09:37:38.886|D PageTest.cpp::timeRun [94]  add style times 60 style num = 62
01-01 09:37:39.050|D PageTest.cpp::timeRun [94]  add style times 61 style num = 63
01-01 09:37:39.213|E lv_obj_style.c::lv_obj_add_style [92]  Asserted at expression: obj->style_cnt != 0
Segmentation fault (core dumped)

``` 
**Suggested solution**
Is it more reasonable to add style's duplicate judgment.If true, remove the corresponding style and add new one

```c++
 void lv_obj_add_style(lv_obj_t * obj, const lv_style_t * style, lv_style_selector_t selector)
{
	uint32_t i;
#ifdef	USE_MODIFY_ST	
      for(i = 0; i < obj->style_cnt; i++) {
        if(obj->styles[i].selector == selector && obj->styles[i].style == style)
        {
        	lv_obj_remove_style(obj, style, selector);
			break;
        }
    }
#endif
    trans_del(obj, selector, LV_STYLE_PROP_ANY, NULL);
   // uint32_t i;
    /*Go after the transition and local styles*/
    for(i = 0; i < obj->style_cnt; i++) {
        if(obj->styles[i].is_trans) continue;
        if(obj->styles[i].is_local) continue;
        break;
    }

    /*Now `i` is at the first normal style. Insert the new style before this*/

    /*Allocate space for the new style and shift the rest of the style to the end*/
    obj->style_cnt++;
    LV_ASSERT(obj->style_cnt != 0);
    obj->styles = lv_realloc(obj->styles, obj->style_cnt * sizeof(_lv_obj_style_t));
    LV_ASSERT_MALLOC(obj->styles);

    uint32_t j;
    for(j = obj->style_cnt - 1; j > i ; j--) {
        obj->styles[j] = obj->styles[j - 1];
    }

    lv_memzero(&obj->styles[i], sizeof(_lv_obj_style_t));
    obj->styles[i].style = style;
    obj->styles[i].selector = selector;

    lv_obj_refresh_style(obj, selector, LV_STYLE_PROP_ANY);

}
```

Would this be more appropriate?

kisvegabor, 2023-04-14:
To be honest, I thought it checks for adding the same style twice :smile: 

Just a little bit of simplification, I think we can unconditionally call `lv_obj_remove_style(obj, style, selector); ` at the beginning as it will iterate through all the styles anyway.

CaoZewei, 2023-04-17:
> To be honest, I thought it checks for adding the same style twice 😄
> 
> Just a little bit of simplification, I think we can unconditionally call `lv_obj_remove_style(obj, style, selector); ` at the beginning as it will iterate through all the styles anyway.

thank you for your reply，it is true to using` lv_obj_remove_style` unconditionally will be  better and easy.

But modifying my colleague's code is not an easy task for me, so I will keep checking it in my code and use `lv_ obj_ remove_ style`  well in next project

kisvegabor, 2023-04-17:
I've just added `lv_obj_remove_style`. :slightly_smiling_face:

commit d34958f5 (2023-04-17): fix(style): be sure to not add a style with the same selector twice

fixes #4133
closed 2023-04-17 by kisvegabor