lvgl/lvgl #4134: Error: unknown type name ‘lv_task_t’; did you mean ‘lv_bar_t’?

shallwing, 2023-04-17:
I have used this LVGL demo and successfully displayed my GUI page on imx6ull, but AN ERROR accured when I add a LVGL task in the lv_demo_widgets.c.

It told me that there were no lv_task APIs in this demo project. How can I solve this problem?

THE ERROR MESSAGE are:
1.   lvgl/demos/widgets/lv_demo_widgets.c:34:1: Error：unknown type name ‘lv_task_t’; did you mean ‘lv_bar_t’?
2.   warning：implicit declaration of function ‘lv_task_create’; did you mean ‘lv_list_create’? [-Wimplicit-function-declaration]
3.   ...(Also the error for lv_task_t)

pete-pjb, 2023-04-17:
Hi @shallwing ,

It sounds like your demo was written for version 7 and you are using version 8.

The ```lv_task_create()``` function was replaced by ```lv_timer_create()`` and the ```lv_task_t``` type was replaced by ```lv_timer_t()``` you should  be able to change the function name/types in your demo code to fix.

I hope that helps...

Cheers,

Pete

shallwing, 2023-04-17:
> Hi @shallwing ,
>
> It sounds like your demo was written for version 7 and you are using version 8.
>
> The `lv_task_create()` function was replaced by `` lv_timer_create()`` and the ``lv_task_t`type was [...]

Thanks, I have discovered the solution from file "lvgl/docs/porting/task-handler.md". It tells me that we can use timer to async call our task_handler.

pete-pjb, 2023-04-17:
Glad you have resolved 🙂 I'll close the issue.

closed 2023-04-17 by pete-pjb

shallwing, 2023-04-17:
> Glad you have resolved 🙂 I'll close the issue.

Good luck next time！Thank you！