lvgl/lvgl #4138: Compilation error status 1

jotaerrec, 2023-04-18:
I am using an esp32 and I am having problems to compile lvgl, without calling any function. I already followed the step by step of the documentation. I get an error
```
exit status 1

Compilation error: exit status 1
```
And in the detail of the compilation it does not give me any information, or error that helps me to deduce it.

I would appreciate the help! Thank you so much

kisvegabor, 2023-04-18:
It's hard to tell anything from it. Please add the whole log, maybe we find something in it.

C47D, 2023-04-25:
Any update @jotaerrec ? I have an ESP32 and can try to help you out setting up your project.
- Which version of ESP-IDF are you using?
- Which version of lvgl?
- Which controller, ESP32, ESP32C3, etc?

closed 2023-05-16 by lvgl-bot