lvgl/lvgl #4149: Consolidating signalling "child removal" to parent and adding pre-removal signal (LV_EVENT_CHILD_PRE_REMOVAL)

sparkles43, 2023-04-20:
### Introduce the problem

I have an object that can be the parent of a varying number of children, e.g. a grouped button container.
If I add a child to it (either during the creation of the child or via the set_parent method), I would like to add a
style to the added child because grouped elements should behave/look differently than non-grouped objects. If I later
remove the child from the parent (either by set_parent or swap), I would like to remove that additional style from the
child before it is removed (in case of deletion this of course does not matter).

Unfortunately, I could not realize such behavior using the `LV_EVENT_CHILD_CREATED`, `LV_EVENT_CHILD_CHANGED` and
`LV_EVENT_CHILD_DELETED` events. This is because the parent is only informed about the removal/change after it took
place (and the parameter included in the event often does not contain information about the child).

The event sources I traced down:

`lv_obj_del` deletes the child and afterwards informs the parent:
```cpp
    lv_obj_send_event(par, LV_EVENT_CHILD_CHANGED, NULL);
    lv_obj_send_event(par, LV_EVENT_CHILD_DELETED, NULL);
```
- The parent now knows that some child changed and was deleted.
- The parent could not act on the child before deletion (or other children using the knowledge about identity of the
deleted child).

`lv_obj_set_parent` first changes the parent and then sends the following events:
```cpp
    /*Notify the original parent because one of its children is lost*/
    lv_obj_send_event(old_parent, LV_EVENT_CHILD_CHANGED, obj);   // 
    lv_obj_send_event(old_parent, LV_EVENT_CHILD_DELETED, NULL);

    /*Notify the new parent about the child*/
    lv_obj_send_event(parent, LV_EVENT_CHILD_CHANGED, obj);
    lv_obj_send_event(parent, LV_EVENT_CHILD_CREATED, NULL);
```
- The old parent now knows that `obj` was changed somehow (added,deleted,moved index) and that some child was removed.
- The old parent does not know which child was deleted.

`lv_obj_move_to_index` sets the new index in the order of children and then sends:
```cpp
    lv_obj_send_event(parent, LV_EVENT_CHILD_CHANGED, NULL);
```
- The parent now knows that the order of its children was changed (but not which child is the reason for the event)

`lv_obj_swap` informs both parents that their own child is removed, then does the swap and then informs both parents
about their new child:
```cpp
    lv_obj_send_event(parent2, LV_EVENT_CHILD_DELETED, obj2);
    lv_obj_send_event(parent, LV_EVENT_CHILD_DELETED, obj1);

    parent->spec_attr->children[index1] = obj2;
    parent2->spec_attr->children[index2] = obj1;

    lv_obj_send_event(parent, LV_EVENT_CHILD_CHANGED, obj2);
    lv_obj_send_event(parent, LV_EVENT_CHILD_CREATED, obj2);
    lv_obj_send_event(parent2, LV_EVENT_CHILD_CHANGED, obj1);
    lv_obj_send_event(parent2, LV_EVENT_CHILD_CREATED, obj1);
```
- Here we have the full information: my child is gone, some child changed, I have a new child.

`lv_obj_class_init_obj` constructs a lvgl object and informs its parent (if non-null) about the new child:
```cpp
    /*Call the ancestor's event handler to the parent to notify it about the new child.
     *Also triggers layout update*/
    lv_obj_send_event(parent, LV_EVENT_CHILD_CHANGED, obj);
    lv_obj_send_event(parent, LV_EVENT_CHILD_CREATED, obj);
```
- Here we have the full information too: my child changed and I have a new child.

As we see, there are several different outcomes which even differ semantically:
- **delete**:        uses `CHANGED` and `DELETED` *without* deleted object as parameter
- **set_parent**:    uses `CHANGED` *with* the pre-existing object and `CREATED` and `DELETED` *without* the
pre-existing object as parameter
- **swap**:          uses `DELETED`, `CHANGED` and `CREATED` *with* pre-existing object as parameter
- **move_to_index**: uses `CHANGED` *without* pre-existing object as parameter
- **init**:          uses `CHANGED` and `CREATED` *with* new object as parameter

It feels somewhat inconsistent. Is there a logic behind this, I don't comprehend?

Back to my example, the only "half-way" solution to my problem seems to be (unfavorable):
When a child is transferred from one parent to a different one, the only solution to my problem would be:
- Maintaining a copy of the list of children and updating it at all child events
- Waiting for a `DELETED` event and comparing the copied-list with the already updated lv_obj_t's list
  - Thereby finding out which child was removed (but not whether it still exists)

### Examples and cases

Dynamically adding and removing buttons from a group button container which adds/removes styles to/from the
added/removed children.

### Suggested solution

I. I think the efficiency of the code handling these events (user code not lvgl code) could be improved in some cases if
the event contained information about the child that was changed.

Of course, I see that dangers may arise from always putting the child in the parameter field of every event (deletion).
I wonder what you think about a `LV_EVENT_CHILD_PRE_DELETE` event. This is actually very similar to the
`LV_EVENT_DELETED` event that is issued to an object itself (but not to its parent).

(I almost wanted to say "let's make `LV_EVENT_DELETED` bubble with the object as parameter", but that would be odd,
reduce performance and would only solve the delete-case and not the remove-from-parent case. ;) )

II. Regarding set_parent and move_to_index, I wonder whether it would be possible to just pass the object among
`DELETED` and `CREATED` from set_parent (similar to swap) and among `CHANGED` in move_to_index.

III. Could it be helpful to align the terminology a bit?
- delete: irreversibly removing an object and freeing its memory -> pass a nullptr to event
- remove: removing a child from a parent (and possibly moving it somewhere else) -> pass object to event
- add: a child was added to a parent (for whatever reason: creation, swap, set_parent) -> pass object to event

kisvegabor, 2023-04-20:
Thank you for opening this issue and summarizing the problem such a clear way. I see that it's really messy know.

Following your terminology I think these type of events shouldn't care about "delete". It shouldn't matter for the parent whether the child was moved to an other parent or was deleted. So we need only "remove" and I think we should rename the event to `LV_EVENT_CHILD_REMOVED`.

So we have 3 actions, let's summarize how we can manage them:
1. Created: Simple, the object is always valid on the given parent. The event is sent when the child fully ready.
2. Removed: Sent after the actual removal and pass the object if it wan't a delete, and pass NULL in case of delete. 
3. Changed: Send it after create or removal, pass the object (except if it was deleted). 

It seems to me if we followed these principles your issue would be solved. What do you think?

sparkles43, 2023-04-21:
> Following your terminology I think these type of events shouldn't care about "delete". It shouldn't matter for the parent whether the child was moved to an other parent or was deleted. So we need only "remove" and I think we should rename the event to LV_EVENT_CHILD_REMOVED.

I think so too. In fact, if we change `LV_EVENT_CHILD_DELETED` to `LV_EVENT_CHILD_REMOVED`, we should analogously change `LV_EVENT_CHILD_CREATED` to `LV_EVENT_CHILD_ADDED`. In both cases, the creation and deletion of a child does not matter. For *adding a child* the child is expected to be valid in both cases creation and adding.

> So we have 3 actions, let's summarize how we can manage them:
>    Created: Simple, the object is always valid on the given parent. The event is sent when the child fully ready.
>    Removed: Sent after the actual removal and pass the object if it wasn't a delete, and pass NULL in case of delete.
>    Changed: Send it after create or removal, pass the object (except if it was deleted).

At first glance I think this is good and we have a clear definition of situations.

However, I am not sure that there aren't any problems with ownership when we are raising events. At the moment for example, *swap* first signals that the children are `DELETED` (while still having their old parents), then swaps parents and then signals `CHANGED` and `CREATED`. Other functions first move the object and then signal `DELETED` (e.g. set_parent).

What I am concerned about is that the event-parameter sometimes still has its old parent when events arise and sometimes it already has its new parent. If someone receives an event, there should be a common pattern or the documentation has to explicitly state that the parent pointers of the objects in the events' parameters must not be used. We should probably discuss this, too.

> It seems to me if we followed these principles your issue would be solved. What do you think?
Yes, I think so. Minimum change: If set_parent would pass the objects in the events' parameters, I could remove the style from the button in all cases, except real deletion (which does not matter to me).

kisvegabor, 2023-04-21:
> we should analogously change LV_EVENT_CHILD_CREATED to LV_EVENT_CHILD_ADDED

I agree!

> However, I am not sure that there aren't any problems with ownership when we are raising events. At the moment for example, swap first signals that the children are DELETED (while still having their [...]

We need to make it consistent. We can either
1. send `REMOVED` after move, so the child is already not part of the parent to which the event is sent. It has the advantage that the parent sees the new child list (without the removed child) so it can do things accordingly.
2. send `TO_BE_REMOVED` (or `PRE_REMOVE`?) before move, so the child is still part of the original parent to which the event is sent. It has the advantage that the child parameter of the event is always valid.

As I think about it more 2) seems better to me. E.g. imagine that we have 3 buttons on a container widget and the 2nd button is deleted. Maybe you want change the style of the other buttons if the middle button disappears. So it's important to know the original index of to-be-removed button.

> Yes, I think so. Minimum change: If set_parent would pass the objects in the events' parameters, I could remove the style from the button in all cases, except real deletion (which does not matter to me).

Okay, it seems we are on the right track!

sparkles43, 2023-04-24:
> As I think about it more 2) seems better to me. E.g. imagine that we have 3 buttons on a container widget and the 2nd button is deleted. Maybe you want change the style of the other buttons if the middle button disappears. So it's important to know the original index of to-be-removed button.

Yes, that sounds good. The only decision left is, whether both events `REMOVED` and `TO_BE_REMOVED` should coexist, or whether only one of them should remain. I can imagine that both situations (1 and 2) could be useful for developers. (Are we willing to create one more event for it?)

Example for case 1:
a child button was removed and we need to do some custom layout recalculation. Therefore we must be sure that the child is really gone from the list of children. (OK, there might be much better examples out there :sweat_smile:)

Another thing:
When we add a TO_BE_REMOVED event, will the `REMOVED` event always have a null pointer as event parameter? The child will be gone and its parent pointer will differ. I think there would be no reason to use the former child after its ownership was transferred to a new parent.

> Okay, it seems we are on the right track!
Indeed :smile:

kisvegabor, 2023-04-24:
> Example for case 1:
a child button was removed and we need to do some custom layout recalculation. Therefore we must be sure that the child is really gone from the list of children. (OK, there might be much better examples out there sweat_smile)

For these kind of things we can use the `LV_EVENT_CHILD_CHANGED` event. If the parameter is `NULL` we don't know which child was changed (maybe multiple children, or a child was deleted).

sparkles43, 2023-05-24:
@kisvegabor I am sorry that the discussion staled. In the last weeks I was pretty busy with other tasks. The overhaul is definitely something that's on my mind, but I am not sure when I can start working on the code.

Are you asking to align this change with the v9 release? I guess it should only be included in a Major version change....

Back to the nitty gritty:
> For these kind of things we can use the LV_EVENT_CHILD_CHANGED event. If the parameter is NULL we don't know which child was changed (maybe multiple children, or a child was deleted).

Initially I was not sure whether "LV_EVENT_CHILD_CHANGED with NULL" would be clear enough as it also occurs when reordering children - but in the end I could not make up a case against it. I think it will do just fine.

sparkles43, 2023-05-24:
> Could you summarize what is latest plan? 🙂
Next week I will have time to see whether I may schedule this change. I guess it's not a too large task but I want to make sure I don't break such a core feature.

_____

To make sure everything will work, we might need to take a look at the bubbling behavior of these events. At least I haven't done it so far.

From the source code:
```cpp
    LV_EVENT_CHILD_CHANGED,       /**< Child was removed, added, or its size, position were changed */
    LV_EVENT_CHILD_CREATED,       /**< Child was created, always bubbles up to all parents*/
    LV_EVENT_CHILD_DELETED,       /**< Child was deleted, always bubbles up to all parents*/
```
It is mentioned that `LV_EVENT_CHILD_CREATED` and `LV_EVENT_CHILD_DELETED` are bubbling up to all parents - I assume `CHANGED` does not.
If `LV_EVENT_CHILD_CHANGED` is the final indicator that an object was deleted and we don't bubble it up, does that impact the behavior of the parents negatively?

kisvegabor, 2023-05-24:
> Next week I will have time to see whether I may schedule this change. I guess it's not a too large task but I want to make sure I don't break such a core feature.

Sounds great, thank you! Actually by "summarizing the latest plan" I mean summarizing the latest idea technically. A lot of things were said and it'd be great to see a short and simple plan about the planned changes.

> To make sure everything will work, we might need to take a look at the bubbling behavior of these events. At least I haven't done it so far.

For `CREATED` we need the "always bubble" for sure, as we can't enable the `LV_OBJ_FLAG_EVENT_BUBBLE` flag on the  children before they are created.  

For `DELETED` (will be renamed  `TO_BE_REMOVED`) it's not mandatory, but for consistency with `CREATED` I think we should keep it.

If we `CHILD_CHANGED` with `NULL` parameter to signal child deletion I think we should always bubble it too.

sparkles43, 2023-05-25:
The problem with bubbeling `CHILD_CHANGED` is that `CHILD_CHANGED` can have a lot of reasons.

> /**< Child was removed, added, or its size, position were changed */

If we decide to make it bubble too. That means higher processing load throughout the tree.
If we decide to only bubble `CHILD_CHANGED` with `NULL`, it will create odd special cases in a otherwise simple event logic.

I wonder whether sacrificing another event for both `CHILD_DELETED` (bubbeling) and `TO_BE_REMOVED` (bubbeling) would keep the complexity down.

kisvegabor, 2023-05-25:
Let's understand what is the purpose of bubbling `CREATE` and `DELETE`. I think it's good because this way the user can add an event e.g. to a single parent, and if any children or grandchildren are created, they can be styled, or "hooked" from this single parent.

Besides "child create" and "child delete/remove" the `CHILD_CHANGED` event is used to tell the parent that now the children's bounding box look different. That is their position or size or index (a.k.a. order)  has been changed. It is really interesting only to the direct parent to update its layout. The grandparent will be notified too if the parent size changes due to child change.

Knowing which child triggered the event is not really useful as the layout needs to be recalculated anyway. 

> If we decide to only bubble CHILD_CHANGED with NULL, it will create odd special cases in a otherwise simple event logic.

We could resolve it by different name to this event. E.g. `LV_EVENT_LAYOUT_UPDATE_REQUEST`.

What do you think?

kisvegabor, 2023-06-09:
Not stale

kisvegabor, 2023-07-03:
I've pinned this issue to not be closed automtically as stale.

kisvegabor, 2023-12-02:
@sparkles43 I would like to add it before releasing v9 so I wonder if you will have time work on it in nearvfeature.

kisvegabor, 2024-01-14:
Closing this issue due to lack of activity.

sparkles43, 2024-03-18:
@kisvegabor I guess, I owe you an apology. Sorry, that I abandoned this issue. At the time it became a bit chaotic and I lacked time to look further into it. I hope this issue will at least serve as a good reference for future development. :disappointed:

kisvegabor, 2024-03-18:
Hi @sparkles43,

It's great to hear from you again! And no worries. I understand that most people are here in their free time. So every single comment is highly appreciated. 

Unfortunately, I didn't have time to pick up this issue. If your life calmer now, and you are still interested in it, I would be happy repopen this issue.

sparkles43, 2024-03-19:
I am definitely interested but I think I will keep this issue in my mind and ask you to reopen it, once I have a foreseeable amount of time to actually finish it. :)

kisvegabor, 2024-03-19:
Okay, reopened :slightly_smiling_face:

kisvegabor, 2024-09-05:
Closing due to inactivity. 

I still believe that fixing this issue is important and if anyone would like to pick it up, please leave a comment.

closed 2024-09-05 by kisvegabor