lvgl/lvgl #4164: Texture LRU cache insertion failure not accounted in the SDL GPU driver

ReimuNotMoe, 2023-04-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When using the SDL GPU driver, it is possible that the description (the "key", idk how to call it) of a texture does not fit in the LRU cache at all. However, in the texture cache management functions (`lv_draw_sdl_texture_cache.c`), the return value of `lv_lru_set` is not checked at all. If the LRU cache insertion failed, the texture pointer will be lost forever, and causing serious GPU memory leaks.

### To Reproduce
- Use the SDL GPU driver and a monitor with 4K or larger resolution
- The SDL version needs to be newer than 2.0.6, to support `LV_GPU_SDL_CUSTOM_BLEND_MODE`
- Create a slider bar that occupies the entire horizontal space
- Drag the slider back and forth and watch the RAM usage grow exponentially

This has been reproduced on Linux + NVIDIA GTX 1660Ti and MacOS + AMD RX580

### Expected behavior
If a texture cannot be put into the LRU cache, it should be destroyed immediately after use.

### Screenshots or video
[image]

[image]

kisvegabor, 2023-04-27:
@mariotaku, could you take look at this?

mariotaku, 2023-04-27:
@kisvegabor Sure!

ReimuNotMoe, 2023-05-05:
> @ReimuNotMoe Would you like to give #4173 a shot? I tested it with Valgrind and it's looking good so far.

Yes, it's working nicely now. Thanks!

mariotaku, 2023-05-05:
@ReimuNotMoe Cool.
@kisvegabor the PR is ready for review :)

kisvegabor, 2023-05-08:
The PR is merged, I close this issue.

PR #4173 (2023-05-08): Destroy Texture After Use if Not Stored in Cache
### Description of the feature or fix

This fixed #4164.

### Checkpoints
- [x] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.p [... cut]