lvgl/lvgl #4165: FPS calculation is wrongly calculating the render time

nicusorcitu, 2023-04-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

The original discussion started from [here](https://github.com/lvgl/lvgl/issues/4016#issuecomment-1521919446)

nicusorcitu, 2023-04-26:
[0001-fix-perf-Fix-the-FPS-calculation-by-measuring-the-ex.patch](https://github.com/lvgl/lvgl/files/11333807/0001-fix-perf-Fix-the-FPS-calculation-by-measuring-the-ex.patch)

This is how it looks like with the patch applied:
[image]

nicusorcitu, 2023-04-26:
Hi Gabor,
I attached a patch which basically moves the FPS calculation after the render is being done.
That was to avoid using the events (render_start and render_ready) as you suggested.

Do you think is a good solution?
Thanks, Nicu.

kisvegabor, 2023-04-26:
Thank you, Nicu. It looks good in general. Just only 1 thing: I think instead of
```c
        if(perf_monitor.frame_cnt == 0) {
            fps = fps_limit;
            time = time_limit;
        }
```

we should add
```c
        if(perf_monitor.frame_cnt == 0) {
            fps = fps_limit;
            time = 0;   //If no frame was rendered, the time should be 0ms 
        }
```

nicusorcitu, 2023-04-27:
If no frame was rendered it means we lost a full frame time. This is why I count the max frame time (time_limit).

nicusorcitu, 2023-04-27:
I mean in standby, when nothing is to be drawn what shall we see?
30 FPS (33 ms) 
or
30 FPS (0 ms)
?

kisvegabor, 2023-04-27:
In idle, I'd say `0 ms`. 

It's really a problem that when the rendering takes longer than 300 ms, the current algorithm thinks it was lightning fast and shows max FPS. I think we should detect this somehow, but it's quite extreme to have < 3 FPS, so I wouldn't focus too much on it.

nicusorcitu, 2023-05-02:
Please see this new patch, with "0 ms" in standby. If you like it, please merge it. If I have to create PR I need to go through the release procedure and is not desired for a single patch.
Thanks.
[0001-fix-perf-Fix-the-FPS-calculation-by-measuring-the-ex.patch](https://github.com/lvgl/lvgl/files/11370239/0001-fix-perf-Fix-the-FPS-calculation-by-measuring-the-ex.patch)

kisvegabor, 2023-05-02:
It's in [lv_sysmon.c](https://github.com/lvgl/lvgl/blob/master/src/others/sysmon/lv_sysmon.c) by @FASTSHIFT. Please try it out.

kisvegabor, 2023-05-18:
The discussion continues in: https://github.com/lvgl/lvgl/issues/4221

closed 2023-05-18 by kisvegabor