lvgl/lvgl #4168: hoHow to use a non zero address in LV_MEM_ADR ?

PeakCrow, 2023-04-26:
### Introduce the problem

> #define LV_MEM_CUSTOM 0
#if LV_MEM_CUSTOM == 0
      /*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
      #define LV_MEM_SIZE (80U * 1024U)          /*[bytes]*/
      /*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
      #define LV_MEM_ADR  0    /*0: unused*/
      /*Instead of an address give a memory allocator that will be called to get a memory pool for LVGL. E.g. my_malloc*/
      #if LV_MEM_ADR == 0
          #undef LV_MEM_POOL_INCLUDE 
          #undef LV_MEM_POOL_ALLOC   
      #endif
#endif

### Examples and cases
i use keil5 and .sct file below:
Now I have an external SRAM section below,But how can I apply it to LV_ MEM_ What is the location here for ADR?

>   EXTERN_SRAM 0x68000000 0x68100000{
    *(.EXTERN_SRAM)
  }

Placing  0x68000000  directly is not working in LV_MEM_ADR 、、、
First of all, thank you for your reply

kisvegabor, 2023-04-26:
> Placing 0x68000000 directly is not working in LV_MEM_ADR

It should work this way. What do you mean by "no working"?
- doesn't compile?
- crashes immediately?
- crashes after some time?
- something else?

PeakCrow, 2023-04-27:
[recv_2023-04-27_21-43-08.txt](https://github.com/lvgl/lvgl/files/11344287/recv_2023-04-27_21-43-08.txt)
LV_MEM_ADR Configured to 0 to fill the screen,but configured 0x68000000,
- can build and 0 errors.
- but crashes after some time.
- screen All white, no filling.

The external SRAM has been successfully initialized and is functioning normally.
I can use the following method to run normally.
> #define MY_DISP_HOR_RES 800
#define MY_DISP_VER_RES 480
__attribute__((section(".EXTERN_SRAM")))   static lv_color_t buf_1[MY_DISP_HOR_RES * MY_DISP_VER_RES];  /*A buffer for 10 rows*/  

The log file has been uploaded.
There's plenty of time, please help me check. Thank you very much

kisvegabor, 2023-04-28:
Just to confirm, the same UI works well with `LV_MEM_ADR 0`, right?

PeakCrow, 2023-04-28:
Yes, it works very well.

kisvegabor, 2023-05-01:
If it's working to some extent but fails after some time, I can think of only a memory configuration issue. :slightly_frowning_face:

PeakCrow, 2023-05-04:
Okay, I'll check the memory configuration again. Thank you!

closed 2023-05-07 by PeakCrow