lvgl/lvgl #4169: Arm-2D applies chromakeying to LV_IMG_CF_TRUE_COLOR images

kjbracey, 2023-04-27:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The Arm-2D accelerator applies chromakeying even when not requested. Observed in 8.3.4, but issue still present by code inspection.

This section of code does not distinguish the values of `cf`.

https://github.com/lvgl/lvgl/blob/b603fb90f13fb473fdcf4369d2629bf1054a0b82/src/draw/arm2d/lv_gpu_arm2d.c#L1078-L1096

### To Reproduce
```c
    imageObj = lv_img_create(lv_scr_act());
    imageDesc.data = (const uint8_t *)lvgl_image;
    imageDesc.data_size = sizeof lvgl_image;
    imageDesc.header.always_zero = 0;
    imageDesc.header.reserved = 0;
    imageDesc.header.cf = LV_IMG_CF_TRUE_COLOR;
    imageDesc.header.w = 224;
    imageDesc.header.h = 224;
    lv_img_set_src(imageObj, &imageDesc);
```
With the image (test bars in my case) containing some pure green, with default config.

### Expected behavior
No pixels should be transparent (unless indicated by alpha).

kisvegabor, 2023-04-27:
@GorgonMeducer, could you take a look at it?

GorgonMeducer, 2023-04-27:
OK, I will fix it. 

@kjbracey Just one question, are you using Cortex-M55 or Cortex-M85 processors?

GorgonMeducer, 2023-04-27:
The issue is fixed in cmsis-pack v8.3.7-dev

kjbracey, 2023-04-28:
> @kjbracey Just one question, are you using Cortex-M55 or Cortex-M85 processors?

M55.

Thanks for the quick fix!

kisvegabor, 2023-04-28:
I've merged the PR.

kjbracey, 2023-04-28:
As I asked on #4179, this looks like it's just disabling the acceleration for non-colorkey images - or does it end up accelerated via another path?

GorgonMeducer, 2023-04-28:
@kjbracey Yes. It will use other basic acceleration.

GorgonMeducer, 2023-05-03:
@kjbracey dedicated acceleration for transform-only-with-opacity is added: https://github.com/lvgl/lvgl/pull/4203
And as the code suggested, you won't get the acceleration until the release of arm-2d v1.1.3.

GorgonMeducer, 2023-05-12:
@kjbracey v1.1.3 is released.

GorgonMeducer, 2023-05-22:
@kjbracey Any update? Or are you happy with this fix?

kjbracey, 2023-05-29:
Okay, tested 8.3.7 + 1.1.3 now, and it's fixed the display error in our app. But I've not profiled it to check performance, and I'm only testing that chromakey isn't applied - I have nothing attempting to use it. Thanks!

GorgonMeducer, 2023-05-30:
@kjbracey Thank you for the confirmation. Don't worry about the performance part. I have done that locally. It is identical to the previous.

closed 2023-06-12 by kjbracey