lvgl/lvgl #4176: MicroPython CI fails

kisvegabor, 2023-04-27:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
After 02a20152d399a16f91c9539c5f3630affbda174e the MicroPython build fails. I've fixed the problem in 144182723b09d4c63a2d98d8888adb619fae0f46 and updated `lvgl` in `lv_bindings` and `lv_micropython` but it still fails.  

It works locally, however the last 2 windows of tests are no closed automatically. 
[image]

### To Reproduce
See https://github.com/lvgl/lvgl/actions/runs/4822889406/jobs/8590683282

### Expected behavior
Pass CI.

@amirgon @embeddedt do you have any idea?

amirgon, 2023-04-28:
@kisvegabor How was updating LVGL submodule supposed to solve this issue?
If you know that latest LVGL fails with the micropython bindings because of some change in LVGL, updating the submodule would just break lv_micropython_binding and lv_micropython.  

So as a first step, before looking into the problem, I'm going to revert your updates to make lv_micropython_binding and lv_micropython pass again. The whole idea of keeping LVGL as submodule was to keep the binding stable even when latest LVGL breaks. Updating a broken LVGL submodule makes no sense.

kisvegabor, 2023-04-29:
Ah, you are right. Sorry for messing it up...

So this part of the log is misleading as the tests are running on multi-thread and not the last item is causing the issue. 
[image]

`lv_example_ime_pinyin_1.c` is working well, but I'll look into it on Monday.

kisvegabor, 2023-05-01:
I've fixed that issue and not the test are pass! Thank you!

closed 2023-05-01 by kisvegabor