lvgl/lvgl #4180: Failed to compile lvgl v8.3.6 examples in Arduino IDE

Damilare1, 2023-04-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Compiling returns this error: `Compilation error: cannot convert 'lv_disp_t* {aka _lv_disp_t*}' to 'lv_disp_drv_t* {aka _lv_disp_drv_t*}' for argument '1' to 'void lv_disp_flush_ready(lv_disp_drv_t*)'`

### To Reproduce

### Expected behavior
Returns this error:
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino: In function 'void my_disp_flush(lv_disp_t*, const lv_area_t*, lv_color_t*)':
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:41:31: error: cannot convert 'lv_disp_t* {aka _lv_disp_t*}' to 'lv_disp_drv_t* {aka _lv_disp_drv_t*}' for argument '1' to 'void lv_disp_flush_ready(lv_disp_drv_t*)'
     lv_disp_flush_ready( disp );
                               ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino: In function 'void my_touchpad_read(lv_indev_t*, lv_indev_data_t*)':
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:49:24: error: 'class TFT_eSPI' has no member named 'getTouch'
     bool touched = tft.getTouch( &touchX, &touchY, 600 );
                        ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino: In function 'void setup()':
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:94:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch( calData );
         ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:100:33: error: cannot convert 'lv_disp_t* {aka _lv_disp_t*}' to 'lv_disp_drv_t* {aka _lv_disp_drv_t*}' for argument '1' to 'void lv_disp_drv_init(lv_disp_drv_t*)'
     lv_disp_drv_init( &disp_drv );
                                 ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:102:14: error: 'lv_disp_t {aka struct _lv_disp_t}' has no member named 'hor_res'
     disp_drv.hor_res = screenWidth;
              ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:103:14: error: 'lv_disp_t {aka struct _lv_disp_t}' has no member named 'ver_res'
     disp_drv.ver_res = screenHeight;
              ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:104:14: error: 'lv_disp_t {aka struct _lv_disp_t}' has no member named 'flush_cb'
     disp_drv.flush_cb = my_disp_flush;
              ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:105:14: error: 'lv_disp_t {aka struct _lv_disp_t}' has no member named 'draw_buf'
     disp_drv.draw_buf = &draw_buf;
              ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:106:37: error: cannot convert 'lv_disp_t* {aka _lv_disp_t*}' to 'lv_disp_drv_t* {aka _lv_disp_drv_t*}' for argument '1' to 'lv_disp_t* lv_disp_drv_register(lv_disp_drv_t*)'
     lv_disp_drv_register( &disp_drv );
                                     ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:110:35: error: cannot convert 'lv_indev_t* {aka _lv_indev_t*}' to '_lv_indev_drv_t*' for argument '1' to 'void lv_indev_drv_init(_lv_indev_drv_t*)'
     lv_indev_drv_init( &indev_drv );
                                   ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:111:15: error: 'lv_indev_t {aka struct _lv_indev_t}' has no member named 'type'
     indev_drv.type = LV_INDEV_TYPE_POINTER;
               ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:112:15: error: 'lv_indev_t {aka struct _lv_indev_t}' has no member named 'read_cb'
     indev_drv.read_cb = my_touchpad_read;
               ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:113:39: error: cannot convert 'lv_indev_t* {aka _lv_indev_t*}' to '_lv_indev_drv_t*' for argument '1' to 'lv_indev_t* lv_indev_drv_register(_lv_indev_drv_t*)'
     lv_indev_drv_register( &indev_drv );
                                       ^
/private/var/folders/6d/s_wqn21s61sb_rjhytc_937m0000gn/T/.arduinoIDE-unsaved2023328-61949-5k32g6.i8u1q/LVGL_Arduino/LVGL_Arduino.ino:127:21: error: 'lv_demo_widgets' was not declared in this scope
     lv_demo_widgets();            // OK
                     ^

exit status 1

Compilation error: cannot convert 'lv_disp_t* {aka _lv_disp_t*}' to 'lv_disp_drv_t* {aka _lv_disp_drv_t*}' for argument '1' to 'void lv_disp_flush_ready(lv_disp_drv_t*)'

### Screenshots or video

C47D, 2023-05-04:
Sounds like this is related to https://github.com/lvgl/lvgl/issues/4011

kisvegabor, 2023-05-05:
It seems you are using LVGL `master`. Please install LVGL v8.3.6 as an Arduino library.

Damilare1, 2023-05-13:
Thanks, that works.

closed 2023-05-13 by Damilare1

skander36, 2023-06-11:
I still get this error : " exit status 1
cannot convert 'lv_disp_t*' {aka '_lv_disp_t*'} to 'lv_disp_drv_t*' {aka '_lv_disp_drv_t*'} " on compiling LVGL_Arduino.ino.
Using ver. 8.3.6. Upgrading to 8.3.7 same result.

kisvegabor, 2023-06-12:
I'd really like to solve this issue but I feel like this way we are proceeding very slowly. Can we organize a quick call and via screen share I could take a deeper look at it?

skander36, 2023-06-12:
Going down to 8.3.3 and it works now. When will finish with current project I will try a fresh install.

kisvegabor, 2023-06-13:
I believe I fixed those issues [here](https://github.com/lvgl/lvgl/blob/release/v8.3/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino). It's still not released but you an try out if this INO files works well.

skander36, 2023-06-13:
You used by Bodmer? 

On my config with TFT_eSPI 2.5 I get:
_exit status 1
'class TFT_eSPI' has no member named 'getTouch'; did you mean 'getColor'?_

ATaimeh, 2023-06-13:
Thanks for all, I've tried the code now and it compiled.

But when i needed to upload an example i got an error saying that:

- Compilation error: 'lv_demo_widgets' was not declared in this scope

noting that i did all the steps included in the link attached (commented) in the sketch.

yusufbender, 2023-06-13:
Compilation error: 'class TFT_eSPI' has no member named 'getTouch'; did you mean 'getColor'?

#define TOUCH_CS PIN_D2 
its open

kisvegabor, 2023-06-13:
> Compilation error: 'lv_demo_widgets' was not declared in this scope

You need to enable it in `lv_conf.h` with `LV_USE_DEMO_WIDGETS     1`.