lvgl/lvgl #4184: I NEED HELP ASAP!! (please read the entire thing)

kdschlosser, 2023-04-29:
OK so I have taken on  straightening out the documentation in LVGL as some of you already know. I have fixed the build system (mostly) and I have converted the majority of the markdown (.md) files  to reST (.rst, RestructuredText) files. reST is what Sphinx uses natively and using it offer a whole lot more functionality.

OK so enough of the background of what I am doing and on to what I need help with.

It is a fantastic thing that LVGL is continually being developed. It adds all kinds of new features and sometimes old features get changed and enhanced.  My issue is the latter of the 2. I don't think there is any developer that would say "I like documenting the things I wrote" the word "developer" and that phrase would never be spoken in the same sentence I can almost gurantee that.

The point is this.. documenting sucks plain and simple. But it is one of those things that needs to be done if we want others to understand how to use the software that has been written. Getting a developer to document things that have been added is hard enough, getting them to maintain that documentation when changes are made..... hell could freeze over first before it happens voluntarily. 

After the  75 or so hours I have invested into the work I have done so far and spending much of those hours staring at the documentation files I noticed that there are a lot of references to components in LVGL that no longer exist or have been renamed. These things need to be corrected. It is even more important now then it was because because the documentation build process actually uses those references to create links to the API section of the documentation. 

I am new to helping out with the LVGL project. I did not write any of the code and I have not been using LVGL long enough to have experience using more than a single version of LVGL. I know some of 9.0 and that is all. I need people that have been around the block to know what has or has not been changed and I need those people to help me with correcting the documentation.

This is kind of the way I look at it. It is something that no one wants  to do. If they could they would run away like it's the black plague. I know how it is and I wish the fabled noofty goofty bird would come and take care of it for us. That's not going to happen and we all know it. It is not something that one person did (or rather didn't do) It was a group effort that caused this problem. It really needs to be a group effort to correct it. All development on LVGL should stop and everyone as a team pulls together and combs over the documentation and fixes/updates it so it is 100% correct with the current source code. If everyone gets together and does it it would  take a fraction of the time that it would take me alone to do it. I could invest easily 100 hours of my time updating it and I would probably not finish it in that time. It's a large amount of reading and checking each and every single reference for its existence in the source code. I can tell you there are a lot of references, 2,123 of them spread over about 115 files. That averages out to about 18 per file.

The good news is they are easily located. but we are going to have to make sure that each one exists, change them if they need to be changed and update the information surrounding the reference if it's function has changed.

as a token for putting in the work I will send each person that actually spends time doing this the money to buy a 12 pack of beer. If you are a non drinker then spend the money on whatever else it is you want o spend it on. One catch tho. the first beer has to be downed during a zoom meeting. call it an LVGL social event if you will. I know people are spread out all over the world but with the technology available today there is no reason why everyone cannot get together for an hour and just BS about anything they want to while having a beverage.

as a group effort it will be done in no time. need to set up a project on GitHub where a list of files can be posted with a checkbox marking it complete and a way to mark it as being worked on. the person that has done the work will have they name attached to that file. That's the "under the bus" part. So if it's not done right the person that did it gets thrown under the bus. For those who have never heard that expression it's when someone tells of another's wrong doing.

I have too much going on with my house (down to the stud remodel of the entire house) to on take this animal by myself. If it is not fixed now it is going to get worse as time passes. Right now it is still manageable and can be corrected fairly quickly. This kind of thing is not unusual to have happen and it happens even more so with an opensource project because after all what is going to happen if you don't update the documentation? You gonna get fired? LOL.. so the best we can do is get it done as a group and try and celebrate a little bit once it is done. again I'll buy the booze for those that participate.

kisvegabor, 2023-04-29:
Hi,

First of all, thank you so much for this great work on the docs! The fact that now we can spot all these issues shows that we are going into the right direction.  

I really like the social part of your writing! :+1:  
It'd be really great to gather and do it live, however I think it's a little bit complicated to organize it due to professional work, family, and time difference. But I hope I'll be wrong. I can write on all of our marketing channels to invite people! 

Could you share the list of issues? I'm really curious about it!

kdschlosser, 2023-04-29:
as an example.

in places the dpcumentation links to other parts of the documentation. it looks like this

```
Have you already started using LVGL in a
`Simulator </get-started/platforms/pc-simulator>`__, a development
board, or on your custom hardware? Was it easy or were there some
obstacles? Are you happy with the result? Showing your project to others
is a win-win situation because it increases your and LVGL's reputation
at the same time.
```

specifically this bit
```
`Simulator </get-started/platforms/pc-simulator>`__
```

There is a built in mechanism to be able to handle this without having to  reload the entire document webpage or opening up a new tab and loading the page all over again. 

if I open up the pc-simulator.rst file I am able to add above the header `.. _simulator:`. it would look like this

```
.. _simulator:

===============
Simulator on PC
===============
```

and in the rst file where the link is I would change that to 

```
Have you already started using LVGL in a
:ref:`simulator`, a development
board, or on your custom hardware? Was it easy or were there some
obstacles? Are you happy with the result? Showing your project to others
is a win-win situation because it increases your and LVGL's reputation
at the same time.
```

The output once compiled end up like this.
```
Have you already started using LVGL in a
Simulator on PC, a development
board, or on your custom hardware? Was it easy or were there some
obstacles? Are you happy with the result? Showing your project to others
is a win-win situation because it increases your and LVGL's reputation
at the same time.
```

where `Simulator on PC` is a clickable link and that link loads the page in the frame instead of reloading the entire page

There are references to LV_IMG_CF_* enumerations which have been removed. There are references to lv_disp_drv_t which I do not believe is available anymore.

Actually there are functions defined in the source code that use lv_disp_drv_t in several source files
```
void lv_draw_vglite_ctx_init(struct _lv_disp_drv_t * drv, lv_draw_ctx_t * draw_ctx);
void lv_draw_vglite_ctx_deinit(struct _lv_disp_drv_t * drv, lv_draw_ctx_t * draw_ctx);
void lv_draw_pxp_ctx_deinit(lv_disp_drv_t * drv, lv_draw_ctx_t * draw_ctx)
void lv_draw_pxp_ctx_deinit(lv_disp_drv_t * drv, lv_draw_ctx_t * draw_ctx)
void lv_draw_pxp_ctx_init(struct _lv_disp_drv_t * drv, lv_draw_ctx_t * draw_ctx);
void lv_draw_pxp_ctx_deinit(struct _lv_disp_drv_t * drv, lv_draw_ctx_t * draw_ctx);
void lv_draw_vglite_ctx_deinit(lv_disp_drv_t * drv, lv_draw_ctx_t * draw_ctx)
void lv_draw_vglite_ctx_deinit(lv_disp_drv_t * drv, lv_draw_ctx_t * draw_ctx)
```

so this is not only going to ensure consistency with the documentation it is also going to correct code issues as well, example is above.

There are functions in LVGL that have wrong parameter names set in the doxygen comments. There is quite a few of them actually.

Call it spring cleaning. from time to time this does need to be done to check the validity of the documentation and also the source code.

Once we get it corrected we can put in place a Python script that will check that documentation has been added for anything new. and we can also check the documentation to see if there are things in there that do not exist in the code. we can check that all parameters for a function are documented and all fields for structures, unions and enumeration items are documented as well. we can also verify the parameter names match what has been documented.

About the social event. Yes there are people all over the world. But I am sure that it can be done on say a Saturday which may be a Sunday for others. where it would not interfere with a work schedule (hopefully). It would also be planned so it would be a few weeks out. I don't expect that many people will be wanting to jump at doing this. ya never know and I have my fingers crossed. I am hoping I can lure a few in with the promise of a refreshment of their choice. I don't drink alcohol any more so I don't even know how much it costs. I would have to check and see.

This is an important thing to do and it is what will keep new people coming and it is also a draw for companies as well. It shows the project is on the ball and everything is how it should be.

Some people may or may not be able to do the Social event thing. They would still get the money for the beverages or whatever they want to spend the money on.

kisvegabor, 2023-05-01:
I see, and I'm absolutely in!  But I suggest waiting a little bit and doing it when all the major changes of v9 are added. There will be 3 major things:
- parallel rendering
- new image decoding
- some widget reworks and general refactoring

I'm afraid if we need it know we needed to do it again in ~3 months.

kdschlosser, 2023-05-01:
good idea on doing it after the next major version release. this way it will not have to get done over again. Maybe this should be one of the things you add to the checklist of things to do for a major version release. getting the documentation squared off with the source code.

kisvegabor, 2023-05-01:
> Maybe this should be one of the things you add to the checklist of things to do for a major version release. getting the documentation squared off with the source code.

Ideally it should be tested in the CI similarly to code formatting errors. Is it possible? I mean figuring out if the docs is consistent with the code, and if not make the CI fail.

kdschlosser, 2023-05-01:
I can do that to some extent. I am able to check to see if things are not documented, or parameters are missing in the docs or they have incorrect names. I can even do a rudimentary spell check as well. Grammar would be a but more difficult to do but not impossible.

I do have an issue open for breathe. breathe is the layer between the doxygen output and sphinx it is what reads the output and converts it into something sphinx is able to use. One of the things I have found is that sphinx directive do not get parsed. So if I wanted to create a link from one section of the C API to another section of the C API I am not able to by doing it in the doxygen docstrings.

While I am sure I could fix it and monkey path the thing so it would do what it should do I really cannot be bothered. if it never gets merged then you would have a piece of non standard library that was being used and that would make the build system fragile.

kdschlosser, 2023-05-01:
what I do need to know is what in the heck makes the friggin changelog. I need to change whatever it is that writes the thing so it will output correct sphinx reST syntax. I have fixed that file like 4 times now and it's starting to make me bonkers. 

I need to know what else gets automatically generated.

kisvegabor, 2023-05-02:
The change log is generated by this script: https://github.com/lvgl/lvgl/blob/master/scripts/changelog_gen.sh

I think it'd be better to have multiple CHANGELOG  files for each release. What do you think?

kdschlosser, 2023-05-02:
I agree. and only put change logs that are specific to a version with that version. No need to have change logs for 8.0 and 7.0 in 9.0.

I am not able to find anything that runs that script in any files in LVGL. what runs it?

kdschlosser, 2023-05-02:
is there anything else that gets run that has a dependency on npm?

I ask this because I believe that `auto-changelog` is specific to the creation of markdown files. I would have to double check on that. I am pretty sure that there is a Sphinx plugin that has the ability to generate the change log without the need to run a shell script (which is not the easiest thing to do on a Windows system). It would also remove the need to install npm if generating the changelog is the only thing that uses npm. The change log would get made when the documentation gets compiled.

kisvegabor, 2023-05-03:
> I am not able to find anything that runs that script in any files in LVGL. what runs it?

@kisvegabor runs them before releasing :slightly_smiling_face: 

> I ask this because I believe that auto-changelog is specific to the creation of markdown files. 

The format is coming from this [template file](https://github.com/lvgl/lvgl/blob/master/scripts/changelog-template.hbs), so we can support rst too.

>  The change log would get made when the documentation gets compiled.

I think it's not a good idea as something I need to edit the change log a little bit. Fix typos, remove and regroup commits if needed, add some overview, etc.  So I'd keep the manual changelog addition.

kdschlosser, 2023-05-03:
ahh ok. so you manually run it and edit the file before you build the documentation.

kisvegabor, 2023-05-04:
Yes, that's it.

kdschlosser, 2023-05-04:
OK that makes sense. Then I did edit the right file. I hope I saved the edited version of the template that would output the correct format. I am not familiar with `auto-changelog` and how to go about setting it up so that only adds changes for a specific version. I will have to do some reading on that.

kisvegabor, 2023-05-05:
E.g. `./changelog_gen.sh v8.3.5` lists commits from the v8.3.5 tag to the last commit.

kisvegabor, 2023-05-22:
Let's get back to it when we get closer to release v9. I've pinned the issue until that.

kdschlosser, 2023-05-22:
sounds good to me. Hopefully by then I will have the PDF bits sorted out.

kisvegabor, 2023-12-02:
Now we add the last features to v9 and start working on the docs update too.

I'll update a dedicated issue for that tomorrow.