lvgl/lvgl #4186: SOS!!

QiaoTuCodes, 2023-04-29:
May I ask why when using the demos of lv_port_linux_frame_buffer, changing disp_drv.hor_res has no effect? Whenever it is changed to 1920 x 1080, the screen becomes scrambled and chaotic, but when changed to 800 x 480, the display is good? i'll want to set hor_res to 1920 or more.

QiaoTuCodes, 2023-04-29:
It seems that there is an invisible framework that strictly limits the display area to 800x480. Changing hor_res has no effect at all, and images with a resolution that does not exceed 800x480 can be displayed normally. Once the setting exceeds this limit, the display becomes chaotic. why?

kdschlosser, 2023-04-29:
what version of LVGL are you using?

QiaoTuCodes, 2023-04-29:
> what version of LVGL are you using?

it's LVGL8.2 , here: https://github.com/lvgl/lv_port_linux_frame_buffer . i'll got the codes on master.

kdschlosser, 2023-04-30:
have you tried 9.0 to see if the same issue exists?

QiaoTuCodes, 2023-04-30:
> have you tried 9.0 to see if the same issue exists?

so from where to download lvgl9 projects? i'll just found top version it's Release v8.3.6 . this all right?

kdschlosser, 2023-05-01:
the master branch of this repository. click on the button that says "CODE" and then click on "download zip" at the bottom of the menu that opens up.

QiaoTuCodes, 2023-05-01:
oh. master it's v9.0. got it. thank u

QiaoTuCodes, 2023-05-01:
> the master branch of this repository. click on the button that says "CODE" and then click on "download zip" at the bottom of the menu that opens up.

but the lv_port_linux_frame_buffer master it's v8.2 , i use : git clone -b master --recursive https://github.com/lvgl/lv_port_linux_frame_buffer.git just download v8.2 , v9.0 not found.
[image]

QiaoTuCodes, 2023-05-01:
you mean just delete lv_conf.h lv_drv_conf.h and delete lvgl 、lv_drivers directory.  

use:  git clone -b master --recursive https://github.com/lvgl/lvgl.git 
use:  git clone -b master --recursive https://github.com/lvgl/lv_drivers.git

recopy lv_conf.h lv_drv_conf.h keep them in v9.0.0 right?

kdschlosser, 2023-05-01:
sorry I wasn't thinking. I didn't realize the 8.3 version bump for the frame buffer driver.

kdschlosser, 2023-05-01:
is there a specific reason as to why you are rendering to a Linux frame buffer?. I ask this because if it is for the purposes of opening up in a window as a display then you can use 9.0 for that and you do not need the frame buffer driver. you need to turn on SDL in the config file and call `lv_sdl_window_create` and you can also set up mouse and keyboard input as well.

QiaoTuCodes, 2023-05-01:
> is there a specific reason as to why you are rendering to a Linux frame buffer?. I ask this because if it is for the purposes of opening up in a window as a display then you can use 9.0 for that and [...]

i think if just useing Linux frame buffer it's can close x-window , the app will very light weight. in practical application i hope split screen display, one screen will set-default multi-user mode to shows dynamic with lvgl UI and the other screen shows x-window and python web applications

QiaoTuCodes, 2023-05-01:
> sorry I wasn't thinking. I didn't realize the 8.3 version bump for the frame buffer driver.

I've figured out that the problem has nothing to do with the fb driver and that you should first set the maximum resolution of the frame buffer using fbset. Run the demo again, you can display the full screen, the default is 1024x600 at the beginning, so no matter how to change lv_conf.h is invalid.

so thank you for help me.

kdschlosser, 2023-05-01:
I don't think I helped at all. Maybe it made you thing about the actual frame buffer in Linux as what was causing the issue.

Glad you solved it. I would close this issue if you are all set

QiaoTuCodes, 2023-05-01:
> 我认为我根本没有帮助。也许它让您想到 Linux 中的实际帧缓冲区是导致问题的原因。
> 
> 很高兴你解决了它。如果你都准备好了，我会关闭这个问题

close

> I don't think I helped at all. Maybe it made you thing about the actual frame buffer in Linux as what was causing the issue.
> 
> Glad you solved it. I would close this issue if you are all set

yah close that. thank you help me.

closed 2023-05-03 by QiaoTuCodes