lvgl/lvgl #4187: Need to place  widgets beside others, such as one is (x,y, w, h), the other is (x, y+h, w, h)

CristXu, 2023-04-29:
### Introduce the problem

### Examples and cases

### Suggested solution

kisvegabor, 2023-05-01:
Do you need to position transformed widgets? If not you don't need to use the `lv_obj_get_transformed_area` for it.

Can [flex layout](https://docs.lvgl.io/master/layouts/flex.html) work for you?

CristXu, 2023-05-04:
Hi, 
    Thank you very much for your quick reply. 
    No, I do not need a position transformed widgets. But after check the code, seems that the lv_obj_get_transformed_area function will always be called when I set a widget visiable:
[image]
    Then inside the lv_obj_get_transformed_area , the area will be enlarge by (5, 5) with lv_area_increase(area, 5, 5); :
[image]
    So why the code always increase the area? and how can I avoid this area-increase?
    And Yes, I have try the flex layout, but seems that can not work when I place two widget alongside others, for one of the area will overlap the others by 5px(lv_area_increase(xx, 5, 5));

Regards,
    Crist

kisvegabor, 2023-05-04:
It's increased just to be sure the are is large enough even after some rounding errors. However,  `lv_obj_get_transformed_area` is not used the position the widgets. It's only used to calculate the figure out which area needs to be redrawn.

CristXu, 2023-05-04:
Thank you. and Have another question, the redrawn area is larger than the widget inside. So if I put two widget as below:
[image]
We have red and black rects, but the redrawn area is the blue one, larger than each widget by 5px. So when redrawn the red one, the redrawn area will overlap the black? Is it correct? 
Thank you again for your help.

kisvegabor, 2023-05-05:
> So when redrawn the red one, the redrawn area will overlap the black? Is it correct?

That's correct and it's really not ideal. This 5px extra size should be removed in v9 .

closed 2023-05-27 by lvgl-bot