lvgl/lvgl #4197: Per object antialiasing control

megaboy2k, 2023-05-01:
It is appeared that for some font sizes antialiasing can improve an appearance and get it worse in other cases.
for example, for tiny font sizes it get it worse and it would be very helpful to disable antialiasing only for such objects
keeping it enabled for the rest. 
This also applies to some mini icons such as GPS/GSM signals and technology name (2D, 3D, E, H, 3G, etc).
It would be highly desirable to add individual antialiasing control, say, in style properties.

kisvegabor, 2023-05-02:
You can convert the fonts with different bpp (bit per pixel). For small fonts you can use higher bpp, for the larger ones small bpp should be enough too.

In the [font converter](https://lvgl.io/tools/fontconverter):
[image]

megaboy2k, 2023-05-02:
@kisvegabor Correct, if I set 1 bpp for a small font the anti aliasing gets effectively disabled. However I can do nothing in case of other object types. E.g. I use lines to cross out a small icon (no signal) and it would appear better if anti aliasing would not apply to these lines.

kisvegabor, 2023-05-04:
I see. We are changing a lot of things for v9, and I'll keep it in mind. We sill see we can add it with a reasonable way.

closed 2023-05-27 by lvgl-bot