lvgl/lvgl #4198: Can you tell me how to connect a mouse as an input device to an RP2040 Arduino?

SocrateCrypto, 2023-05-01:
I am using the following libraries:
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages =
framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#3.1.1
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
lib_archive = yes
lib_deps =
sekigon-gonnoc/Pico PIO USB@
adafruit/Adafruit TinyUSB Library@1.17.0
bodmer/TFT_eSPI@^2.5.23
lvgl/lvgl@^8.3.4

board_build.f_cpu = 120000000L
build_flags = -fstack-protector"

The problem is that when I connect LVGL, the mouse stops sending coordinates. Does anyone know a working solution?"

kisvegabor, 2023-05-02:
I see that you have opened [this forum topic](https://forum.lvgl.io/t/can-you-tell-me-how-to-connect-a-mouse-as-an-input-device-to-an-rp2040-arduino/11647) too so I close this issue.

closed 2023-05-02 by kisvegabor

SocrateCrypto, 2023-05-02:
> I see that you have opened [this forum topic](https://forum.lvgl.io/t/can-you-tell-me-how-to-connect-a-mouse-as-an-input-device-to-an-rp2040-arduino/11647) too so I close this issue.

ок