lvgl/lvgl #4201: Lots of new and exciting things happening with LVGL

kdschlosser, 2023-05-02:
There is a lot of new development taking place in LVGL. Version 9 is going to have a lot of new changes and updates to it. There have been a lot of conversations happening for all the new development.  I have run into a bump with something that I am working on and it falls right into the conversations that have taken place about API and naming convention standards. The problem I am having is due to the lack of standards. While there is nothing wrong with the code the differences in how things are done and naming conventions can cause headaches and or problems. This specifically holds true if trying to write code that automates the process of doing something.  @amirgon sir I tip my hat to you for the MicroPython binding and managing to code around the snafu's caused by naming conventions and code styles. I would very much like to hear your input and experience with dealing with the differences in the MicroPython binding.

 As some of you already know I am working on making a CPython Binding to LVGL. I have come across some challenges because of naming conventions and how things are being declared in the C code. I started to code around the differences but then I started thinking about what a maintenance nightmare it would become and how fragile the build process would be. It could easily get broken and it would make the build system pretty chunky in size and also increase the complexity to the point of making it hard to understand when the code is read. I now realize why the program that creates the MicroPython binding is as complex as it is.

We have had the addition of a Ruby binding. There is a developer working on a C++ binding and I am working on a CPython binding. This is going to really expand where LVGL is going to get used. The more bindings there are the better. it allows people to write code in a language they are most comfortable using and makes development easier for them. We need to make it easier to create these bindings. API standards and naming standards are the way to achieve that..

I would like for @kisvegabor to create a repo where API specs/rules can be collected so they can be compiled into documentation that is easily read. I would be more than happy to handle the maintenance and upkeep of this repository if no one else wants to handle it. The maintainer (s) of this repository are not in change of coming up with the riles/specs. They are only there to add, remove and update. They would be responsible for keeping the information in a manner that it is easy to read and has a standard for how the information is organized.

Having this not only makes it easier for attachments/bindings to be made it also makes it easier to read the source code in LVGL.

Automatic enforcement of the specs/rules can be done. I wrote the mechanism that is able to do it and it is included in the most recent changes in the documentation build system. This would keep everyone honest, The system would be able to collect information that is located in the API repo so it knows what it should be looking for when running the checks.

I would like to know what peoples thoughts are on this.

kdschlosser, 2023-05-02:
Currently there are callbacks that are declared that do not take a structure as the first parameter. some of these do have a parameter for passing user_data and some do not. Some of those functions have the parameter named as "user_data" while others do not. There is a mixture of callbacks declared using typedef and being declared in a structure or parameter list of a function. Due to limitations when making a binding callback functions need to be declared using typedef. So far I have found 3 naming schemes used for callbacks they have a suffix of one of the following:

* "_xcb_t": callback that does not take a structure as the first parameter
* "cb_t": random on how it is used
* "_f_t": just a different naming of the one above.

This is inconsistent and additional code has to be added to handle this.

In the MicroPython binding there is quite a bit of code just to handle all of the inconsistencies and variations of callbacks. there is checking of the first parameter type and if it is a structure then a check is done to see if there is a user_data field. it also checks if there is a named parameter of user_data.

While this sounds like no big deal it's not.

I am currently going through and reading each source file and looking for callbacks.  due to the different ways they are declared there is no easy way to do a search and each file needs to be read to locate them all.

This is what I propose that we should use as rules on how the code should be concerning callback functions. 

Callback Naming conventions and API specification.

* 1: Callback functions **MUST** be declared as a type and that type is what gets used in structures and functions. The declaration of a callback cannot occur in a field of a structure and it cannot be in the declaration of a function.
* 2: Callback function parameters **MUST** be named.
* 3: A private callback should not appear in any public function or structure.
* 4: A private callback name **MUST** be prefixed with an underscore "_"
* 5: Callback type names **MUST** have a suffix of "_cb_t".
* 6: The first parameter of a callback function **MUST** be a structure and that structure **MUST** have a field in it named "user_data" and that field **MUST** have a type of `void *`

**WRONG**
```
typedef void (*lv_anim_exec_xcb_t)(void *, int32_t);
```
```
typedef struct {
    lv_draw_ctx_t base_draw;

    /** Fill an area of the destination buffer with a color*/
    void (*blend)(lv_draw_ctx_t * draw_ctx, const lv_draw_sw_blend_dsc_t * dsc);
} lv_draw_sw_ctx_t;
```
```
typedef lv_res_t (*lv_img_decoder_info_f_t)(struct _lv_img_decoder_t * decoder, const void * src,
                                            lv_img_header_t * header);
```
```
void lv_disp_set_flush_cb(lv_disp_t * disp, void (*flush_cb)(struct _lv_disp_t * disp, const lv_area_t * area,
                                                             lv_color_t * px_map));
```

**RIGHT**
```

typedef void (*lv_group_focus_cb_t)(struct _lv_group_t *);
typedef void (*lv_group_edge_cb_t)(struct _lv_group_t *, bool);

/**
 * Groups can be used to logically hold objects so that they can be individually focused.
 * They are NOT for laying out objects on a screen (try layouts for that).
 */
typedef struct _lv_group_t {
    lv_ll_t obj_ll;        /**< Linked list to store the objects in the group*/
    struct _lv_obj_t ** obj_focus; /**< The object in focus*/

    lv_group_focus_cb_t focus_cb;              /**< A function to call when a new object is focused (optional)*/
    lv_group_edge_cb_t  edge_cb;               /**< A function to call when an edge is reached, no more focus
                                                    targets are available in this direction (to allow edge feedback
                                                    like a sound or a scroll bounce) */

    void * user_data;

    uint8_t frozen : 1;         /**< 1: can't focus to new object*/
    uint8_t editing : 1;        /**< 1: Edit mode, 0: Navigate mode*/
    uint8_t refocus_policy : 1; /**< 1: Focus prev if focused on deletion. 0: Focus next if focused on
                                   deletion.*/
    uint8_t wrap : 1;           /**< 1: Focus next/prev can wrap at end of list. 0: Focus next/prev stops at end
                                   of list.*/
} lv_group_t;

/**
 * Set a function for a group which will be called when a new object is focused
 * @param group         pointer to a group
 * @param focus_cb      the call back function or NULL if unused
 */
void lv_group_set_focus_cb(lv_group_t * group, lv_group_focus_cb_t focus_cb);

/**
 * Set a function for a group which will be called when a focus edge is reached
 * @param group         pointer to a group
 * @param edge_cb      the call back function or NULL if unused
 */
void lv_group_set_edge_cb(lv_group_t * group, lv_group_edge_cb_t edge_cb);

```

amirgon, 2023-05-02:
> I would very much like to hear your input and experience with dealing with the differences in the MicroPython binding.

I think that my goals with the Micropython Bindings and your goals are slightly different.  

While the main use of the Micropython Bindings is to provide Micropython API for LVGL, the same script is also used to provide Micropython API to other libraries.  
That is why I tried to take LVGL "as is" as much as possible with minimum changes, and generate the bindings for it. Ideally I would have wanted it to provide Micropython API for any C API, but realistically we must make some minimal changes in the C API.

I'm collaborating with @kisvegabor for long time regarding coding conventions and the minimal set of conventions that can make a usable and complete Micropython API for LVGL and these are [described in the docs](https://docs.lvgl.io/master/get-started/bindings/micropython.html#the-micropython-binding-is-auto-generated).

I believe having as few "conventions" as possible is a good thing:
- Makes it easy for the C developer and maintainers. It's hard to keep track of lots of rules like these, even if we have automations in place to enforce them (and such automations are difficult to build and maintain).
- Makes the Micropython Binding more general. It can be used with libraries other than LVGL with minimal changes.

If you look at LVGL history you can find lots of discussions about LVGL API conventions and Micropython compatibility.
Here are examples: https://github.com/lvgl/lvgl/issues/1763, https://github.com/lvgl/lvgl/issues/1036

> started thinking about what a maintenance nightmare it would become and how fragile the build process would be

The way we keep the conventions and build stability is by automated tests that are running on the LVGL repo itself.  
Some volunteers, specifically @uraich, made huge effort to translate all C examples and some of the demos to Micropython. These examples and demos are running as automated tests on every commit and PR. When some API changes these tests should catch it. That's not full-proof but even if it catches 90% of the cases then it's not bad. This also ensure we always have functional examples as a reference for Python developers.  
The goal is to keep translating new C examples to Micropython when new features/widgets are added, both for the reference and for the API testing.

> This is what I propose that we should use as rules on how the code should be concerning callback functions.

Callbacks are a special challenge especially without [libffi](https://en.wikipedia.org/wiki/Libffi) (which is not supported on many embedded platforms). In C the programmer simply creates a function pointer, but in order to implement this for dynamic languages like MicroPython (or Python) we need to pass through some context to track back the calling object. This is the purpose of `user_data`. We discussed callbacks a lot (for example in https://github.com/lvgl/lvgl/issues/1036 and other issue) and reached the callback conventions we use today and that are [described in the docs](https://docs.lvgl.io/master/get-started/bindings/micropython.html#callbacks).

As said above, I still wanted to keep some flexibility for passing `user_data` so it would match other libraries, or require relatively small changes in the C API to enable Micropython API.

kisvegabor, 2023-05-04:
I'm absolutely open to find the simplest rules and update LVGL accordingly. I think adding typedefs for the callbavks are pretty straight forward and we can easily do it. Having a list about the problematic callbacks  would be really helpful.

> The first parameter of a callback function MUST be a structure and that structure MUST have a field in it named "user_data" and that field MUST have a type of void *

Besides, the `user_data` should point to the data set when the callback was attached. 

> I would like for @kisvegabor to create a repo where API specs/rules can be collected so they can be compiled into documentation that is easily read. 

Do we really need a new repo for it? We started to embrace the idea of the monorepo to reduce versioning issues, so I wonder why can't we add it in this repo?

GorgonMeducer, 2023-05-04:
About the function pointer and typedef, I have some suggestions:

```c
typedef void (*lv_group_focus_cb_t)(struct _lv_group_t * group);
typedef void (*lv_group_edge_cb_t)(struct _lv_group_t * group, bool focus_changed);

...
typedef struct _lv_group_t {
    ...
    lv_group_focus_cb_t focus_cb;              /**< A function to call when a new object is focused (optional)*/
    lv_group_edge_cb_t  edge_cb;               /**< A function to call when an edge is reached, no more focus
                                                    targets are available in this direction (to allow edge feedback
                                                    like a sound or a scroll bounce) */

...
} lv_group_t;
```

I prefer a different style:

```c
typedef void lv_group_focus_cb_t(struct _lv_group_t * group);
typedef void lv_group_edge_cb_t(struct _lv_group_t * group, bool focus_changed);

...
typedef struct _lv_group_t {
    ...
    lv_group_focus_cb_t *focus_cb;              /**< A function to call when a new object is focused (optional)*/
    lv_group_edge_cb_t  *edge_cb;               /**< A function to call when an edge is reached, no more focus
                                                    targets are available in this direction (to allow edge feedback
                                                    like a sound or a scroll bounce) */

...
} lv_group_t;
```

As you can see, we typedef function prototypes rather than function pointers, and with this method, the function pointer really looks like an ordinary pointer. Further more, if we insist that for function pointers, you should not ignore the "*" or "&", then the whole syntax will be as simple as normal pointers.

kdschlosser, 2023-05-04:
if you wanted to have a place where any of these kinds of standards are documented another repo is a good start.  until it gets organized enough to add into the documentation that is. No sense in adding as we go if things are not yet concrete with all of the changed that are occurring in LVGL. It is also where you would document naming conventions as well. and that at the moment I do not believe has be fully decided on. so as ideas come about that is where they would go and once it is hammered out all of the information is in a single location and can be viewed all at once. from there it can be organized and sorted out.

Ever had to go searching for something that someone said a bunch of weeks or even a month ago and you know it was said but end up spending 30 minutes looking for it to only come up empty handed? I know I have had that happen a few times.

I don't remember off the top of my head what widget it is but it has come to my attention that the _xcb_t suffix on a typed callback is for "private"(internal) use. This is fine if the functions that the callback was used in started with an underscore, which has also come to my attention is a naming convention standard. Not sure where there is a special suffix and not an underscore added to the beginning of the name which would mark it as private. 

While having the `lv_*_private.h` files are a great way of declaring private functions declaring anything else in them is going to cause an issue when using an automated script to create a binding. This is because those header files are not included in any of the public header files which is what gets used to generate the code for the binding. The private header files are only included in source files. So a structure like `lv_disp_t` is not able to be accessed properly because there is no code that gets generated for it. I am not sure what is used in the MicroPython binding as far as the preprocessor output that gets read by pycparser. Is it just the preprocessed output of lvgl.h or is it the output from all of the header files and source files? I am guessing it is the latter because `lv_disp_t` is available and all fields have code to be able to access them.

Unfortunately I am dealing with limitations with the program that writes the code for the extension module which is what is causing the callback issue.

I have the start of a list of callbacks.

```
lv_disp.h
function lv_disp_set_flush_cb
function lv_disp_set_draw_ctx

lv_disp_private.h
struct _lv_disp_t
flush_cb
draw_ctx_init
draw_ctx_deinit
wait_cb

lv_indev.h
function lv_indev_set_read_cb

lv_indev_private.h
struct _lv_indev_t
read_cb
feedback_cb

lv_obj_class.h
struct _lv_obj_class_t
constructor_cb
destructor_cb
event_cb

lv_img_cache.h
struct lv_img_cache_manager_t
open_cb  - no user_data
set_size_cb  - no user_data
invalidate_src_cb  - no user_data

lv_fs.h
struct _lv_fs_drv_t
ready_cb
open_cb
close_cb
read_cb
write_cb
seek_cb
tell_cb
dir_open_cb
dir_read_cb
dir_close_cb

lv_fragments.h
struct _lv_fragment_class_t
constructor_cb  - no user_data
destructor_cb  - no user_data
attached_cb  - no user_data
detached_cb  - no user_data
create_obj_cb  - no user_data
obj_created_cb  - no user_data
obj_will_delete_cb  - no user_data
obj_deleted_cb  - no user_data
event_cb  - no user_data

lv_draw.h
struct _lv_draw_ctx_t
init_buf
draw_rect
draw_arc
draw_img_decoded
draw_img
draw_letter
draw_line
draw_polygon
draw_transform
wait_for_finish
buffer_copy
buffer_convert
buffer_clear
layer_init
layer_adjust
layer_blend
layer_destroy

lv_gpu_nxp_pxp.h
struct lv_nxp_pxp_cfg_t
pxp_interrupt_init  - no user_data
pxp_interrupt_deinit  - no user_data
pxp_run  - no user_data
pxp_wait  - no user_data

lv_draw_sw.h
struct lv_draw_sw_ctx_t
blend

gifdec.h
struct gd_GIF
plain_text  - no user_data
comment  - no user_data
application  - no user_data

```

kdschlosser, 2023-05-04:
@GorgonMeducer 

can you do that with callbacks in C99? I thought you have to declare it as a function pointer. The other thing is I opened an issue with Doxygen because of it not reading a typed function pointer properly. the response I got was `void (* some_func_name)` is not standard syntax. and that the parenthesis are redundant. IDK if that is the case or not. You C guys would know if that is the case or not.

For the CPython binding cffi makes everything a pointer anyhow because there is no way to pass any actual python data to C code, only a pointer to a memory location can be passed. so in my use cause it doesn't matter if it is a pointer to a function or not because it ends up being turned into a pointer anyhow.

GorgonMeducer, 2023-05-05:
> can you do that with callbacks in C99?

Yes, I did this in my daily work.
https://github.com/ARM-software/Arm-2D/blob/29f89ed5252b8f1f396277243ba58fd8bb554fd0/Helper/Include/arm_2d_helper_pfb.h#L363-L442

> I thought you have to declare it as a function pointer. 

That is not necessary.

kdschlosser, 2023-05-05:
I am not that proficient in C. I know enough to edit it and to make bindings from Python. small programs for an Arduino and that's about all.

GorgonMeducer, 2023-05-05:
This trick isn't that famous but useful.

kdschlosser, 2023-05-05:
so as a question what is the difference between this

```
typedef void (*lv_group_edge_cb_t)(struct _lv_group_t * group, bool focus_changed);
```

and this

```
typedef void *lv_group_edge_cb_t(struct _lv_group_t * group, bool focus_changed);
```

GorgonMeducer, 2023-05-05:
```c
typedef void (*lv_group_edge_cb_t)(struct _lv_group_t * group, bool focus_changed);
```

It is a type for function pointer, the return type is `void`

```c
typedef void *lv_group_edge_cb_t(struct _lv_group_t * group, bool focus_changed);
```

It is a type for function prototype, the return type is `void *`

kdschlosser, 2023-05-05:
@amirgon 

I was reading over #1763 and I am hitting my head against the wall for most of the same things you mentioned in that discussion. Gotta be Déjà vu seeing this as it is pretty much a duplicate of what was discussed 2.5 years ago.

While you made the gen_myp script so it could be used on other c libraries and not just LVGL makes no real difference in this case because the binding is still being made to Python. 

There are little things like lv_list and functions like lv_obj_del that end up not playing nice nice with python. prefixing these things with an underscore (which is what is done in the gen_mpy script) does the exact same thing as if you prefix something in C code. It means that it is a private object. If using an IDE that is made for Python when the IDE sees objects that are prefixed with _ it does not show up in any code insights for an object. anything that begins with an underscore does not get imported when using this to do an import `from module import *`. The simple solution is to rename functions like `lv_obj_del` to `lv_obj_delete` which would make more sense to do instead instead of mangling the name so it will work properly in Python. 

The funny thing about Python is the ability to set an attribute like this `setattr(obj, 'del', None)` but you have the inability to access it using `obj.del` but it can be accessed using `getattr(obj, 'del'). so the binding would probably compile correctly with a function named 'del' but the only way to be able to get access to it would be using `getattr(obj, 'del')`.

This is more than likely a newbie question, I am trying to get a better understanding of C code. Can something like this code be used instead of what is currently being used in LVGL?
```

typedef struct {
    lv_obj_t obj;
    char * text;
    union {
        char * tmp_ptr; /*Pointer to the allocated memory containing the character replaced by dots*/
        char tmp[LV_LABEL_DOT_NUM + 1]; /*Directly store the characters if <=4 characters*/
    } dot;
    uint32_t dot_end;  /*The real text length, used in dot mode*/

#if LV_LABEL_LONG_TXT_HINT
    lv_draw_label_hint_t hint;
#endif

#if LV_LABEL_TEXT_SELECTION
    uint32_t sel_start;
    uint32_t sel_end;
#endif

    lv_point_t size_cache; /*Text size cache*/
    lv_point_t offset; /*Text draw position offset*/
    lv_label_long_mode_t long_mode : 3; /*Determine what to do with the long texts*/
    uint8_t static_txt : 1;             /*Flag to indicate the text is static*/
    uint8_t recolor : 1;                /*Enable in-line letter re-coloring*/
    uint8_t expand : 1;                 /*Ignore real width (used by the library with LV_LABEL_LONG_SCROLL)*/
    uint8_t dot_tmp_alloc : 1;          /*1: dot is allocated, 0: dot directly holds up to 4 chars*/
    uint8_t invalid_size_cache : 1;     /*1: Recalculate size and update cache*/
} _lv_label_t;     // <<------- Because it is an internal use structure ("private")

typedef lv_obj_t lv_label_t;

lv_label_t * lv_label_create(lv_obj_t * parent);

void lv_label_set_text(lv_label_t * obj, const char * text);

```

even better would be to change the name of the existing lv_label_t structure to lv_label_config_t

This would make it easier to organize functions when making a binding. it also makes it really clear from a user standpoint of what is expected to be passed to functions.

This would also solve naming conflicts like `dropdown` and `dropdown _list` which was mentioned in #1763 which is the reason why I ask the question.

I don't know C that well and I do not know if an `lv_label_t` typed object can be passed to a function where the parameter has a type of `lv_obj_t`. I would think it should be able to be passed, don't know if the compiler would complain at all.

kdschlosser, 2023-05-05:
@GorgonMeducer 

excuse my ignorance here. I am  confused then as to why someone over at doxygen called it "redundant parentheses" when I opened an issue because of doxygen not parsing a `void (* name)(arg_type arg_name)` statement properly. 

It may have been declared in a structure or it could have been a typedef I would have to go back and double check.

I have never seen a function prototype begin with typedef before. what would be the difference between this 

```c
typedef void *lv_group_edge_cb_t(struct _lv_group_t * group, bool focus_changed);
```

and this

```c
void *lv_group_edge_cb_t(struct _lv_group_t * group, bool focus_changed);
```

kisvegabor, 2023-05-05:
@kdschlosser 
That list of callbacks doesn't look that bad. It seems no too complicated to fix them. 

@GorgonMeducer 
As an experiment I've tried:
```c
typedef void lv_disp_flush_cb_t(struct _lv_disp_t * disp, const lv_area_t * area, lv_color_t * px_map);

//Option1
void lv_disp_set_flush_cb(lv_disp_t * disp, lv_disp_flush_cb_t * flush_cb);

//Option2
void lv_disp_set_flush_cb(lv_disp_t * disp, lv_disp_flush_cb_t flush_cb);

lv_disp_set_flush_cb(disp, flush_cb);
```

And both are working without warnings. I didn't know that prototypes can be  `typedef`-ed too.  But I wonder what is actual meaning of  not passing a function as pointer. What value will be actually passed?

XuNeo, 2023-05-05:
There is a good explanation [here](https://stackoverflow.com/questions/6893285/why-do-function-pointer-definitions-work-with-any-number-of-ampersands-or-as) about function and its pointer.

So it becomes a coding style disscusion when both work identically.

kisvegabor, 2023-05-05:
Ah, thanks! Really interesting! For consistency with normal pointers  I would also prefer 
`void lv_disp_set_flush_cb(lv_disp_t * disp, lv_disp_flush_cb_t * flush_cb);`.

GorgonMeducer, 2023-05-06:
> Ah, thanks! Really interesting! For consistency with normal pointers I would also prefer `void lv_disp_set_flush_cb(lv_disp_t * disp, lv_disp_flush_cb_t * flush_cb);`.

Glad you like this idea.

kdschlosser, 2023-05-06:
I just realized something. There is a problem with some of the callbacks. You have a structure that is able to hold 3 more than 1 callback but only a single user_data field. How's that supposed to work?

When a callback gets set the user_data field would point to the "real" callback so to speak. In the CPython binding I am working on I am only able to provide a single entry point for any declared function pointer. What that means is I have to provide a callback that is hard coded in both Python and also in C code. so it's like a function pointer to a function pointer to python basically.

I am only able to pass a C function pointer and there is a bridge that CFFI makes between that C function pointer and the one that is hard coded in Python. The user_data is where the handle to the real callback gets stored. But if there is more than one callback for a structure and only a single user_data field this is not going to work properly because I am not going to be able to register more than one callback because I am only able to store a single handle to the Python function.

amirgon, 2023-05-07:
> You have a structure that is able to hold 3 more than 1 callback but only a single user_data field. How's that supposed to work?

In Micropython the `user_data` is used not to store the callable object itself, but a dictionary of callable objects where the key is the name of the function.
See [`get_callback_dict_from_user_data`](https://github.com/lvgl/lv_binding_micropython/blob/e4cc7668cd6ed170d71c874ab2ba15cc8804fd4a/gen/gen_mpy.py#L1383-L1389).

kdschlosser, 2023-05-07:
You are storing a C representation of a python dictionary.. what I am able to do with CFFI in Python is very limited. I am only able to declare function prototypes that point to the functions in LVGL. I am not able to add a function unless I write a C source file for it and header file and include the header and compile the source. Using the Python C API and getting it to work is above my skill set. To do it would require a fair amount of knowledge about the Python C API. The gil adds complexity to it and also how Python deals with reference counting.

How CFFI works with callbacks is that there is really only a single callback that gets assigned to each of the function pointers. That callback is a C callback and has a connection to a single python function. The Python function is what calls the actual user registered callback. In the MicroPython code you have a C function that makes the call to the Python function the user has registered. There is an additional layer I am dealing with. Whatever it is that gets passed to user_data has to be able to be passed back into Python and then converted into a python object and that object has to provide some way of identifying the user callback function.

I have to do some more investigation on this. I don't know what CFFI does when things get passed into Python.

kisvegabor, 2023-05-08:
Unfortunately, I can't comment much on it as I don't know libffi. 

But maybe it helps: I was thinking about adding a `lv_binding_t binding_data` to all types where it might make sense. If we don't a generate binding it should be ignored, of course, but for a binding we could define any data here, and `binding_data` could be used instead of `user_data`. It doesn't solve all the issues, as we don't know how many `void *` to add in it for the callbacks.  Having `lv_obj/anim/event/style_binding_t` would be also an option.

kdschlosser, 2023-05-08:
I think I might be able to work around the problem by wrapping a python dictionary and passing a handle to that wrapper as the user data. The (key, value) pairs in it would be ("function_name", python callback). because the python code is also generated I can add in the callback function name to get the correct callback function. IDK if CFFI will allow me to pass a handle of a wrapped dictionary. I don't see why not. The wrapper would have to be created as a global so it would never go out of scope causing it to get garbage collected causing the pointer to it in C code to become invalid.

The biggest issue I am having  is the whole private/public thing. If CFFI is not able to read the actual contents it is not going to be able to correctly set the size for it. void * has special handling to have a variable size.

kdschlosser, 2023-05-22:
What CFFI does and how it does I am really still learning. The documentation is next to useless, doesn't show any example code and the descriptions fomr the internals is very vague. some of the terms used really don't align properly.

here is an example.

So when I create a python representation of a C structure that object is called "cdata". It doesn't matter if it's a structure, union, primitive or array doesn't matter the object always has a type of "cdata". Now if that wasn't enough of a pain to deal with a way has been provided to get the real "type" of the object. using that you would think would return something that makes is easy to determine the type. no no no that would be way too easy. instead it returns an object that has a type "ctype". OYE!!!. Now comes the really fun part. That object has properties that can help determine what is actually being dealt with. This properties are not consistent and change depending on if you are dealing with a structure, array, pointer, ... etc... The real fun part is if a type in LVGL has been declared as `uint8_t *` and say that type is for a parameter of a function. What exactly should I be passing to that parameter an array or a pointer to a single integer? In C these are both fundamentally the same, in Python they are not. CFFI expects an integer to get passed and not an array but LVGL might actually be expecting an array.  `lv_disp_flush_cb_t` is an example of this where the px_map parameter is of type `lv_color_t *`. This comes through CFFI as a pointer to an lv_color_t structure and not an array of lv_color_t structures. This kind of handling is even further complicated by the use of void * because there is really no way to know how to handle those. When passing from Python to C I am able to figure out how to handle the data that is being passed. but going from C to Python it's a complete mystery how to deal with it.

If LVGL expects an array then it should be declared as `type[]`, doesn't matter if it's a union, structure, function, function prototype or a pointer to a function prototype any place that is exposed to the user needs to be done this way. That is the only way I am able to get some insight to be able to provide a wrapper class to handle the data properly. If it's a pointer that is being passed then `type *` is the way it needs to be done. In the case of `void *` the same would apply. I can wiggle my way around what the actual type is by taking a more dynamic approach with it by doing a "hunt and peck" to locate a proper or close to proper representation in Python for the type that has actually been passed.

CFFI complains about nested structures and it's not able to see what is inside of a structure that is nested. So this kind of code

```
/*Data of color picker*/
typedef struct {
    lv_obj_t obj;
    lv_color_hsv_t hsv;
    struct {
        lv_point_t pos;
        uint8_t recolor     : 1;
    } knob;
    uint32_t last_click_time;
    uint32_t last_change_time;
    lv_point_t last_press_point;
    lv_colorwheel_mode_t mode  : 2;
    uint8_t mode_fixed            : 1;
} lv_colorwheel_t;
```

causes some issue. in a dynamic setup it's less of a problem because no insight is used and leaves it up to the user to get things right. I am also not able to initialize the nested structure either because no python representation for that structure exists. setting fields directly only in the dynamic approach works but in a coded approach IDK how I would go about handling that situation. where the problems are is the lack of knowing what data type the fields have inside of the nested structure. it's a hard gap to bridge between C and Python without having it be really complex for the user and things like above make it even harder.

kisvegabor, 2023-05-22:
God, it really sounds crazy... 

> The real fun part is if a type in LVGL has been declared as uint8_t * and say that type is for a parameter of a function. What exactly should I be passing to that parameter an array or a pointer to a single integer? In C these are both fundamentally the same, in Python they are not.

The intention is to use `[]` for arrays, but it seems we missed it on one of the most important places (flush_cb)... However it's implemented e.g. [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/line/lv_line.c#L66).

>  I am also not able to initialize the nested structure either because no python representation for that structure exists. setting fields directly only in the dynamic approach works but in a coded approach IDK how I would go about handling that situation. 

We can assume that the user don't need to set the struct fields manually but only via API calls. If it's not true in some cases we should fix those cases.

> so an integer in Python is any of the int*_t or uint*_t types and also signed int and unsigned int as well. handling that can be done if I know what the expected data types are in the C code. 

But you know what is expected, right? As you know, in v9 I'd like to review the API names, and I'd like to use `int32_t` where possible. Maybe it helps a little.

kdschlosser, 2023-05-22:
Use the proper container for the size data that is being stored. It's the little things like the `[ ]`s that is really twisting me up. I have all of the callbacks sorted out and those are working fine. I have not tested everything and I am sure there are bound to be some glitches It this point I would say 95% of it is functional. All the most commonly used things should be in a working state. when getting into the really advanced stuff is where some problems are possibly lurking.

I know you are running Linux.  You should install PyCharm (it's free) and clone my repo. The make a new project in PyCharm that points to that repo. The compile the binding. You will understand why a stub file or having the entire thing statically coded with  data types is important. I cannot begin to tell you how much easier it is to write code when those things are in place. I m going to add in pulling the documentation from LVGL and adding it to the binding. Not to be compiled into HTML output, to be available in an IDE.

kdschlosser, 2023-05-22:
The use of `[ ]`s is not consistent. The crappy thing is there is no way to automatically correct them. There is no way to even search the files to locate things that should have them but instead have a \*. Have to manually go in and look at every structure, union, function parameter and function return type reading the code to see what it's used for. That's what I would have to do, someone that wrote the majority of LVGL would be able to locate the items that need to be changed much faster then someone who doesn't know that code that well.  😁 😉

kisvegabor, 2023-05-23:
> I know you are running Linux. You should install PyCharm (it's free) and clone my repo. 

Downloading... :slightly_smiling_face: 

> I m going to add in pulling the documentation from LVGL and adding it to the binding. Not to be compiled into HTML output, to be available in an IDE.

Do you mean to make it available on auto-complete? 

> That's what I would have to do, someone that wrote the majority of LVGL would be able to locate the items that need to be changed much faster then someone who doesn't know that code that well.

No worries we can review it as well when we will update the nameing.

kdschlosser, 2023-05-23:
if you use an IDE that is specifically made for Python all kinds of "goodies" are available if the code is written in a manner that allows those goodies to function. Things like being able to read the documentation from inside the IDE instead of having to go to a website and searching for what you want.

While object oriented languages have their upsides they also have complexity because of variables not having to be statically typed or declared a specific kind. It is easy to make a mistake and pass an incorrect type to a function. If that function is not a common use thing because the code is not compiled in the typical sense there is no errors or warnings that an incorrect data type is being passed until it actually gets passed and the program errors out. By adding types to the code an IDE is able to tell you that you are passing an incorrect type to that function. Typechecking in Python is not a runtime feature it is a feature that an IDE is able to use to tell you that you are passing an incorrect type.

In a Python IDE if I type in "lv." because it knows I imported the LVGL module as "lv" it will display a list of all available items in the module. as I type it narrows down the results.  so if i want to use the "refr_now" function I may only have to type in "ref" and then hit enter to fill in the rest.. it auto completes. It also pops up a dialog that tells me the parameters and the data types  for those parameters so I know what has to be passed to it. If I don't fill in one of the parameter it tells me that something is wrong by showing a red squiggly line under the function call.

kisvegabor, 2023-05-24:
I've just tried is out and it worked! Auto complete was working well too. I used Python in Eclipse in the past, it also supported these "IDE features" but PyCharm really looks great!

kdschlosser, 2023-05-24:
PyCharm is geared to Python where as Eclipse is not really designed to be specific to a language. There are plugins that can be gotten to allow it to work with other languages but what it is able to do is not going to have the same kind of feature set as an IDE that is specifically made for a language.

OH.... JetBrains (that's who makes PyCharm) Maakes all kinds of IDE's. They have one for JavaScript, PHP, C, SQL, GO, .NET, Ruby and Java. You can get all of them, commercial versions with all the bells and whistles turned on for the fantastic price of FREE.

https://www.jetbrains.com/community/opensource/#support

don't get me wrong PyCharm is a fantastic IDE but it does have it's glitches and issues. The biggest one is it being written in Java. It's a memory hog big time. and it doesn't handle really large code files all that well. I am talking files that are > 10,000 lines of code. It can have some strange behavior when working with files that large.

kdschlosser, 2023-05-24:
The really cool thing with having the python binding is the ability to inject code into it from an outside source real time. as in while it is running. Full blown python is able to execute code in the global namespace without needing to restart. So something like SquareLine Studio would not need a play button in order to run the GUI. It could be running and be edited at the same time.

You can actually code Squareline studio in Python using wxWidgets or QT which have Python bindings. the code can be compiled into a C extension using Cython which gives it the speed of C code and it also makes it so the source is not able to be seen because it has been compiled into a dynamic library essentially. If you used wx you can utilize portions of the GUI framework that allows you to have dockable panels. In the image below you can see the docking guides as an example.

[image]

kisvegabor, 2023-05-24:
> The really cool thing with having the python binding is the ability to inject code into it from an outside source real time. as in while it is running. Full blown python is able to execute code in [...]

What we do now is sending e.g. a `btn1.set_pos(10, 20)` MicroPython command to a "server" (MicroPython+LVGL compiled to an exe) and `exec` this command. How would it be different with real Python?

kdschlosser, 2023-05-24:
you do not have the ability to do more than one thing at a time. and they run as 2 completely independent programs. Where as with CPython there would not be that separation.... Python instead of being an "extra" or a secondary can be used directly as in You compile the C code for Squareline into a python extension module and Python would be able to interact with it. data can be shared between the 2 directly. Right now you have the ability to move objects or maybe even set their size while the script is running and that's about it. I am talking adding, deleting and changing any property while it is running. You can actually place a completely transparent window that renders the grab handles or any other graphic the user has to be able to see over the top of the running GUI program.

Take a look at this.

[image]

There are some glitches in the rendering for whatever reason that would need to get hammered out. Not sure why they are happening but that is LVGL being rendered to a completely transparent window.

The flexibility of Python is really amazing. a second module can be written that would wrap the binding code and those wrappers would hold the code for C, MicroPython or Python. when you want to generate the code a simple iteration over the child objects and a call to a function and the code gets spat out. Every single thing in LVGL can be wrapped very easily. Those wrappers can be automatically generated too.

There is a world of things that can be done using LVGL with a Python binding.

IDK what you are using as a GUI framework for Squareline or if you are custom rendering everything. you could use LVGL to actually render the GUI for Squareline if you wanted to. You would want to use wxPython and mutliprocessing to handle this so that you would be able to run multiple LVGL instances at the same time. tie the 2 running instances together so that one sits over the top of the other.

so much that can be done

kdschlosser, 2023-05-24:
OH and I know what those black marks are from. they are from the background color being used even tho the background color is set to fully transparent. Maybe I have to tell LVGL that I am using a display that supports transparency???

kisvegabor, 2023-05-24:
Having an integrated "Python server" to avoid the infinite number of possible issues of communication  2 applications would really amazing. 

We area already working SquareLine v2, in which we could partially solve this problem. I wish I could tell more about it because it would be a really meaningful discussion about pros and cons.

> Maybe I have to tell LVGL that I am using a display that supports transparency???

You need `lv_disp_set_color_format(disp, LV_COLOR_FORMAT_ARGB8888);`.

kdschlosser, 2023-05-25:
> You need lv_disp_set_color_format(disp, LV_COLOR_FORMAT_ARGB8888);.

PERFECT!!!!

Check this out...

https://youtu.be/-ci66hzU4lE

LVGL can be used to make OSD displays on  Windows PC's

This is a mechanism that can be used in Squareline Studio. 2 instances of LVGL running one on top of the other. The one that is on top is where all of the  grab bars would be located for resizing and changing position. the second would be for the running GUI program. If a grab bar is clicked on then mouse events don't propigate to the lower copy. Otherwise the clicks would get sent down and the UI is able to be interacted with.

kdschlosser, 2023-05-25:
[image]

kdschlosser, 2023-05-25:
I have a curious problem tho. ted and blue bits are reversed for the colors. don't know if it's an issue in LVGL or something in wxPython. I am going to have to run a test and see.

kisvegabor, 2023-05-25:
Wooha, looks sooooo amazing!!! I'm smiling like a kid :smile:  

> don't know if it's an issue in LVGL or something in wxPython

LVGL supports only a given RGB order. I think the order we support is the more common one and was enough so far, so I believe it's a wxWidget config somewhere.

kisvegabor, 2023-05-25:
> If I compile the code I write into a C extension it will get upwards of a 200% speed boost.

Compared to wxPython?

> Lots and lots of potential I can see for uses of LVGL on a PC with CPython.

Me two :slightly_smiling_face: Actually 3 things are required to make LVGL suitable for desktop app development:
1. Binding to a modern "desktop" language: Python is more then perfect
2. OpenGL, or similar support for faster rendering (especially on 4k screens)
3. A few missing features: would ~1 week of work to add.

---

Did you know that [LVGL's blog](https://blog.lvgl.io/) is open for everyone to write posts about LVGL related topics? So if you would like to share your project in more detail with fellow developers, I'd be happy add some marketing boost to you post. That post could be sent to https://hackaday.com/ too to reach a super large audience. Actually we did it with [this post about MicroPython](https://blog.lvgl.io/2019-02-20/micropython-bindings) and the page of lvgl.io (actually littlevgl.com at that time) was peaked to 5 times higher than usual.

kdschlosser, 2023-05-26:
You can promote it in any way you would like. 

I would only announce the availability of it when you are ready to release version 9

Just make sure to mention that is has both the C API and also the MicroPython API which allows for faster development. The C API would need to be converted to actual C code which wouldn't be all that difficult to do once a person has their UI running in Python.

BTW, Thank you for the comments on the blog about the documentation I do appreciate it. Hopefully the changes should make it easier for people to locate the information they are looking for. Still have to go through it and verify the external links and get all of the internal ones set up as references instead of links and update the information that is contained in it. We will do that just before version 9 gets released.

I am going to start working on the removal of the extra config file in the documents folder since it is causing some issues. It should be a pretty simple thing to fix.

kdschlosser, 2023-05-31:
question. would it make sense to create some macros for setting up callback structures since there are so many callbacks in LVGL

something along these lines

```c
#define CB_FIELD(type, name) type field;

#define CB_STRUCT(name, fields) \
struct { \
    void * user_data; \
    fields \
} name;
```

That would be used in this manner

```c

typedef void(* lv_test1_cb_t)(void);
typedef void(* lv_test2_cb_t)(void);
typedef void(* lv_test2_cb_t)(void);

typedef struct {
    uint8_t field1;
    uint8_t field2;
    CB_STRUCT(callbacks, CB_FIELD(lv_test1_cb_t, callback1) \ 
                       CB_FIELD(lv_test2_cb_t, callback2) \ CB_FIELD(lv_test3_cb_t, callback3))
    uint8_t field3;
} some_structure_t;

```

which would produce the following code

```c
typedef void(* lv_test1_cb_t)(void);
typedef void(* lv_test2_cb_t)(void);
typedef void(* lv_test2_cb_t)(void);

typedef struct {
    uint8_t field1;
    uint8_t field2;
    struct { 
        void * user_data; 
        lv_test1_cb_t callback1;
        lv_test2_cb_t callback2;
        lv_test3_cb_t callback3;
    } callbacks;
    uint8_t field3;
} some_structure_t;
```

kisvegabor, 2023-05-31:
Unless I miss something we have many callbacks only in `lv_anim_t` and I'm planning to rework it to event based. So there will be only a few callbacks there too.

Usually I'd like to avoid macro magic as it's hard to read and debug (the compiler errors related to macros are usually vague).

kdschlosser, 2023-05-31:
an event based system would be much easier to deal with. single entry point for all callbacks. need to be able to supply objects to marry a callback to or optionally None for a catchall for all objects. That would eliminate almost all of the function pointers and "user_data" fields in the various structures. 

A single structure could be used but an LVGL coding rule would have to be broken in order for it to work. The structure would have to contain 2 fields in it that are void * so they can have any kind of object stored in them. Function pointers would still have to be declared and used so the user will know what kind of parameters are being expected. Those types is what would go into one of the void * fields. The other would be for an object that is married to the function. A 3rd field would be needed to specify the event. That structure would also contain a user_data field and that structure would need to get passed to the function call as the first parameter.

This would make it super easy on the binding side of things because a single function would be needed specifying no object so it would be a catch all. A simple check to see if user_data is NULL and if it isn't then call the function contained within the user_data field. On the Python side of things There is no need to specify any additional parameters because it can bet up as a var_args. Iterate over the var_args coupled with type checking in order to determine the Python wrapper code that needs to be used for an arg. The same thing is able to be done in MicroPython and I am pretty sure the same thing is also able to be done in CPP. IDK about Rust/

kisvegabor, 2023-06-02:
>  need to be able to supply objects to marry a callback to or optionally None for a catchall for all objects. 

I'm not sure I understand correctly. So do you suggest setting the object for callback and not the callback for the object?

kdschlosser, 2023-06-02:
This is what I am seeing.

I am going to write this in python because it is easier to do.

a global function for registering and unregistering for events.

```python
class event_container_t:
    user_data = None
    callback = None
    event = None
    obj = None    

global_event_storage = []

def register_event(event, callback, obj, user_data):
    ec = lv.event_container_t()
    
    ec.obj = obj
    ec.event = event
    ec.callback = callback
    ec.user_data = user_data
    
    if obj is not None:
        # array inside of object to hold the events
        obj.event_callbacks.append(ec)
        
    else:
        # store event in a global catchall area so that every object that 
        # receives this event would call the callback
        global_event_storage.append(ec)
        
    # the returning of the ec is the "identifier" to be able to unregister an event    
    return ec

def unregister_event(ec):
    if ec.obj is not None:
        if ec in ec.obj.event_callbacks:
            ec.obj.event_callbacks.remove(ec)
    elif ec in global_event_storage:
        global_event_storage.remove(ec)
        
        
def signal_event(event, obj):
    for ec in obj.event_callbacks:
        if ec.event == event:
            ec.callback(ec)
            
    global_ec = event_container_t()
    global_ec.obj = obj
    global_ec.event = event
    
    for ec in global_event_storage:
        if ec.event == event:
            global_ec.callback = ec.callback
            global_ec.user_data = ec.user_data
            ec.callback(global_ec)
```

the event_container_t structure can be extended if additional parameters need to get passed. maybe have some kind of a marker in the structure so LVGL is able to identify its type. once it's type is identified then it can be cast properly in order to collect any parameters that might be stored inside of it that are specific to the event being used.

an example would be additional parameters that need to get passed to the callback. The user would be aware of the type of event they are registering for and they would be able to declare/define the callback function so the correct type is used for the first parameter. LVGL would be able to set that marker and cast to event_container_t for the return value and then cast to the correct type when a callback is being called.

something along these lines.

```python
class event_container_t:
    user_data = None
    callback = None
    event = None
[... 58 lines cut ...]
        ec = cast(item, event_container_t)
        
        if ec.event == event:
            if ec.type == 1:
                temp_ec = anim_event_container_t()
                ec = cast(ec, anim_event_container_t)
                _anim_event_copier(ec, temp_ec)
                global_ec = cast(temp_ec, event_container_t)
                
            else:
                global_ec = event_container_t()

            global_ec.obj = obj
            global_ec.event = event
            global_ec.callback = ec.callback
            global_ec.user_data = ec.user_data
            ec.callback(global_ec)

```

I know it would be more complex in C code but it gives a general idea. The only thing the event handler needs to know is the type of structure to use. because there are matching field names that would be shared across all event container types and the arrangement of the common fields shared between them allows for casting between them with any issues when accessing any of those common fields. Passing off to a function that knows how to deal with a specific container to copy the data stored within is the key.

So say a button gets pressed. there would be an event for that. LVGL objects get first positions in the event list because well they get added on creation of the object. so when a touch input event comes in that event gets sent out to all objects along with the x, y coords of the touch. the objects do a hit test and if there is a hit and they are the top most object the object changes its state and then sends out an event for it's state changing. Because the array of event structures is stored in the object it can simply iterate over those structures, locate the ones to send the event for, add the state data to the structure, cast the structure to event_container_t and call a function that handles making the callback.

You can speed up this process so you do not have to iterate over what could be a large array of events by separating user events from internal events using 2 different event systems that are set up almost identical. one gets used internally and is private and the other for events the user is able to register. This would allow for bitwise or'ed of event id's so multiple events would be able to be added to a single callback at one time,.. In this case the user would be responsible for getting the event type and casting it to what it needs to be.

kdschlosser, 2023-06-02:
You would have to cast to the proper callback type to make the callback but that would not be that hard to do because the event system knows what event type it is dealing with.

kisvegabor, 2023-06-05:
In v9 we have something similar with the `lv_event_dsc_t` and `lv_event_t`. The main difference is that there is no global list where events not belonging to any widgets could be added. But I wonder what is the practical use case for having these events detached from the objects. For example what problem it solves if a `my_pressed_event_cb` is called on any widget press?

kdschlosser, 2023-06-05:
It's easier from a binding standpoint. If the events can be or'd together and not attached to an object then only one callback would need to get registered between the binding and LVGL. What I am having an issue with is the MicroPython API portion of the CPython binding. The objects are broken into classes and the classes have methods that are specific to that class only. An example is the lv_label_set_text function. This function gets turned into a method and attached to the lv.label class. But the underlying C object is lv_obj_t type. So when a callback is made from lvgl that lv_obj_t structure is what gets passed to the callback function on the python side. I have no way of knowing what class the thing belongs to because it is not the same object that was passed when the callback was registered. It has a different memory address. It does contain the correct data but I am not able to do an equality test to pair it with the Python class it belongs to. I tried creating a handle to the Python class instance and passing that into the user data but I end up with memory corruption because of it. That corruption is more than likely because of a C object being contained inside that instance. It's funky.

If we could register a single callback that is not attached to a specific object I could handle the storage of the callbacks on the Python side of things. I can do this when the user registers for a callback. I would store the event the function and also the python class using a GUID as a way to identify the object from the C side of the code. I would set the GUID into the user data instead of a function. so if an event happened for an object that object would get passed to the binding callback and I can collect the GUID from the user_data and use that to collect the python object, registered events and any functions associated to those events. See if an event is registered and if it is I call the registered function passing the python class instance to the function.

Here is an example...

This is an example of a callback function used to set some specific feature an arc has when the knob for the arc gets clicked on

``` python
def add_mask_event_cb(e,mask_map):

    code = e.get_code()
    obj = e.get_target_obj()

    obj.set_knob_offset(10)
```

The return type attached to the `lv_event_get_target_obj` function is of type `lv_obj_t` and that is all I am able to determine I am not able to tell if the if the object is an arc, label or any other kind of widget and because I am not able to tell I cannot provide aa proper class to attach the C object to so the proper methods would be assigned.

```
    obj = e.get_target_obj()
```

the micropython binding handles this aspect of things on one side of the binding. it does it in C code where as I need to do it on the opposite side. and it would be easier to handle by only having to write a wrapper around a single function pointer and all events would be received in the binding via a single callback function on the Python side of things.

I cannot think of an easier way to explain it. 

You would have to go and look in the binding at the generated `__init__.py` file if you search that file for `@_lib_lvgl.ffi.def_extern(` there are 61 matches. there are 61 different callback functions so far that have their types declared as typedefs. that's a lot. The number is larger than that, it's just that I have not moved all of the function pointers out of the structures yet. 

What complicates it further is some have a function to register a callback and some of those functions have a user_data param and some do not. some have a separate function for setting the user_data. Then you have things like the event callback which takes an obj_t as the first parameter and that structure has does have a user_data field in it but that structure does not get passed to the callback function. instead what gets passed back is a structure that does have a user data field but it is not an object that can be married to what the event is actually attached to. I have no way to make a connection between the object that was used to register an event and the object that is returned by `lv_event_get_target_obj` other than storing some kind of a marker in the user_data which is already being occupied by a handle to the python function. I cannot keep a reference to that function stored somewhere because if the function is a method of a class if the instance of that class goes out of scope it should get deleted and garbage collected but if I am holding a reference to any part of it that would not be able to take place.

It's a really twisted mess of things that have to be handled in just the right way to not run into behavior that is not wanted. I can make a weak reference to it but that is going to come at a big cost of having to iterate over a large storage container and opening up each weak reference and doing an equality test to see if the function I have store is the same as the one that is in the user_data field of the structure. on larger UI's that is going to be a bottleneck having to do that every time an event takes place.

Now having to manage that mess of crap along with there being over 61 functions to do it for!?!??! for lack of better words it would be a shit show.

kisvegabor, 2023-06-06:
The callback conventions are really messy. I think one part of the confusion comes from the name of `user_data`. I think we should have `user_data` which is really for the user (and the binding does nothing with it), and a `binding_data` which will be enabled/added only if we are building a binding. So `lv_obj_t` would have both `user_data` and `binding_data`.

With events we also have 2 `user_data`: one when the event is registered and one when an event is sent. Let's call these `reg_data` (for registration) and `event_data` (for sending).

For this on its own makes things lot more cleaner for me.

---

I still don't understand something about your idea:
```py
if obj is not None:
        # array inside of object to hold the events
        obj.event_callbacks.append(ec)
        
    else:
        # store event in a global catchall area so that every object that 
        # receives this event would call the callback
        global_event_storage.append(ec)
```
This registers an event either on an object or in the list. But registering and event to the list (not to an object) is not that useful, so it won't really happen in the practice. Therefore we will have the same mechanism as now, that is registering events on the objects.

kdschlosser, 2023-06-06:
if there is a way to combine all of the callbacks in LVGL into a single place. a single function to register no matter what the event is or the object is. and a single entry point for "signaling" or calling callbacks registered to an event.

The purpose of an event is not necessarily to pass any information to the callback function. the intention should be focused on simply calling the function and passing the object that has triggered the event. The code inside of the function would then be responsible for collecting whatever information it needs by calling other functions to collect that information.

There is a smoke and mirrors kind of approach to how the events are handled currently. I pass an lv_obj_t structure to register for an event but that object does not get sent back when the callback is made. there is a disconnect. There is no user_data available in the event_t structure to collect the handle to the function that needs to be called.. in my case CFFI does some kind of kung fu on the lv_ob_t structure that get passed. when the event gets registered and the structure gets passed to the callback is it not the same instance of it. It has the same data but a different memory location leaving me zero way to compare it to what is already stored on the python side.

if there is a way for me to identify the "sub type" it would make it easier to deal with. On the C side of things you have the object classes but they are not type definitions so I cannot use them as a way of identifying if an lv_obj_t  structure is aan lv_arc_class or an lv_label_class. An lv_arc_class and an lv_label_class both have the same type which is lv_obj_class_t.

```c
typedef lv_obj_class_t lv_arc_class_t;

const lv_arc_class_t lv_arc_class  = {
    .constructor_cb = lv_arc_constructor,
    .event_cb = lv_arc_event,
    .instance_size = sizeof(lv_arc_t),
    .editable = LV_OBJ_CLASS_EDITABLE_TRUE,
    .base_class = &lv_obj_class
};

now I would be able to identify the object.

What I do not know is if you can put an object that has the lv_arc_class_t type into say a field of a structured that has a type of lv_obj_class_t without having to cast it into that field and then back out of that field.

I am looking at  a typedef as kind of like subclassing in Python. The child class has the same type as the parent class but the parent class does not have the same type as the child. if that makes any sense?

```python
class Parent:
    pass

class Child(Parent):
    pass

parent = Parent()
child = Child()

print(isinstance(child, type(parent)))
print(isinstance(parent, type(child)))
```

The output from that is

```
True
False
```

can the following be done in C code without having to cast?

```c
typedef struct {
    
} Parent;

typedef Parent Child;

typedef struct {
    Parent *some_field;
} SomeStruct;

Child * child;

SomeStruct some_struct = {.some_field=child};
```
 the casting means you have to have prior knowledge of what type is being expected.  This is what I do not have in the binding when something is being sent from the C side to the Python side that was not directly asked for.

kisvegabor, 2023-06-08:
Note that callbacks are not only events. There can be other kind of callbacks too. Maybe they all can be converted to events, but I can't guarantee it for the future.

> I pass an lv_obj_t structure to register for an event but that object does not get sent back when the callback is made.

It does, but inside the `lv_event_t * e` pointer. E.g. `lv_obj_t * obj = lv_event_get_target(e);`

> I am looking at a typedef as kind of like subclassing in Python. The child class has the same type as the parent class but the parent class does not have the same type as the child. if that makes any sense?

Do these help?

```c
/**
 * Check the type of obj.
 * @param obj       pointer to an object
 * @param class_p   a class to check (e.g. `lv_slider_class`)
 * @return          true: `class_p` is the `obj` class.
 */
bool lv_obj_check_type(const lv_obj_t * obj, const lv_obj_class_t * class_p);

/**
 * Check if any object has a given class (type).
 * It checks the ancestor classes too.
 * @param obj       pointer to an object
 * @param class_p   a class to check (e.g. `lv_slider_class`)
 * @return          true: `obj` has the given class
 */
bool lv_obj_has_class(const lv_obj_t * obj, const lv_obj_class_t * class_p);

/**
 * Get the class (type) of the object
 * @param obj   pointer to an object
 * @return      the class (type) of the object
 */
const lv_obj_class_t * lv_obj_get_class(const lv_obj_t * obj);
```

E.g. 
```c
if(lv_obj_has_class(slider1, &lv_bar_class)) -> true: as bar is the ancestor type of slider
if(lv_obj_has_class(arc1, &lv_bar_class)) -> false
```

Noe that `&lv_bar_class` is a variable not  a tpye.

kdschlosser, 2023-06-08:
are all `lv_*_class` variables?

This is where it would get to be a problem because of the example you just pointed out.

In MicroPython the widgets are a subclass of lv_obj I do not believe that there is any other subclassing that is done. In the example you pointed out if lv_slider is a child of lv_bar and lv_bar would be a child of lv_obj I have no way of knowing that when building the API. This is because the only thing I have the ability to do it to check the return type from the create function and the return type for both lv_bar and lv_slider is lv_obj_t. because the return type if lv_obj_t the parent class would be an instance of lv.obj

I am not sure how I would go about making an attachment from lv_slider to lv_bar. I do not believe this attachment is even done in the MicroPython API.

also... how would this evaluate?

```c
lv_obj_has_class(bar1, &lv_slider_class)
```

kisvegabor, 2023-06-09:
In Micropython for example for the slider the `obj` and `slider` APIs are added to `slider1.set/get`. I don't how this magic happens. :smile: Probably @amirgon can tell more about it.

kdschlosser, 2023-06-10:
The MicroPython binding does not have a structure that is consistent with what `lv_obj_has_class` dictates. if it did then this would not be False.

```python

slider = lv.slider(None)

isinstance(slider, lv.bar)
```

which it is False

what you have said is the slider is a child class of bar and if that is the case then the above would not return False in MicroPython. because of that I am not going to be able to use that function in order to determine the structure to mimic them API of the MicroPython binding unless I hard code the use of that function in which I would need to know the inheritance arrangement of all of the classes

This is what I mean by inheritance

obj is a parent class of bar is a parent class of slider. 

so 

```c
lv_obj_has_class(slider, &lv_obj_class) == true
lv_obj_has_class(slider, &lv_bar_class) == true
lv_obj_has_class(slider, &lv_slider_class) == true
```

But this is what I am seeing in the header file for the slider.

```c
extern const lv_obj_class_t lv_slider_class;
```

which leads me to believe that slider in not a child class of bar.

but then in the source file there is 

```c
    .base_class = &lv_bar_class
```

animimg is a child of img
canvas is a child of img
keyboard is a child of btnmatrix
list button is a child of btn
list label is a child of label

now this one if interesting. 
lv_templ_class has a bass_class of lv_templ_class

the problem is the base class is not defined in a header file it is defined in the source file so there is no knowledge to the binding of this arrangement. You have to remember that pycparser is reading text from a file and nothing more so when header files get preprocessed there are no includes for any of the source files and because of that there is not going to be anything included in that preprocessor output that would tell pycparser of the class inheritance. I am not going to be able to build a proper representation and neither is the MicroPython binding either.

amirgon, 2023-06-10:
> In Micropython for example for the slider the `obj` and `slider` APIs are added to `slider1.set/get`. I don't how this magic happens.  Probably @amirgon can tell more about it.

@kisvegabor It became harder for me to follow this issue, sorry.  Too little time and too much text to digest in here.
If you need some answer from me, could you please shortly summarize? What "magic" are you referring to?

kisvegabor, 2023-06-12:
I think the main question is how do you figure out the real widget type (class) from `lv_obj_t *`?

amirgon, 2023-06-12:
> I think the main question is how do you figure out the real widget type (class) from `lv_obj_t *`?

Thank you for a very short and focused question!  

I'm calling `lv_obj_get_class` to find out the lv_obj class, and create the appropriate Micropython object accordingly.
Here is the relevant code: https://github.com/lvgl/lv_binding_micropython/blob/e0f97fc773ffa57ea61f2f41eebca8e7844329c6/gen/gen_mpy.py#L930-L937

kdschlosser, 2023-06-13:
> No. As discussed several times in the past, all LVGL object classes inherit from lv_obj.

of course they do. If the inheritance was like this

`lv_obj -> lv_bar -> lv_slider`

lv_slider would inherit from lv_obj. which is why this question has come up.

I was not aware that lvgl has inheritance between the widgets and not just lv_obj. and because you did not mention this inheritance in LVGL in our previous discussions I was working off the information I was provided which was the widgets only inherit from lv_obj.

Thank you for taking the time to clarify that. I am trying to duplicate the API in the CPython binding and I am not going to be able to do that unless I know these things. Had I used the function that @kisvegabor pointed out it would have caused different behavior. So now I am back at square one because I am not able to use that function to convert a C object to a Python object due to it not reporting the same inheritance that would align with the MicroPython API.

amirgon, 2023-06-13:
> of course they do. If the inheritance was like this
> 
> `lv_obj -> lv_bar -> lv_slider`
> 
> lv_slider would inherit from lv_obj. which is why this question has come up.

@kdschlosser I'm not sure I understand.
In the Micropython bindings every widget inherits from lv_obj (except lv_obj itself).  Pretty simple.
It was discussed and agreed with @kisvegabor - see https://github.com/lvgl/lv_binding_micropython/issues/19

Why do you want the inheritance to work differently in the Python bindings?

[41 more replies]

closed 2023-07-23 by lvgl-bot