lvgl/lvgl #4205: keypad input device callback is not triggered

tranvnhan, 2023-05-05:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2023-05-05:
Are you sure the textarea is focused? 

Please call `lv_group_focus_obj(textarea);` to be sure the ta is focused.

tranvnhan, 2023-05-06:
> ```c
> keypad_indev 
> ```

Hi. The ta is focused correctly.
I found out that the problem is due to I declared `keypad_indev` inside `setup()`.
When I declare it as a global variable, the program can run appropriately.

```c
// Declare as global variable
static lv_indev_t * keypad_indev;
```

I can now use the keypad to control the display.
I will close the issue now.
Thanks.

closed 2023-05-06 by tranvnhan