lvgl/lvgl #4208: How not to send LV_EVENT_CLICKED when the touch screen slides

qinshijing, 2023-05-05:
### Introduce the problem

In a touch screen scenario, the user presses a button, but the user expects no LV_EVENT_CLICKED to be generated after moving the finger.

### Examples and cases

Although I can add a piece of code like the following to each event response, it doesn't feel very elegant.
//////////////////////////////////////////
    case LV_EVENT_CLICKED: 
        lv_indev_t* indev = lv_event_get_indev(e);
        if (!indev || LV_ABS(indev->proc.types.pointer.scroll_sum.x)
                >= LV_INDEV_DEF_SCROLL_LIMIT
            || LV_ABS(indev->proc.types.pointer.scroll_sum.y)
                >= LV_INDEV_DEF_SCROLL_LIMIT) {
            sr_info("be scrolled, not click.\n");
            return;
        }
//////////////////////////////////////////
### Suggested solution

How about adding it directly in lv_indev.c? Or is there a better way?
//////////////////////////////////////////
         if (proc->types.pointer.scroll_sum.x <= LV_INDEV_DEF_SCROLL_LIMIT
                && proc->types.pointer.scroll_sum.y <= LV_INDEV_DEF_SCROLL_LIMIT)
                lv_event_send(indev_obj_act, LV_EVENT_CLICKED, indev_act);
//////////////////////////////////////////

qinshijing, 2023-05-08:
Yes, it should be reasonable if the finger does not move obj to generate clicked. But there is a scenario where the user's finger does not leave obj, but the finger movement distance occurs. I want to use short_clicked to handle this kind of scene, but short_clicked only has time judgment, no distance judgment. When there is a gesture response and short_clicked scene, if no gesture is triggered, short_clicked will be generated, so you think the short_clicked event increases the scroll_sum judgment reasonable?

kisvegabor, 2023-05-08:
I'd like to understand the scenario when this kind of judgement is required. Could you describe your use-case?

closed 2023-05-30 by lvgl-bot