lvgl/lvgl #4210: lv_scale

C47D, 2023-05-05:
### Introduce the problem
I'm currently working on the horizontal `lv_scale` widget and getting familiar with the drawing mechanisms of LVGL. It seems like the drawing gets masked at the far left.

Here the tick starts at 25 and has 10 ticks, with major ticks every 5.

[image]

### Examples and cases
None

### Suggested solution
Current implementation here: https://github.com/C47D/lvgl/tree/lv_scale

kisvegabor, 2023-05-08:
Hi Carlos,

Be default the widgets can't draw outside of their coordinates.  The add some extra draw area you need to use the `LV_EVENT_REFR_EXT_DRAW_SIZE` event. See an example here https://docs.lvgl.io/8.3/examples.html#slider-with-extended-drawer

kisvegabor, 2023-05-23:
Let's continue in the PR: https://github.com/lvgl/lvgl/pull/4196

closed 2023-05-23 by kisvegabor