lvgl/lvgl #4212: Memory not be fully freed when lv_scr_load_anim() time of animation is 0

SendToSouthEast, 2023-05-07:
LVGL Version: tag 8.3.6 (my esp-idf version cannot compile tag 8.3.7)
Platform: ESP32

Description:

It is very strange that when I use `lv_scr_load_anim(scr, LV_SCR_LOAD_ANIM_FADE_ON, 400, 0, true);` to load the screen with a time of animation of 0, the memory can be correctly loaded and freed. However, when I use `lv_scr_load_anim(dictScene,LV_SCR_LOAD_ANIM_NONE,0,0,true);` the memory cannot be completely freed, which leads to a crash after repeatedly loading the interface multiple times. Through mem monitor, I found that the memory is not completely freed. When I use `lv_scr_load_anim(dictScene,LV_SCR_LOAD_ANIM_NONE,10,0,true);`, the memory can be correctly released.
For example, when a screen is opened, it may use 7.1K of memory, but when another screen is opened and then returned to the original screen, the memory usage may increase to 8.3K.

Here is my test code.

```
lv_obj_t* scr;
lv_obj_t* scr1;

void win_test1(void)
 {
     scr = lv_obj_create(NULL);
     lv_obj_set_size(scr, LV_HOR_RES, LV_VER_RES);
     static lv_style_t style;
     lv_style_reset(&style);
     lv_style_init(&style);
     lv_style_set_bg_color(&style, lv_palette_main(LV_PALETTE_RED));
     lv_style_set_text_color(&style, lv_color_white());
     lv_obj_add_style(scr, &style,NULL);

     lv_obj_t* label = lv_label_create(scr);
     lv_obj_set_style_text_color(label, lv_color_hex(0x990000), LV_PART_MAIN | LV_STATE_DEFAULT);
    

     lv_label_set_text(label, "This is a test 1");

     lv_scr_load_anim(scr, LV_SCR_LOAD_ANIM_FADE_ON, 400, 0, true);
     printf("win test 1 is created ");
 }

 void win_test2(void)
 {
     scr1 = lv_obj_create(NULL);
     lv_obj_set_size(scr1, LV_HOR_RES, LV_VER_RES);
     static lv_style_t style;
     lv_style_reset(&style);
     lv_style_init(&style);
     lv_style_set_bg_color(&style, lv_palette_main(LV_PALETTE_BLUE));
     lv_style_set_text_color(&style, lv_color_white());
     
     lv_obj_add_style(scr1, &style,NULL);

     lv_obj_t* label = lv_label_create(scr1);
     lv_obj_set_style_text_color(label, lv_color_hex(0x990000), LV_PART_MAIN | LV_STATE_DEFAULT);

     lv_label_set_text(label, "This is a test 2");
     
     lv_scr_load_anim(scr1, LV_SCR_LOAD_ANIM_FADE_ON, 0, 0,true);
     printf("win test 2 is created ");
 }
```

```
while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(10));
        keyCheakTask();
        lv_task_handler();

        static uint16_t i;
        if (++i > 100) {
            i = 0;
            static bool flag;
            if (flag)
            {
                flag = false;
                win_test2();
            }
            else
            {
                flag = true;
                win_test1();
            }
            
      	}

    }
```

Another issue that is not related to the above test code is that I was not familiar with the usage of screens earlier, so I manually created a interface using `Scene = lv_obj_create(lv_scr_act());` and created interface widgets with Scene as the parent object. When I call `lv_del_obj(Scene)` to log out of this interface, the memory cannot be completely released. It can only release 30% to 70% of the memory. I want to know if this is because I did not use the function correctly or for other reasons. However, the symptoms of this issue appear very similar to the previously mentioned issue.

Thank you for your response!

SendToSouthEast, 2023-05-14:
> Hi,
> 
> It works normally for me: [image] [ [image] ](https://user-images.githubusercontent.com/7599318/236854565-4a47aa5b-9daa-4d59-b85c-36eb56def17c.gif) [ ](https://user-images.githubusercontent.com/7599318/236854565-4a47aa5b-9daa-4d59-b85c-36eb56def17c.gif)
> 
> Do you use a LVGL's allocator or a custom one? (see `LV_MEM_CUSTOM` in `lv_conf.h`)

Hi！
this is my memory settings（on esp32）:
[image]

I tried running on the codeblock emulator and found it to work fine. However, on the ESP32 platform, there is a memory problem.
Here is the video

https://github.com/lvgl/lvgl/assets/93421418/7081bd55-4e40-4fa0-a697-f5d2dc68625c

Thankyou！

kisvegabor, 2023-05-18:
Thank you, I've found the problem and pushed a fix to the `release/v8.3` branch.

commit 1caafc55 (2023-05-18): fix(disp): fix memory leak lv_scr_load_anim with auto_del and time=0

fixes: #4212
closed 2023-05-18 by kisvegabor