lvgl/lvgl #4216: Drag a label not work

satar1980, 2023-05-10:
static void drag_event_handler(lv_event_t * e)
{
    lv_obj_t * obj = lv_event_get_target(e);

    lv_indev_t * indev = lv_indev_get_act();
    if(indev == NULL)  return;

    lv_point_t vect;
    lv_indev_get_vect(indev, &vect);

    lv_coord_t x = lv_obj_get_x(obj) + vect.x;
    lv_coord_t y = lv_obj_get_y(obj) + vect.y;
    lv_obj_set_pos(obj, x, y);
}

void main(void)
{
    lv_obj_t * label = lv_label_create(lv_scr_act());
    lv_obj_add_event_cb(label, drag_event_handler, LV_EVENT_PRESSING, NULL);
    lv_label_set_text(label, "Drag me");
    lv_obj_center(label);

}

kisvegabor, 2023-05-11:
Labels are not clickable by default so you should add
```c
lv_obj_add_flag(label, LV_OBJ_FLAG_CLICKABLE);
```

kisvegabor, 2023-05-12:
Great! :slightly_smiling_face:

closed 2023-05-12 by kisvegabor