lvgl/lvgl #4217: Regarding the issue of some driver chips requiring even row and column addresses or related lengths.

snoooopy, 2023-05-11:
### Introduce the problem

I am working on the driver of the chip RM67162/RM69330. The user manual describes "The SC [9:0] and EC [9:0] - SC [9:0]+1 must be visible by 2."
[image]

How should this be handled in Lvgl? The current display of Spinner or screen refresh is incorrect, as shown in the following image.
[image]

### Examples and cases

### Suggested solution

kisvegabor, 2023-05-11:
Hi,

You can use the `rounder_cb`. See [here](https://docs.lvgl.io/8.3/porting/display.html).

snoooopy, 2023-05-12:
> Hi,
> 
> You can use the `rounder_cb`. See [here](https://docs.lvgl.io/8.3/porting/display.html).

It's really helpful. Thank you so much. I have modified the round_cb according to the requirements of the driver chip,and  the screen is working properly now!
[image]

closed 2023-05-12 by snoooopy

Robox-Robotics, 2023-06-30:
Hello @snoooopy , I'm developing a smartwatch based on the same display I bought from Alibaba. Can you help me with steps you followed to make the custom driver for RM69330?