lvgl/lvgl #4218: Map css cubic-bezier(x1, y1, x2, y2) to lv_bezier3 parameters

XuNeo, 2023-05-12:
### Introduce the problem

Normally in css, the animation path is specified via `cubic-bezier(x1, y1, x2, y2)` and the UI designer also tends to use it.
I cannot figure out how to map the x1, y1, x2, y2 parameter to u0,u1,u2,u3 in lv_bezier3 function. 

For example, `ease-in-out` is specified by `cubic-bezier(0.42, 0.0, 0.58, 1.0)`, and `lv_bezier3` uses equilent(shift 10bits) parameter (0, 0.0488, 0.9297, 1).

Is there an equation to map those parameters? 

Thanks.

### Examples and cases

### Suggested solution

kisvegabor, 2023-05-18:
According to this [comment](https://stackoverflow.com/a/21962457/1999969):
```
x(t) = (1.0 - t^2) * p0.x + 2.0 * (1 - t) * t * p1.x + t^2 * p2.x;
y(t) = (1.0 - t^2) * p0.y + 2.0 * (1 - t) * t * p1.y + t^2 * p2.y;
```

Here the problem is that we increment `t` gradually and we get `x` and `y`. And `x(t)` depends on only `p.x` and `y(t)` only on `p.y`. But what we need is `y(x)`. So you move on the x axis (time in case of an animation) and get the y coordinates (the value of the animation).  

[image]

However, it's not so simple to rearrange `x(t)` to get `t` for each gradually increasing `x` and than calculate `y(t)`. To be honest I don't know how and what CSS calculates but it seems to me the simple bezier equations can't be used for animations as they are.  When I implemented `lv_bezier3` I found that there is a version where the x cooridnate of the control points are fixed at 0.33 and 0.66. This way we really have only a `y(t)` function.

Having a real bezier curve animation would be really great, so if you have any ideas, I'd highly appreciate it :slightly_smiling_face:

kisvegabor, 2023-05-18:
Useful reading, but unfortunately the linked book is no available: https://math.stackexchange.com/questions/841169/fastest-way-to-obtain-the-parametric-value-t-of-a-bezier-curve-for-a-given-set

XuNeo, 2023-05-22:
Thanks for your great comments, I now can understand what the problem is and found an implementation lib for `js` [here](https://github.com/gre/bezier-easing).

For reference, the cubic-bezier function is defined as
```
X(u) = (1-u)^3 * X0 + 3*(1-u)^2 * u * X1 + 3*(1-u) * u^2 * X2 + u^3 * X3
Y(u) = (1-u)^3 * Y0 + 3*(1-u)^2 * u * Y1 + 3*(1-u) * u^2 * Y2 + u^3 * Y3
```

where u is in range of `[0, 1]`, and `[X0, Y0]` is `[0, 0]`, `[X3, Y3]` is `[1, 1]`.
Calculate the corresponding X(u) and Y(u) give us the curve normally shows in tutorial.

I think the most important condition for cubic-bezier function in css is that given a value `x`, there's only one `u` can be found, with the value `u` found, then we can calculate `y` use above equation. 

For `lvgl`, the x maps from `t` which represents time, and `y` maps to `value` which will be set to obj's style etc.

The next step for me is to check how to port `js` lib to lvgl. Until then, I'll close this issue for now, thanks.

Useful links:
- https://stackoverflow.com/questions/7348009/y-coordinate-for-a-given-x-cubic-bezier 
   - The first answer uses binary search, which is easiest to understand.
   - Second one uses mathematic solution, the comment also suggests that when you found mutilple roots, there's only one suits for cubic-bezier, because `u` is limited to [0, 1]. It's also explained in below link.
- https://algorithmist.wordpress.com/2009/10/15/bezier-y-at-x-algorithm/

closed 2023-05-22 by XuNeo

kisvegabor, 2023-05-22:
Very nice! It's great that you look into it :hugs: