lvgl/lvgl #4219: lv_tick_inc

kdschlosser, 2023-05-12:
this is a continuation of the last couple of posts located 
[here](https://github.com/lvgl/lvgl/commit/42f70fecf0c3ab3afb1591aace197be0974b54d9)

Now it was brought up about the use of SDL_tick. Using the mechanism that SDL uses on the Windows OS the time measurement over one second is 1014.0053. using a different mechanism in Windows I get a time measurement over 1 second of 1002.7444. 12 milliseconds is a whole lot of inaccuracy and considering LVGL uses millisecond timing 12 milliseconds equates to over 1/2rd of the refresh delay that comes set. 

So SDL has inaccurate timekeeping....

My original suggestion of being able to add a callback at runtime to get the number of ticks that has elapsed would allow for better handling of timer execution especially during lengthy frame buffer flushes. It would also eliminate the need to have an interrupt that has the sole purpose of incrementing the tick.

kisvegabor, 2023-05-18:
So you mean the runtime configurable option of this?
https://github.com/lvgl/lvgl/blob/c4ead473743f18cf5f603c7ae379e560c05ca44f/lv_conf_template.h#L90-L97

kdschlosser, 2023-05-18:
If you look at the latest version of my binding I added this to the code. It is not a requirement to use the way I did it. There is also an example written in the example folder for it so you can see how it works.

kisvegabor, 2023-05-19:
Changing the interface is a sensitive topic as it's one of the basic interfaces of LVGL. We can update it but it requires careful planning. 

So I'd postpone it a little bit and focus more on the callbacks and similar topics.

kdschlosser, 2023-05-19:
it is something that I had noticed would be a way of getting accurate time for LVGL, was easy to add.  LVGL is able to be ran with out without it's use and it doesn't break any of the current API at all...

closed 2023-06-11 by lvgl-bot