lvgl/lvgl #4221: performance monitor frames per second is very inaccurate.

kdschlosser, 2023-05-14:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
the frames per second calculation in the performance monitor produces an incorrect/inaccurate number.

### To Reproduce

I have to add the code to do the FPS calculations to a repo I have for a CPython binding. Once I do that you would have to clone the repo I have for a cpython binding of LVGL compile the binding using the included LVGL and install wxPython then run the wx_python script that is included. This would show you the FPS that LVGL calculates and it would print out to the terminal the real FPS.

### Expected behavior
For the FPS to be correct

### Screenshots or video
This is after I had already made some changes to improve how LVGL is doing the calculations.

[image]

It was much further off then what you see in the above image.

[image]

The printed numbers are the same but LVGL was reporting a way higher number.

The problem is multifold. There is not a singular cause. I will list the issues here.

* assumption that exactly 1000 milliseconds (1 second) has passed between when calculations are done.
* doing elapsed time calculation each time a frame does or does not get rendered.
* frame count gets incremented even when nothing gets drawn to the display

The calculation should be

`fps = 1 / ((stop_time - start_time) / frame count / time_resolution)`

the start time gets set to the current ticks when the calculation is done. The frame count gets reset to zero. The only thing that is updated when the display buffer gets written is the frame count. The frame count should ONLY be updated if there is data that needs to be written to the display. When the timer for the performance monitor expires it doesn't mean that is when the calculation actually gets done. It is when `lv_task_handler` gets called and that is completely dependent on the user. 

Setting an IRQ to increment the tick is not a good thing to do. This is because you will end up with really inaccurate time keeping in LVGL. If you have an interrupt run at less than 20 millisecond intervals it starts to really cause performance issues. There is a lot of work the MCU has to do for an interrupt. It has to store the current state of the CPU in memory which costs CPU time to do then load the interrupt code and run it and when finished it has to restore the state of the CPU. This all takes processor time to accomplish. The numbers seem small but if you then multiply them into say a single minute of running time you will see what the real cost is.

LVGL should be able to request a new timestamp when it needs to handle timing related operations. The easiest way to be able to do this is by using runtime callback that the user would supply. This can be set up one of 2 ways, the function can either return the amount of time that has passed or it can return the timestamp. I am in favor of returning the amount of time that has passed. This would allow the user to get better precision by using a higher resolution timestamp and handling the loss of precision when converting the time to milliseconds.

LVGL already has some of the framework to accomplish this in place, It is done when compiling using the Arduino IDE.  It is something that has to be done at compile time. The reason why I say at runtime is so bindings would be able to use it as well. Because LVGL has the ability to be run on a multitude of different platforms it would not make sense to add code for each type of platform. Let the user handle how they want to do the calculations.

The code below is the callback  function that is used for refreshing the display. It is also the function that gets called by `lv_refr_now`. `lf_refr_now` has the use of this function hard coded into it. It should not be hard coded, it should use whatever function has been given to the refresh timer. If the user has a custom function for some reason they would not be able to utilize `lv_refr_now`. there is use of this function in LVGL if SDL is being used. If there was no use of the function by any LVGL internal code then there would be no issue with how it is done currently.

Can you see the problem?

```c

void _lv_disp_refr_timer(lv_timer_t * tmr)
{
    REFR_TRACE("begin");

    uint32_t start = lv_tick_get();

    if(tmr) {
        disp_refr = tmr->user_data;
#if LV_USE_PERF_MONITOR == 0 && LV_USE_MEM_MONITOR == 0
        /**
         * Ensure the timer does not run again automatically.
         * This is done before refreshing in case refreshing invalidates something else.
         */
        lv_timer_pause(tmr);
#endif
    }
    else {
        disp_refr = lv_disp_get_default();
    }
[... 268 lines cut ...]
    void * buf_off_screen = disp_refr->draw_buf_act;
    void * buf_on_screen = disp_refr->draw_buf_act == disp_refr->draw_buf_1
                           ? disp_refr->draw_buf_2
                           : disp_refr->draw_buf_1;

    lv_coord_t stride = lv_disp_get_hor_res(disp_refr);
    uint32_t i;
    for(i = 0; i < disp_refr->inv_p; i++) {
        if(disp_refr->inv_area_joined[i]) continue;
        disp_refr->draw_ctx->buffer_copy(
            disp_refr->draw_ctx,
            buf_off_screen, stride, &disp_refr->inv_areas[i],
            buf_on_screen, stride, &disp_refr->inv_areas[i]
        );
    }

    refr_clean_up();
    refr_cache_clean_up();
    draw_mask_cleanup();
    lv_disp_send_event(disp_refr, LV_EVENT_REFR_FINISH, NULL);
    REFR_TRACE("finished");
}
```

now the event ONLY gets called when actual data has been written to the display.

kdschlosser, 2023-05-14:
If you wanted to capture the frame rate accurately the start and stop times should be captured by the callback function and the difference between the 2 should be calculated in the event callback. This would remove the time that gets used to make the callback from being added into the FPS calculation. I know it is a really small number but it is time that should not be factored in.

kisvegabor, 2023-05-18:
What if we use `LV_EVENT_RENDER_READY`? We just need to move it from [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L303) to [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L329). 

This way `LV_EVENT_REFR_START/FINISH` can used to do something in each rendering cycle, and `LV_EVENT_RENDER_START/READY` to catch the real rendering if any

kdschlosser, 2023-05-18:
The end game is it is better to use a large sample period instead of having to deal with possible inconsistencies with lv_tick_inc and when it gets called While I do know it is such a small thing but not having to do the math to get the difference between start and stop times for each frame being rendered and adding that time to a total amount of elapsed time is just better resource management IMHO. set the timestamp when it first starts up and set it again each time the calculation is done not 20, 30 or 40 times a second just once is all that is really needed.

kdschlosser, 2023-05-18:
I am sure there is also a better way to to code the `_lv_disp_refr_timer` function then what I did. what I did works and it gives a proof of concept. You should run a test version without changes against one with changes and see what the difference is when running whatever test it is you use to certify devices/displays with. It would be best to run it on an actual display/device instead of on a PC. The thing that I noticed was the faster the refreshing took place the greater the error became. so on a PC if I set the default refresh value to 1 and called lv_tick_int and lv_task_handler every millisecond the error amount grew to almost 100% when rendering simple things on the display. 

OH!! The other thing.. when the display is sitting there idle currently what will happen is the FPS is going to max out at 30. This is because of the default refresh timer value being set to 33. It shouldn't do that.  if nothing is being changed on the display the number should drop to 0 not spike to 30.

I also wanted to know the reasoning behind the 33 value for the refresh timer?  It is not a multiple of 1000 and unless a user specifically knows to change that mostly undocumented macro value it would cause one hell of a performance impact. I say mostly because the documentation for this is not really clear on what it actually does. You also have the same macro used to set the indev callback timer..... Those 2 things should not be tied together.

I was thinking about a way to handle this and it would make more sense to be able to have this done again at runtime. `lv_init` takes no parameters and would be an ideal place to be able to pass the update speeds used. I know the values can be changed in the timer But the push to make lv_disp_t and lv_indev_t private would mean that the fields where the timers are located in those structures would not be known (well at least to a binding they wouldn't be). That would remove the ability to change the vales at runtime... 

You could store both values in a single byte by the user being able to input a value for each in the range 0 to 50. Then use `lv_map` to scale each of them into the 0-15 range and bit shift one of them over 4 bits for storage. 

This is what it would look like.

on the input side.
```c
uint8_t storage = lv_map(indev_input, 0, 50, 1, 15) << 4 | lv_map(indev_input, 0, 50, 1, 15);
```

and collecting the values from the storage
```c
uint8_t indev_refr_timer = lv_map(storage >> 4, 0, 15, 1, 50);
uint8_t disp_refr_timer = lv_map(storage & 0xF, 0, 15, 1, 50);
```

Yes lose some resolution by doing this but nothing I don't feel would be unacceptable. This is what the remap of the values ends up looking like.
```
input values: 0 1 2 3
stored_value: 1
set_timer_value: 4

input values: 4 5 6 7
stored_value: 2
set_timer_value: 7

input values: 8 9 10
stored_value: 3
set_timer_value: 10

input values: 11 12 13 14
stored_value: 4
set_timer_value: 14

input values: 15 16 17
stored_value: 5
set_timer_value: 17

input values: 18 19 20 21
stored_value: 6
set_timer_value: 20

input values: 22 23 24
stored_value: 7
set_timer_value: 23

input values: 25 26 27 28
stored_value: 8
set_timer_value: 27

input values: 29 30 31 32
stored_value: 9
set_timer_value: 30

input values: 33 34 35
stored_value: 10
set_timer_value: 33

input values: 36 37 38 39
stored_value: 11
set_timer_value: 36

input values: 40 41 42
stored_value: 12
set_timer_value: 40

input values: 43 44 45 46
stored_value: 13
set_timer_value: 43

input values: 47 48 49
stored_value: 14
set_timer_value: 46

input values: 50
stored_value: 15
set_timer_value: 50
```

Maybe separating the indev and the disp into 2 different macros and really documenting them in a place that is right up there in front so it can be easily seen and understood. I can tell you that the typical duration a person is in actual contact with a display when touching it for something like tapping a button is < 20 milliseconds. so having a refresh timer check every 33 milliseconds is going to mean missed touches. but setting a display refresh time of something like 15 milliseconds would mean extra work if the users UI is not something that is very active. so it would be a waste of resources.

kisvegabor, 2023-05-19:
> I am not sure if that would correct the problem because the FPS is still being updated for frame that never actually get rendered. 

It's not the case as if we move the `RENDER_READY` event [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L329), it's before the `clean_up`s. And if there was no areas to redraw it [goes to clean up](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#LL299C36-L299C55).

I think the simplest would be to use these RENDER events. We just need to move one line of code to make it work correcrly.

> if nothing is being changed on the display the number should drop to 0 not spike to 30.

It also makes sense. But won't it suggest that the UI is heavily lagging? 

> lv_init takes no parameters and would be an ideal place to be able to pass the update speeds used. [...]  know the values can be changed in the timer But the push to make lv_disp_t and lv_indev_t private would mean that the fields where the timers are located in those structures would not be known (well at least to a binding they wouldn't be). That would remove the ability to change the vales at runtime...

I'd rather add a `lv_disp/indev_set_period()` function.

> You could store both values in a single byte by the user being able to input a value for each in the range 0 to 50. Then use lv_map to scale each of them into the 0-15 range and bit shift one of them over 4 bits for storage.

It sounds too complex to save 3-4 bytes :slightly_smiling_face: 

> Maybe separating the indev and the disp into 2 different macros and really documenting them in a place that is right up there in front so it can be easily seen and understood.

I think using 33ms for both is good starting point. It's a trade off between the number of config options and level of configurability. IMO 99% a single value works, and 1% could use the `lv_disp/indev_set_period()`.

kdschlosser, 2023-05-19:
those functions you mentioned `lv_indev_set_period` and `lv_disp_set_period`.. where are they? I don't see them anywhere and there is nothing in the documentation about them either.

maybe a better choice of a name if they are added would be better. something along the lines of `lv_indev_set_refr_timer_duration` and `lv_disp_set_refr_timer_duration` or something along those lines. Provides a little bit more insight into what the function does.

kdschlosser, 2023-05-19:
if it is showing 0 FPS it indicates that no updates to the screen are being done. for the purposes of the performance monitor being used as a diagnostic aide the FPS not giving a readout of work that is actually being done is not going to be helpful... maybe it  should be expanded instead of changed?.

maybe something like

refresh FPS (number of times per second that refresh is being called)
flush FPS (number of times per second that data is being written to the display) real FPS
render time (how long on average it is taking to render)
flush time (average amount of time it is taking to write the data to the display)

typically the performance monitor is a tool one would use to locate a problem. while it is an informative tool the information it provides can give insight to a user where a bottle neck is occurring.

Here is a fantastic example of such a case.
https://forum.lvgl.io/t/esp32-320x480-low-fps-animating/11799/3

The user is interpreting the low FPS to something being wrong in LVGL when the actual problem is the hardware. This is where a little bit of additional information would be helpful in locating the bottle neck.

I am not the only one that has also said that the FPS calculation is not correct. The reason why this topic has come up from myself and others is because they have added code in their flush function that does a calculation as well and when they see it is not the same number as what they have derived it's a logical conclusion that the way LVGL is doing it is not correct. data being written to the buffer counts as a single frame. and even this is not going to be 100% true either because a lot of times the amount of data being written is not a complete frames worth of data, it is only a small portion of a frame.

kisvegabor, 2023-05-22:
> those functions you mentioned lv_indev_set_period and lv_disp_set_period.. where are they? I don't see them anywhere and there is nothing in the documentation about them either.

By "Id rather add" I meant these needs to be added :slightly_smiling_face: 
In v9 I'd like to get rid of a lot of abbreviations, so `lv_display_set_refresh_period` sounds the best to me. 

> if it is showing 0 FPS it indicates that no updates to the screen are being done.

Actually it"s really logical. There were 0 frames per second. 30 FPS would mean that LVGL drawn 30 frames. However, what to show if LVGL rendered 1 frame in 2 ms. It could be 500 FPS (so very fast), but only 1 frame. Showing 1 FPS would be definitely misleading. 

> refresh FPS (number of times per second that refresh is being called)
flush FPS (number of times per second that data is being written to the display) real FPS
render time (how long on average it is taking to render)
flush time (average amount of time it is taking to write the data to the display)

What about this?
```
CPU 38%, 30 FPS (12 redraws)
Render: 2ms, Flush: 1ms
```
Here FPS is really `min(1000 / (render_time + flush_time), max_fps)`. I'd call it "perceived FPS".

kdschlosser, 2023-05-22:
You have to get rid of that 1000. There is no gurantee that 1000 milliseconds has actually passed. It could be 1250 ms before the user calls lv_task_handler.... I have a saying... "assumptions are the mother of all F*** ups". You should not make the assumption that the calculations are being done exactly when they are supposed to be done. This is why stashing the current ticks and using that when the calculations are done to get the exact amount of time that has passed coupled with the number of frames that have been rendered during that time you end up with a more accurate result. If the callback is added to LVGL for the ticks you add in more accuracy.

This is not something that runs all the time so having additional memory use and processor use is OK. I say the more metrics the better. Give the user the ability to cherry pick the ones they want to see. 

What would be cool is using the same layer to display a "hidden" menu. So on a touch display of the user taps the upper right corner, then the lower left corner, then the upper right corner and maybe the the lower right corner after that and they do that within 2 seconds the menu comes up and performance monitor settings can be changed. 

That would be something that is optional. and LVGL can be compiled without it. It wouldn't consume much in the way of resources when the menu is not open. Put a button in that menu to load the logs if logging is turned on... The user would have to be able to tell LVGL where to save the file but I cannot imagine that would be too hard to do because LVGL already has the file system handlers built into it.

kisvegabor, 2023-05-23:
> You have to get rid of that 1000. There is no gurantee that 1000 milliseconds has actually passed.

Why can't we measure the the time between to calls of a `monitor_timer` using `lv_tick_elaps`. It should be around 1000, but if it's 1379 we will use that as a basis.

 > That would be something that is optional. and LVGL can be compiled without it. It wouldn't consume much in the way of resources when the menu is not open. Put a button in that menu to load the logs if logging is turned on... The user would have to be able to tell LVGL where to save the file but I cannot imagine that would be too hard to do because LVGL already has the file system handlers built into it.

What else shall we display besides:
```
CPU 38%, 30 FPS (12 redraws)
Render: 2ms, Flush: 1ms
```

kdschlosser, 2023-05-23:
ya never know what else may come up. IDK how the code is set up in LVGL so I don't know how things can be timed and it's performance monitored. possibly the monitoring the render times of individual widget is something that might get added in the future. 

OH here is another one. monitoring the times between calls to lv_task_handler and lv_refr_now. so the user is able to know what their code execution time is

kisvegabor, 2023-05-24:
For complex things @FASTSHIFT added a [profiler](https://docs.lvgl.io/master/porting/profiler.html). We can add a lot of profiler calls  to various places of the code  and write a script that draws a chart from it.

kisvegabor, 2023-06-08:
Not stale

kdschlosser, 2023-06-23:
not stale

kisvegabor, 2023-07-10:
@kdschlosser 
The performance monitor was updated by @FASTSHIFT. Please check if the issues listed here were fixed.

kdschlosser, 2023-07-11:
I will check the code to see what is being done. give me a 24 hours to do that.

kdschlosser, 2023-07-11:
nope it is still going to be off.

That is because it is counting when there is nothing that needs to be updated. The `_lv_disp_refr_timer` function gets called not when the display needs to be redrawn but instead it gets called by use of a timer. So the fps is going to be relative to that timer value and not the data being written to the display. The frames per second count is going to be skewed because a frame number gets added to the count and has zero added to the time so the average is going to be higher. The purpsoe for FPS is to measure how long it takes to draw to the display, not how often the timer gets called. 

Look at how the function is.

```c
void _lv_disp_refr_timer(lv_timer_t * tmr)
{
    LV_PROFILER_BEGIN;
    REFR_TRACE("begin");

    if(tmr) {
        disp_refr = tmr->user_data;
#if LV_USE_PERF_MONITOR == 0 && LV_USE_MEM_MONITOR == 0
        /**
         * Ensure the timer does not run again automatically.
         * This is done before refreshing in case refreshing invalidates something else.
         */
        lv_timer_pause(tmr);
#endif
    }
    else {
        disp_refr = lv_disp_get_default();
    }

    if(disp_refr == NULL) {
        LV_LOG_WARN("No display registered");
        return;
    }

    if(disp_refr->draw_buf_size == 0) {
        LV_LOG_WARN("draw_buf_size == 0");
        return;
    }

    if(disp_refr->draw_buf_1 == NULL) {
        LV_LOG_WARN("No draw buffer");
        return;
    }

    lv_disp_send_event(disp_refr, LV_EVENT_REFR_START, NULL);

    /*Refresh the screen's layout if required*/
    lv_obj_update_layout(disp_refr->act_scr);
    if(disp_refr->prev_scr) lv_obj_update_layout(disp_refr->prev_scr);

    lv_obj_update_layout(disp_refr->bottom_layer);
    lv_obj_update_layout(disp_refr->top_layer);
    lv_obj_update_layout(disp_refr->sys_layer);

    /*Do nothing if there is no active screen*/
    if(disp_refr->act_scr == NULL) {
        disp_refr->inv_p = 0;
        LV_LOG_WARN("there is no active screen");
        goto refr_finish;
    }

    lv_refr_join_area();

    refr_invalid_areas();

    if(disp_refr->inv_p == 0) goto refr_finish;

    /*If refresh happened ...*/
    /*Call monitor cb if present*/
    lv_disp_send_event(disp_refr, LV_EVENT_RENDER_READY, NULL);

    if(!lv_disp_is_double_buffered(disp_refr) || disp_refr->render_mode != LV_DISP_RENDER_MODE_DIRECT) goto refr_clean_up;

[... 60 lines cut ...]
```
    while(disp_refr->flushing);
```

really makes zero sense why it's there because when the flushing occurs that field gets set to a 1 BEFORE the flush callback gets called. and it gets set back to a zero once lv_disp_flush_ready gets called. So it is actually stalling LVGL so it is not able to write anything to the second buffer while the first one is emptying.

You really cannot do double buffering with only a single marker indicating that the flush has finished. You would need to have 2 markers, one for each buffer. because at any point in time if one buffer is in the middle of being flushed the flag is going to be True and it is going to stop the second buffer from getting populated. 

The filling and flushing of the buffer occurs here

```
refr_invalid_areas();
```

Then you stall the program here
```   
while(disp_refr->flushing);
```
which is going to be while the first buffer is being emptied. once the buffer has finished being emptied then the buffers get flipped???

That doesn't make any sense to do because the buffer that was just filled is going to be empty because of the stall to the program...

There would be zero benefit to using double buffering, it would end up being a major slow down. That could be where the issue is coming from between version 8.3 and 9 and the reports of 9.0 being slower. 

There is already a stalling of the program before the flush is made. that covers sending a second buffer into the flush function while the first has not finished emptying. The swapping of the buffers should occur directly after that stall but it doesn't occur until after the flush function is called. If it occurred before the flush was called there would be no need to have any other stalls anywhere in the code for waiting for the flush buffer to empty only in that one single location right before the callback gets called to flush the second one. By also swapping the buffers before the callback get made would mean there would be no need to check what buffer is being flushed and what buffer is being emptied anywhere else in the code because the one that is able to be written to will always be in a single location

kdschlosser, 2023-07-12:
I was following the data path through the refresh code and this could be an issue.

```c
    while(disp_refr->flushing);
    /*The buffers are already swapped.
     *So the active buffer is the off screen buffer where LVGL will render*/
    void * buf_off_screen = disp_refr->draw_buf_act;
    void * buf_on_screen = disp_refr->draw_buf_act == disp_refr->draw_buf_1
                           ? disp_refr->draw_buf_2
                           : disp_refr->draw_buf_1;

    lv_coord_t stride = lv_disp_get_hor_res(disp_refr);
    uint32_t i;
    for(i = 0; i < disp_refr->inv_p; i++) {
        if(disp_refr->inv_area_joined[i]) continue;
        disp_refr->layer_head->buffer_copy(
            disp_refr->layer_head,
            buf_off_screen, stride, &disp_refr->inv_areas[i],
            buf_on_screen, stride, &disp_refr->inv_areas[i]
        );
    }
```

This is in the same function mentioned above..

The buffer_copy field in the layer only gets set if software rendering is enabled. If it's not the buffer_copy field will be a NULL pointer and there is no testing to see if the pointer is NULL or not. 

I am not 100% sure what the need is to make a copy from one buffer to the other.

I created a PR that you will be able to see what I am taking about a little bit better..

#4363

closed 2023-08-03 by lvgl-bot