lvgl/lvgl #4223: The CHANGELOG of release v8.3.7 isn't accessible

GorgonMeducer, 2023-05-15:
The CHANGELOG of release v8.3.7 isn't accessible.

https://github.com/lvgl/lvgl/blob/master/docs/CHANGELOG.md

https://github.com/lvgl/lvgl/releases/tag/v8.3.7

### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

break2133, 2023-05-17:
yes

billybednar, 2023-05-17:
The changelog for v8.3.7 is here: https://github.com/lvgl/lvgl/blob/v8.3.7/docs/CHANGELOG.md

The broken link is in all releases since they started being created using GitHub Actions. It has always been to the latest changelog in master and the file was recently renamed during a format conversion. This was somewhat fixed going forward in 8c99b359d21e0d60de64265e724d245b24730cd9, but I think it would be better to link to the changelog in the release rather than the latest one in master. It should be possible to edit the existing releases as well.

kisvegabor, 2023-05-18:
I've manually fixed the link of the CHANGELOGs in the v8.3x to point to `https://github.com/lvgl/lvgl/blob/v8.3.x/docs/CHANGELOG.md`

I've also added the v8.3.7 changelog in CHANGELOG.rst in master.

I' wouldn't complicate the [release.yml](https://github.com/lvgl/lvgl/blob/v8.3.7/.github/workflows/release.yml) to handle `release/v8.3` and `master` differently. Instead I can change the CHANGELOG link manually once in a month for v8.3 patch releases. 

However for v9 we could really improve it to add links by tag to CHANGELOG.rst.

GorgonMeducer, 2023-05-18:
> Instead I can change the CHANGELOG link manually once in a month for v8.3 patch releases.

This would be the best workaround.

closed 2023-05-18 by GorgonMeducer