lvgl/lvgl #4225: Textarea does not work properly with pointer type input devices in grid navigation

yinntian, 2023-05-16:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When POINTER input devices click on a textarea managed by grid navigation, only the cursor blinks, but the KEYPAD input device cannot enter text on it

### To Reproduce
```c
void test() {
    lv_group_t * g = lv_group_get_default();

    lv_obj_t * panel = lv_obj_create(lv_scr_act());
    lv_obj_center(panel);
    lv_obj_set_size(panel, lv_pct(50), lv_pct(50));
    lv_obj_set_flex_flow(panel, LV_FLEX_FLOW_COLUMN);
    lv_group_add_obj(g, panel);
    lv_gridnav_add(panel, LV_GRIDNAV_CTRL_ROLLOVER);

    lv_obj_t * btn_1 = lv_btn_create(panel);
    lv_obj_t * textarea_1 = lv_textarea_create(panel);
    lv_obj_t * textarea_2 = lv_textarea_create(panel);
    lv_obj_t * btn_2 = lv_btn_create(panel);
}
```

### Expected behavior
textarea responds correctly to POINTER input device focus under grid navigation

### Screenshots or video
[image]

kisvegabor, 2023-05-18:
Hi,

it works well here in v8.3. Which LVGL version do you use?

[image]

yinntian, 2023-05-18:
I'm sorry, I forgot to remove textarea from the default group, I don't use the default group in my code, so I should add these lines at the end, my version is 8.3.3
```c
    lv_group_remove(btn1);
    lv_group_remove(btn2);
    lv_group_remove(textarea_1);
    lv_group_remove(textarea_2);
```

kisvegabor, 2023-05-19:
I see now and it's tricky corner case. 

You can do 2 things:
- `lv_obj_clear_flag(textarea1, LV_OBJ_FLAG_CLICK_FOCUSABLE);` after this the textareas wont be focused on click
- add an `LV_EVENT_FOCUSED` event callback to the text areas in which you call `lv_gridnav_set_focused(panel, lv_event_get_target(e), LV_ANIM_OFF);` to manually make the clicked (focused) item as the focused gridnav item.

yinntian, 2023-06-07:
The suggestion you gave me seems a bit problematic, there will still be two textareas in focus at the same time. However, I found a workaround that required group-focused settings for container objects in event callbacks.

Here is the complete code
```c
void textarea_handle(lv_event_t * e){
    lv_obj_t * cont = lv_event_get_user_data(e);
    lv_obj_t * obj = lv_event_get_target(e);
    lv_group_focus_obj(cont);
    lv_gridnav_set_focused(cont, obj, LV_ANIM_OFF);
}

void test() {
    lv_group_t * g = lv_group_get_default();

    lv_obj_t * panel = lv_obj_create(lv_scr_act());
    lv_obj_center(panel);
    lv_obj_set_size(panel, lv_pct(50), lv_pct(50));
    lv_obj_set_flex_flow(panel, LV_FLEX_FLOW_COLUMN);
    lv_group_add_obj(g, panel);
    lv_gridnav_add(panel, LV_GRIDNAV_CTRL_ROLLOVER);

    lv_obj_t * btn_1 = lv_btn_create(panel);
    lv_obj_t * textarea_1 = lv_textarea_create(panel);
    lv_obj_t * textarea_2 = lv_textarea_create(panel);
    lv_obj_t * btn_2 = lv_btn_create(panel);

    lv_group_remove_obj(btn1);
    lv_group_remove_obj(btn2);
    lv_group_remove_obj(textarea_1);
    lv_group_remove_obj(textarea_2);

    lv_obj_clear_flag(textarea_1, LV_OBJ_FLAG_CLICK_FOCUSABLE);
    lv_obj_clear_flag(textarea_2, LV_OBJ_FLAG_CLICK_FOCUSABLE);
}
```

closed 2023-06-07 by yinntian