lvgl/lvgl #4226: Missing object init in msgbox and test

billybednar, 2023-05-17:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
`lv_msgbox_create` and `test_event_object_deletion` create objects using `lv_obj_class_create_obj` but never call `lv_obj_class_init_obj` on them. As a result the object constructor never runs and the created objects do not have the default flags nor any theme styles. For the message box, this does not cause problems in most cases; the object in question does not have any styles in the default theme and the missing flags generally don't make a difference when no custom styles are applied.

https://github.com/lvgl/lvgl/blob/6a47c6f33ac51a69cf241d43b74933c09c5d175f/src/widgets/msgbox/lv_msgbox.c#L113-L115
https://github.com/lvgl/lvgl/blob/6a47c6f33ac51a69cf241d43b74933c09c5d175f/tests/src/test_cases/test_event.c#L23-L25

### To Reproduce
```c
static const char * const buttons[] = { "Ok", "" };
lv_obj_t *msgbox = lv_msgbox_create(NULL, NULL, "1\n2\n3\n4\n5\n6\n7\n8\n9\n10", buttons, true);
// Uncomment to make scrolling work
//lv_obj_class_init_obj(lv_msgbox_get_content(msgbox));
lv_obj_set_style_max_height(lv_msgbox_get_content(msgbox), 50, 0);
```

kisvegabor, 2023-05-18:
Thank you for reporting it and also describing the problem. I've just pushed a fix.

closed 2023-05-18 by kisvegabor

kisvegabor, 2023-05-18:
And a fix for the fix: 6a89bd2d7 :slightly_smiling_face:

commit 6843c191 (2023-05-18): fix(msgbox): add missing lv_obj_class_init_obj

fixes #4226
commit e3876cbf (2023-05-18): fix(msgbox): add missing lv_obj_class_init_obj

fixes #4226