lvgl/lvgl #4229: Performance Monitor Error

kimyj1503, 2023-05-19:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The background and fonts on the performance monitor are not showing as expected. There are 'broken pixel' on the performance monitor. 

### To Reproduce

lv_conf.h - related configurations
```c
#define LV_DISP_DEF_REFR_PERIOD 30      /*[ms]*/
#define LV_COLOR_SCREEN_TRANSP 1

#define LV_USE_PERF_MONITOR 1
#if LV_USE_PERF_MONITOR
    #define LV_USE_PERF_MONITOR_POS LV_ALIGN_TOP_MID     /*I moved the position but the bug is not fixed*/
#endif
```

main.c
```c
#include <stdbool.h>
#include <stdio.h>

/* FreeRTOS */
#include <FreeRTOS.h>
#include <task.h>

/* bl808 c906 std driver */
#include <bl808_glb.h>

#include "demos/lv_demos.h"
#include "lv_port_disp.h"
#include "lv_port_indev.h"
#include "lvgl.h"

static void lvgl_task(void *param)
{
    while (1) {
        lv_task_handler();
        vTaskDelay(1);
    }
    vTaskDelete(NULL);
}

void main()
{
    lv_init();

    lv_port_disp_init();
    lv_demo_benchmark();
    lv_task_handler();

    xTaskCreate(lvgl_task, (char *)"lvgl task", 512, NULL, 15, NULL);
}
```

### Expected behavior

I expected there is no 'broken pixels', i.e. the background is gray and the font is white without random unwanted pixels.

### Screenshots or video

The first photo shows the fonts that is working perfectly.
The second photo shows the performance monitor with 'broken pixels'.

[image]
[image]

kisvegabor, 2023-05-22:
What is the value of `LV_COLOR_DEPTH` in `lv_conf.h`?

kimyj1503, 2023-05-22:
> What is the value of `LV_COLOR_DEPTH` in `lv_conf.h`?

It's defined as 16.
`#define LV_COLOR_DEPTH LCD_COLOR_DEPTH
`

kisvegabor, 2023-05-23:
It's really strange. Which LVGL version do you use?

kimyj1503, 2023-05-30:
> It's really strange. Which LVGL version do you use?

I tested it on both v8.3.1 and v8.3.7.

I changed the opacity value from 50% to 100% with `lv_obj_set_style_bg_opa(perf_label, LV_OPA_100, 0);` on `src/core/lv_refr.c`.

Due to my camera problem, it looks quite blurry, but I can confirm that the 'broken pixels' are no longer visible. I'll further check if my board has some problem with the opacity settings.

[image]

closed 2023-06-22 by lvgl-bot