lvgl/lvgl #4231: access violation error

kdschlosser, 2023-05-19:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I am working on writing the CPython binding for LVGL as you know. I keep on getting an access violation error when setting the opacity of an object. The problem is caused by the `convert_cb` function in lv_draw_sw_img.c file. Line 264

I generated pdb files when I compiled it so I ran the program using the Visual Studio debugger and that was where it indicated the problem to be.

 
[image]

kdschlosser, 2023-05-22:
This is all set. It turned out to be an issue with the builtin memory management specifically using lv_malloc. The allocations weren't correct somehow. When I changed over to using malloc, free, realloc, ...etc. from the stdlib everything started to play nice.

How was your holiday?

kisvegabor, 2023-05-23:
Maybe is way just an out of memory issue than. 

> 
This is all set. It turned out to be an issue with the builtin memory management specifically using lv_malloc. The allocations weren't correct somehow. When I changed over to using malloc, free, realloc, ...etc. from the stdlib everything started to play nice.

> How was your holiday?

A little bit colder than I expected but we could make our feet wet in the sea :slightly_smiling_face:

closed 2023-05-23 by kisvegabor