lvgl/lvgl #4233: Question about how the code is written that handles styles.

kdschlosser, 2023-05-19:
in lv_style.c there is this function.

```c
static void lv_style_set_prop_internal(lv_style_t * style, lv_style_prop_t prop_and_meta, lv_style_value_t value,
                                       void (*value_adjustment_helper)(lv_style_prop_t, lv_style_value_t, uint16_t *, lv_style_value_t *));
```

If you look at the last parameter for that function you can see there is a function pointer that is expected to be passed to that function.

Now if we look at another function that calls the above function

```c
void lv_style_set_prop(lv_style_t * style, lv_style_prop_t prop, lv_style_value_t value)
{
    lv_style_set_prop_internal(style, prop, value, lv_style_set_prop_helper);
}
```

we can see the last parameter being passed is `lv_style_set_prop_helper`. now if we look at the prototype for the `lv_style_set_prop_helper` function.

```c
static void lv_style_set_prop_helper(lv_style_prop_t prop, lv_style_value_t value, uint16_t * prop_storage,
                                     lv_style_value_t * value_storage);
```

notice that it is not a pointer.

Is that correct? should it be like that?. I know it runs like that but the question is why does it work?.. I bring this question up because I am having an issue #4321 when setting the opacity. The error occurs some other place but I cannot help thinking that the error might be caused by this and not with where the debugger says the problem is. 

The other question I have is when a structure gets passed to a function by value does the structure get copied? From what I have read is if a structure is passed as a value a copy of that structure gets made.. if we start off at the point of origin for setting a style, lets say the opacity of the background as an example a structure gets made in the first function and that structure gets passed to the next by value. it actually gets passed 5 times by value until it finally gets set. in the case of setting the opacity that ends up equating to 20 bytes of memory used to store a 1 byte property. I say 25 bytes because the 1 byte property gets converted into a uint32_t.

I know this memory allocation is not a static allocation and doesn't get held after the function(s) return but it sure cannot help with fragmentation of memory. which has a higher impact on a binding like the micropython binding that only frees the memory periodically and not when the function(s) return. can a pointer be passed instead and the data copied from that pointer into the structure that is stored instead? would it make more sense to do that to reduce dynamic memory allocation?

kisvegabor, 2023-05-22:
It seems correct to me.  `lv_style_set_prop_internal` needs a callback as last parameter and it gets `lv_style_set_prop_helper`.

>  I am having an issue #4321 when setting the opacity

I think you meant #4231. I commented there.

> The other question I have is when a structure gets passed to a function by value does the structure get copied?

Yes, it will be copied. 

> I say 25 bytes because the 1 byte property gets converted into a uint32_t.

Do you mean this?
```c
typedef union {
    int32_t num;         /**< Number integer number (opacity, enums, booleans or "normal" numbers)*/
    const void * ptr;    /**< Constant pointers  (font, cone text, etc)*/
    lv_color_t color;    /**< Colors*/
} lv_style_value_t;
```

If so, it's only 4 bytes (on 32 bit) as it's an union 

.

kdschlosser, 2023-05-22:
I understand that it's a union that is only 4 bytes of data. But it get passed as a value through 5 different functions before it gets to where it needs to be. so if it is copied each time that adds up to 20 right there. The last 5 is one for the initial uint8_t for setting the opa and the last 4 is for when it gets placed into the very same structure that was passed to the first function,

A copy means a duplicate, as in now you have 2 instead of 1. twice the memory consumption.

If it was passed as a pointer you could eliminate all but 4 bytes (size of a 32 bit pointer)

kisvegabor, 2023-05-23:
> A copy means a duplicate, as in now you have 2 instead of 1. twice the memory consumption.

However it's in stack. Using 20-30 extra bytes from stack is not a big deal. It's the same if we passed `int32_t` 5 level deep. Passing 4 byte is 1 instruction and the used memory will be freed when the function exists.

> If it was passed as a pointer you could eliminate all but 8 bytes, 4 bytes for the union being initialized and then 4 bytes for the pointer.

Passing a pointer is also 4 bytes and passing it 5 times is 20 bytes.

> I cannot imagine that is doesn't add to memory fragmentation.

Stack is different from heap:
- stack: local variables and function arguments are "stacked"  on top of each other and  when the function exists it frees all the memory it used with one instruction. (just changes the stack pointer which points to the last free memory)
- heap: malloc works in heap and it really can be fragmented.

kdschlosser, 2023-05-23:
I understand the heap and stack thing and memory management works differently in Python then it does with C code (even tho they kind of wound together). Things have to get garbage collected so even a local that is out of scope doesn't get freed right away. It probably does in C code I am sure.

kisvegabor, 2023-05-24:
However after calling e.g. `lv.obj_set_style_border_width(obj, 2, 0)` from Python everything happens in  C, so C memory management applies, right?

kdschlosser, 2023-05-24:
nope it doesn't. that's the goofy part. CFFI generates an 80k line source file that wraps every aspect of the LVGL code. the memory management largely gets handled by Python. This is the reason why I was having issues with the builtin memory management. Windows is more sensitive and the extension module needs to be compiled using the same compiler and version that was used to compile Python. Things have to be done largely the same in both the extension module and also in Python, that includes how memory gets allocated. It's a real bitch to contend with. sorry about the bad language there is simply no other word that describes it in the correct way.

kisvegabor, 2023-05-24:
I mean you see only `lv_obj_set_style_border_width()`. It calls
```
lv_obj_set_local_style_prop -> lv_style_set_prop -> lv_style_set_prop_internal
```

But when you call `lv.obj_set_style_border_width` (from Python) it calls `lv_obj_set_style_border_width` (in C) so the last 3 calls are in C, right? So from passing the `value` parameter 5 times 1 or 2 happens in Python the rest is in C.

kdschlosser, 2023-05-25:
just because it takes place in C doesn't mean the memory gets handled in the same way.  I know there is some really wacky stuff that CFFI does with respect to memory. 

As an example. If you pass a structure from Python into C code via a function and if that function passes that structure to another function by value which would make a copy right? There needs to be a python reference to that object even tho it has technically been copied. If that reference is no longer available the data even in the copy is no longer usable.

kisvegabor, 2023-05-25:
I mean you parse only the headers of LVGL, so CFFI doesn't know anything about what happens after the top level API calls (as the internal magic are hidden into C files, which are not processed by CFFI).

kdschlosser, 2023-06-05:
OK I have another question that is basically just a coding question.

```c
static inline uint16_t lv_color16_to_int(lv_color16_t c)
{
    uint16_t * p = (uint16_t *)&c;  <-- WHYY???
    return *p;
}
```

in C parameters passed to a function is done as a "copy" unless it is specifically stated to use a pointer.

Why is there the need to create a pointer to the value of a pointer that gets returned as a value?
Type punning can lead to undefined behavior I also think it is slower code than doing the bitwise operation.

The most efficient is using a union but that has changed and the `lv_color*_t` are no longer unions. They have been changed to a structure. That remove the fast way of doing conversion from the r, g, b color values into they integer representations. That is going to make a pretty large impact when writing color data to a pixel buffer. I know that people have mentioned performance losses between versions, if there is a performance drop between 8.3 and 9.0 I would imagine this is a large portion of that loss. 

It stinks that c99 doesn't support anonymous structures and unions because I believe the reasoning behind this change was to get rid of the second level ("ch") in the unions but doing so comes at a cost.

```c
static inline uint16_t lv_color16_to_int(lv_color16_t c)
{
    return  (uint16_t) c.red << 11 | c.green << 5 | c.blue
}
```

Another option is to use the following. I read that it might cause endianness issues but I could not find anything that really explained the use case that would cause that issue.
```c
static inline uint16_t lv_color16_to_int(lv_color16_t c)
{
    uint16_t p;
    lv_memcpy(&p, &c, sizeof(c));
    return p;
}
```

kisvegabor, 2023-06-05:
I've removed the `full` element from the union because There is no integer data type we could use for `RGB888` colors. There is no `uint24_t`.

> Type punning can lead to undefined behavior I also think it is slower code than doing the bitwise operation.

Here we know what we are doing, so it's safe :slightly_smiling_face:

>  return  (uint16_t) c.red << 11 | c.green << 5 | c.blue

In the end it should really matter if you do `c16.red` or `(some_int >> 11) & 0x1F` as the compiler need to do the same thing. So it shouldn't affect the performance.

kdschlosser, 2023-06-06:
> There is no integer data type we could use for RGB888 colors. 

I see what you are doing now. took me a bit to see it. This is kind of what you are doing now.

```c

typedef struct {
   uint8_t blue;
   uint8_t green;
   uint8_t red;
} lv_color24_t;

typedef struct {
   uint8_t blue;
   uint8_t green;
   uint8_t red;
   uint8_t alpha
} lv_color32_t;

#if LV_COLOR_DEPTH == 24
    #define lv_color_t lv_color24_t
#else
    #define lv_color_t lv_color32_t

lv_color_t color_data = {.red = 100, .green = 50, .blue = 75};

#if LV_COLOR_DEPTH == 24
    lv_color_t color_array[4];
    color_array[0] = color_data;
    color_array[1] = color_data;
    color_array[2] = color_data;
    color_array[3] = color_data;
#else
    lv_color_t color_array[3];
    color_array[0] = color_data;
    color_array[1] = color_data;
    color_array[2] = color_data;

uint8_t buf[] = (uint8_t *) color_data;

```

you can advance the array using `sizeof(lv_color_t)` to write single lv_color_t structures or to create an lv_color_t structure from a specific position in the buffer.

I had to key it out to see it.

my code above is probably wrong, it was only for the purposes of making me think.

kisvegabor, 2023-06-06:
> you can advance the array using sizeof(lv_color_t) to write single lv_color_t structures or to create an lv_color_t structure from a specific position in the buffer.

It's not that simple in the practice because some platform/architectures doesn't support  unaligned access. E.g. read `uint32_t` from 0x1234567 address.

Anyway, this part will change soon. Take look at lv_color.h in the [parallel rendering branch](https://github.com/lvgl/lvgl/blob/feat/parallel-rendering/src/misc/lv_color.h). Here `lv_color_t` is always RGB888 regardless of `LV_COLOR_DEPTH`.

kdschlosser, 2023-06-06:
it is???

```c
typedef struct {
    uint8_t blue;
    uint8_t green;
    uint8_t red;
} lv_color_t;

typedef struct {
    uint16_t blue : 5;
    uint16_t green : 6;
    uint16_t red : 5;
} lv_color16_t;

typedef struct {
    uint8_t blue;
    uint8_t green;
    uint8_t red;
    uint8_t alpha;
} lv_color32_t;
```

then why the different `lv_color*_t` structures?

kdschlosser, 2023-06-06:
I think the field types in this is a boo boo.

```c

typedef struct {
    uint16_t blue : 5;
    uint16_t green : 6;
    uint16_t red : 5;
} lv_color16_t;
```

kisvegabor, 2023-06-07:
> then why the different lv_color*_t structures?

These are just helpers in LVGL renders to/from 16 or 32 bit color formats.

> I think the field types in this is a boo boo.

What's the boo-boo here? It looks good to me and it's working too.

kdschlosser, 2023-06-07:
uint16_t data type for each of the colors? shouldn't it be uint8_t?

kisvegabor, 2023-06-08:
`:5` means use 5 bits. :slightly_smiling_face:

kdschlosser, 2023-06-08:
Oh that's right so it doesn't matter what the data type is so long as it can hold 5 bits.

But if one was after correctness it should be uint8_t

pete-pjb, 2023-06-08:
Hi @kdschlosser ,

With this type of 'C' declaration you are declaring variables in numbers of bits which are then concatenated together into the root type, so it crams all the structure members into one uint16_t type.

So the structure:
```
typedef struct {
    uint16_t blue : 5;
    uint16_t green : 6;
    uint16_t red : 5;
} lv_color16_t;
```

looks like this in memory:
```
Memory Bits:      15 14 13 12 11 10 09 08 07 06 05 04 03 02 01 00
Colour Bits:      B4 B3 B2 B1 B0 G5 G4 G3 G2 G1 G0 R4 R3 R2 R1 R0
```

I hope that makes it clearer. 😊 

Kind Regards,

Pete

kdschlosser, 2023-06-08:
OK that makes sense.

still learning

closed 2023-06-23 by kdschlosser