lvgl/lvgl #4238: imgbtn images cannot be loaded using variables

corinna-bxb, 2023-05-23:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ v] Update lvgl to the latest version
- [ v] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

imgbtn images cannot be loaded using variables
### To Reproduce

 char i_icone[] = "H:/Downloads/DECK/60_60/StreamDeckIcon_VDM-4051.png";
avatar3= lv_imgbtn_create(panel1);
 lv_obj_set_size(avatar3 , 60, 60);
lv_imgbtn_set_src(avatar3 , LV_IMGBTN_STATE_RELEASED, NULL, i_icone, NULL);

### Expected behavior

When I use img, I can use variables to load pictures, but when I use imgbtn, there is no way, the program will always be wrong and cannot be opened

ex:
lv_img_set_src(avatar, i_icone);  ====>ok
lv_imgbtn_set_src(avatar3 , LV_IMGBTN_STATE_RELEASED, NULL, i_icone, NULL);  ===>error
### Screenshots or video

kisvegabor, 2023-05-23:
Hi,

Which LVGL version do you use? It was working for me with `master`:
```c
  lv_obj_t * obj = lv_imgbtn_create(lv_scr_act());
  lv_imgbtn_set_src(obj, LV_IMGBTN_STATE_RELEASED, NULL, "A:lvgl/demos/widgets/assets/avatar.png", NULL);
```

Please also enable the [logging](https://docs.lvgl.io/master/porting/log.html).

corinna-bxb, 2023-05-24:
Version v8.3
I want imgbtn to use variables, because the file name will change
It is ok for me to fill in the location directly like you, but there is no way to use variables,
I tested that img can use variables, but imgbtn will be wrong
log crazy display"
[Warn] (34.420, +10) _lv_img_cache_open: Image draw cannot open the image resource (in lv_img_cache.c line #125)
[Warn] (34.430, +10) lv_draw_img: Image draw error (in lv_draw_img.c line #81)

kisvegabor, 2023-05-24:
Ahh, sorry, I misread you comment and thought files are not working. 

I tink the problem is that you have missed an `&` here:
```c
lv_imgbtn_set_src(avatar3 , LV_IMGBTN_STATE_RELEASED, NULL, &i_icon, NULL);
```

But you should see at least a compiler warning about it.

kisvegabor, 2023-05-29:
Do you see any compiler warnings?

corinna-bxb, 2023-05-29:
NO
Excuse me, how is your test result?

kisvegabor, 2023-05-30:
You can try out `lv_example_imgbtn_1();` which use variables and work well for me.

corinna-bxb, 2023-05-31:
I am referring to the variables for accessing files as described above, which is different from the following examples
`void lv_example_imgbtn_1(void)
{
    LV_IMG_DECLARE(imgbtn_left);
    LV_IMG_DECLARE(imgbtn_right);
    LV_IMG_DECLARE(imgbtn_mid);

    /*Create a transition animation on width transformation and recolor.*/
    static lv_style_prop_t tr_prop[] = {LV_STYLE_TRANSFORM_WIDTH, LV_STYLE_IMG_RECOLOR_OPA, 0};
    static lv_style_transition_dsc_t tr;
    lv_style_transition_dsc_init(&tr, tr_prop, lv_anim_path_linear, 200, 0, NULL);

    static lv_style_t style_def;
    lv_style_init(&style_def);
    lv_style_set_text_color(&style_def, lv_color_white());
    lv_style_set_transition(&style_def, &tr);

    /*Darken the button when pressed and make it wider*/
    static lv_style_t style_pr;
    lv_style_init(&style_pr);
    lv_style_set_img_recolor_opa(&style_pr, LV_OPA_30);
    lv_style_set_img_recolor(&style_pr, lv_color_black());
    lv_style_set_transform_width(&style_pr, 20);

    /*Create an image button*/
    lv_obj_t * imgbtn1 = lv_imgbtn_create(lv_scr_act());
    lv_imgbtn_set_src(imgbtn1, LV_IMGBTN_STATE_RELEASED, &imgbtn_left, &imgbtn_mid, &imgbtn_right);
    lv_obj_add_style(imgbtn1, &style_def, 0);
    lv_obj_add_style(imgbtn1, &style_pr, LV_STATE_PRESSED);

    lv_obj_align(imgbtn1, LV_ALIGN_CENTER, 0, 0);

    /*Create a label on the image button*/
    lv_obj_t * label = lv_label_create(imgbtn1);
    lv_label_set_text(label, "Button");
    lv_obj_align(label, LV_ALIGN_CENTER, 0, -4);
}`
please help to try to access images in the file system, see if there is a solution, thank you

kisvegabor, 2023-05-31:
I see now.  The problem with 
```c
char i_icon[] = "H:/Downloads/DECK/60_60/DeckIcon_VDM-4051.png";
```

is that LVGL saves only the address of the variable and if `i_icon` is a local variable it will the dropped when its function exists. You can do this instead:
```c
static char i_icon[256]; //Global or in a function

//Later
sprintf(i_icon,"H:/Downloads/DECK/60_60/DeckIcon_AAA45678.png");
```

corinna-bxb, 2023-06-16:
Thanks, it's ready to use
Each image variable of imgbtn needs to be set as a global variable and cannot be shared
Very different from img, img test does not have this requirement, because imgbtn has its own independent settings for pressing and releasing, so is this required?

kisvegabor, 2023-06-16:
For both lv_img and lv_imgtn only the address of the image source is saved, so you need to be sure that the variable you pass as image source is no a local variable.

> img test does not have this requirement

What do you mean by "img test"?

corinna-bxb, 2023-06-19:
I use lv_img, there is no need to use global variables, you can use regional variables, the actions are all normal, only lv_imgbtn needs to use global variables

closed 2023-07-12 by lvgl-bot