lvgl/lvgl #4244: lv_roller does not translate mouse movements when widget is rotated

kdschlosser, 2023-05-25:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
if setting a transform_angle on a roller widget no matter the angle it is set to only up and down moves the roller. The roller should move perpendicular to the angle it is set at. so if it is set at 45° the mouse movement should be 135° degree mouse angle for rolling down and 315° for rolling up. 

### To Reproduce
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/57325dfa8742b5fee09ccf4f370ce23656fcb315/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/57325dfa8742b5fee09ccf4f370ce23656fcb315/examples/widgets/roller/lv_example_roller_3.py&script_direct=66cb28cf70f182ea486e134c69a412740ea0c216

### Expected behavior
to be able to move the mouse in the angled direction and have the roller move  like it is supposed to

kisvegabor, 2023-05-29:
Hi,

I've just fixed it.

closed 2023-05-29 by kisvegabor

kdschlosser, 2023-05-29:
sweet. Thanks!!! I am messing about with different parts and pieces and seeing if there are any issues or odd behaviors. While I know that V9 is not considered "stable" for all intents and purposes I didn't think that the mouse movements was something that was being tinkered with in version 9 and that it is an actual bug.

kisvegabor, 2023-05-29:
I wouldn't call it a bug but rather a not implemented feature :smile:

kdschlosser, 2023-05-29:
OK that would explain it. Did it have to be implemented directly into the roller widget? If so I am willing to bet that you probably need to go through all of the widgets to add this because this is the second one I have found where the issue exists.

kisvegabor, 2023-05-30:
> I am willing to bet that you probably need to go through all of the widgets to add this because this is the second one I have found where the issue exists.

No all actually. Almost all widgets are using the built in scrolling of LVGL which handles  transformations automatically. However there area a few widget which do some custom magic and they needs to be fixed case-by-case.

commit 87fbaad4 (2023-05-29): feat(roller): make scrolling handle transformations

fixes #4244