lvgl/lvgl #4256: In new master branch compile error

zhzhn1234, 2023-05-29:
I want to use ssd1306 128x64 oled in my lvgl code(master branch v9.0), ssd1306 is 1bit per pixel,so i add 
#define LV_COLOR_DEPTH 1  
but lvgl comipile error as below:

===============================================================
n file included from ../../src/misc/lv_log.h:19:0,
                 from ../../lvgl.h:24,
                 from lv_demo_flex_layout_main.h:17,
                 from lv_demo_flex_layout_view.c:10:
../../src/core/../misc/lv_color.h: 在函数‘lv_color_set_int’中:
../../src/core/../misc/lv_color.h:359:16: 警告： 隐式声明函数‘lv_color1_set_int’ [-Wimplicit-function-declaration]
     LV_CONCAT3(lv_color, LV_COLOR_DEPTH, _set_int(c, v));
                ^
../../src/misc/lv_types.h:86:30: 附注： in definition of macro ‘_LV_CONCAT3’
 #define _LV_CONCAT3(x, y, z) x ## y ## z
                              ^
../../src/core/../misc/lv_color.h:359:5: 附注： in expansion of macro ‘LV_CONCAT3’
     LV_CONCAT3(lv_color, LV_COLOR_DEPTH, _set_int(c, v));
     ^~~~~~~~~~
../../src/core/../misc/lv_color.h: 在函数‘lv_color_to_int’中:
../../src/core/../misc/lv_color.h:386:23: 警告： 隐式声明函数‘lv_color1_to_int’ [-Wimplicit-function-declaration]
     return LV_CONCAT3(lv_color, LV_COLOR_DEPTH, _to_int(c));
                       ^
../../src/misc/lv_types.h:86:30: 附注： in definition of macro ‘_LV_CONCAT3’
 #define _LV_CONCAT3(x, y, z) x ## y ## z
                              ^
../../src/core/../misc/lv_color.h:386:12: 附注： in expansion of macro ‘LV_CONCAT3’
     return LV_CONCAT3(lv_color, LV_COLOR_DEPTH, _to_int(c));
            ^~~~~~~~~~
../../src/core/../misc/lv_color.h: 在函数‘lv_color_from_buf’中:
../../src/core/../misc/lv_color.h:418:23: 警告： 隐式声明函数‘lv_color1_from_buf’ [-Wimplicit-function-declaration]
     return LV_CONCAT3(lv_color, LV_COLOR_DEPTH, _from_buf(buf));
                       ^
../../src/misc/lv_types.h:86:30: 附注： in definition of macro ‘_LV_CONCAT3’
 #define _LV_CONCAT3(x, y, z) x ## y ## z
                              ^
../../src/core/../misc/lv_color.h:418:12: 附注： in expansion of macro ‘LV_CONCAT3’
     return LV_CONCAT3(lv_color, LV_COLOR_DEPTH, _from_buf(buf));
            ^~~~~~~~~~
../../src/core/../misc/lv_color.h:418:23: 错误： 将‘int’返回为‘lv_color_t {或称 union <匿名>}’时类型不兼容
     return LV_CONCAT3(lv_color, LV_COLOR_DEPTH, _from_buf(buf));
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../src/misc/lv_types.h:86:30: 附注： in definition of macro ‘_LV_CONCAT3’
===============================================================

**what should I do to make ssd1306 work well in lvgl?**

kisvegabor, 2023-05-29:
Hi,

The master branch is still under development and not all parts of it are fully stable. Please use the `release/v8.3` branch.

zhzhn1234, 2023-05-29:
> Hi,
> 
> The master branch is still under development and not all parts of it are fully stable. Please use the `release/v8.3` branch.

how to use realease/v8.3 to work on ssd1306 1bit/pixel ?
need #define LV_COLOR_DEPTH 1 ? any other settings else? could you give me some suggestions?

kisvegabor, 2023-05-30:
With `LV_COLOR_DEPTH 1` you will still get 1 byte/pixel.
If you need a more dense format, you can use the [`set_px_cb`](https://docs.lvgl.io/8.3/get-started/platforms/arduino.html).

closed 2023-06-02 by zhzhn1234