lvgl/lvgl #4258: add image decoder hooks for GPU special requirements

terry0012, 2023-05-29:
### Introduce the problem
some GPUs have special requirements for image process.
1. special requirements for buffers when rendering images.  For example: address alignment
2. After the decoder completes, there may be some cache synchronization operations that need to be performed.

### Examples and cases

### Suggested solution
1. we need to provide a private allocation function to hardware vendors during image decoding.
2. we also need a callback function for post-processing by the image decoder.

Can this patch (https://github.com/lvgl/lvgl/pull/3294) meet these two requirements?
If so, when is the expected time for the patch to be merged?
If not, we would like to implement this requirement ourselves and contribute it to the community.

kisvegabor, 2023-05-29:
Hi Terry,

Please take a look at this part in https://github.com/lvgl/lvgl/pull/3294#issuecomment-1374512677

This `event_cb` idea looks quite flexible but I'd like to understand better what you reaaly need:
> we need to provide a private allocation function to hardware vendors during image decoding.

What kind of decoder do you have? Is it a custom format or a common format already supported by LVGL?
It might be related to https://github.com/lvgl/lvgl/pull/4241

> we also need a callback function for post-processing by the image decoder.

If I understand correctly it should be called when the image is decoded for cache invalidation. Are there any other requirements besides cache invalidation?

terry0012, 2023-05-30:
> What kind of decoder do you have? Is it a custom format or a common format already supported by LVGL? It might be related to #4241

A: The common format already supported by LVGL, like png, gif.

> If I understand correctly it should be called when the image is decoded for cache invalidation. Are there any other requirements besides cache invalidation?

A: Yes, it should be called after image is decoded.
Besides cache invalidation, there maybe stride alignment for image buffer in GPU.
In addition, "the event_cb idea" in https://github.com/lvgl/lvgl/pull/3294#issuecomment-1374512677 seems to meet post-processing requirement.

terry0012, 2023-05-31:
> To support stride we should modify the decoders (e.g. png) too. Doesn't seem impossible, just mentioning.
> 
> Can you confirm that using a platform specific `lv_malloc_draw_buf(size, flags)` functions to allocate buffers for decoding can solve the issue of "we need to provide a private allocation function to hardware vendors during image decoding."

A:  yes, lv_malloc_draw_buf can solve this issue.

> Implementing the `event_cb` on images would take some time, and there are a lot of open questions. Can we manage the problem of cache invalidation by always calling a `lv_draw_buf_invalidate_cache(buf, buf_size)` function after decoding?

A: I agree with it.

kisvegabor, 2023-05-31:
Great! In this case let's continue in https://github.com/lvgl/lvgl/pull/3294.

It seems with this we can easily solve multiple issues at once.

closed 2023-05-31 by kisvegabor